warmcat/lws-esp32-factory #2: Failing to Mount ROMFS

tedwoodt25, 2017-05-02:
V (1220) event: exit default callback
D (1220) esp_ota_ops: found ota app max = 1
D (1230) esp_ota_ops: finding factory app......
D (1230) nvs: nvs_open lws-station 1
D (1230) nvs: nvs_get_str_or_blob ssl-pub.der
D (1240) nvs: nvs_get_str_or_blob ssl-pri.der
I can build flash and upload to my Core Board V2. But startup fails with this message:

D (1240) nvs: nvs_close 3
4: No SSL certs... using port 80
4: Initial logging level 63
4: Libwebsockets version: 2.2.0 tedwood@Teds-MacBook-Pro.local-v2.0.0-365-gff151d0
4: IPV6 not compiled in
4:  Threads: 1 each 30 fds
4:  mem: platform fd map:   120 bytes
4:  Compiled with OpenSSL support
4:  SSL disabled: no LWS_SERVER_OPTION_DO_SSL_GLOBAL_INIT
4:  mem: per-conn:          432 bytes + protocol rx buf
1: Failed to mount ROMFS at 0xdbb34 0xdbb34
4: setting mount default to factory

lws-team, 2017-05-02:
What does the factory build process say about the ROMFS?  Here it says

```
             Original length: 0x0ba470 (  762992)
    After ROMFS + Build info: 0x0e3589 (  931209)
```

and if I look in build, I see

```
-rw-rw-r--. 1 agreen agreen  931209 May  2 18:56 lws-esp32-factory.bin
```

Did the build proceed OK otherwise on your Mac?

tedwoodt25, 2017-05-02:
I'm seeing:

 Original length: 0x000000 ( 0)
 After ROMFS + Build info: 0x02a92d (  174381)

and

-rw-r--r--  1 tedwood  staff   174381  2 May 12:28 /Users/tedwood/lws-esp32-factory/main/../build/lws-esp32-factory.bin

lws-team, 2017-05-02:
Mmmm sounds like your build is broke somewhere.  So far we know it works well on Fedora or Ubuntu, both Linux.  Mac should not be a million miles away.

Can you delete ./build and then cut and paste here the whole build log.

lws-team, 2017-05-02:
(and if you made any changes to get it this far, please let me know them).

tedwoodt25, 2017-05-02:
I had to add gstat in place of OSX's stat, and make sure that ranlib called the ESP32 version - at the moment it calls whichever ranlib it finds first in the path.  I fixed it by just making sure it looks in the /bin directory of the ESP32 compiler first, but this isn't a particularly robust solution

make all
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o mconf.o mconf.c
flex -L -P zconf -o zconf.lex.c zconf.l
zconf.l:255: warning, -s option given but default rule can be matched
bison -t -l -p zconf -o zconf.tab.c zconf.y
sed -E "s/\\x0D$//" zconf.gperf | gperf -t --output-file zconf.hash.c -a -C -E -g -k '1,3,$' -p -t
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o zconf.tab.o zconf.tab.c
lxdialog/check-lxdialog.sh -check cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE -lncurses 
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/checklist.o lxdialog/checklist.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/util.o lxdialog/util.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/inputbox.o lxdialog/inputbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/textbox.o lxdialog/textbox.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/yesno.o lxdialog/yesno.c
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o lxdialog/menubox.o lxdialog/menubox.c
cc -o mconf mconf.o zconf.tab.o lxdialog/checklist.o lxdialog/util.o lxdialog/inputbox.o lxdialog/textbox.o lxdialog/yesno.o lxdialog/menubox.o -lncurses 
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o conf.o conf.c
cc -o conf conf.o  zconf.tab.o -lncurses 
GENCONFIG
cp: /Users/tedwood/lws-esp32-factory/main/../build/lws-esp32-factory.bin: No such file or directory
stat: cannot stat '/Users/tedwood/lws-esp32-factory/main/../build/lws-esp32-factory.bin': No such file or directory
             Original length: 0x000000 (       0)
    After ROMFS + Build info: 0x02a92d (  174381)
CC src/bootloader_flash.o
CC src/bootloader_random.o
CC src/efuse.o
CC src/esp_image_format.o
CC src/flash_encrypt.o
CC src/flash_partitions.o
CC src/secure_boot.o
CC src/secure_boot_signatures.o
AR libbootloader_support.a
CC log.o
AR liblog.a
CC spi_flash_rom_patch.o
AR libspi_flash.a
CC micro-ecc/uECC.o
AR libmicro-ecc.a
CC esp32/brownout.o
CC esp32/cpu_util.o
CC esp32/rtc_clk.o
CC esp32/rtc_init.o
CC esp32/rtc_pm.o
CC esp32/rtc_sleep.o
CC esp32/rtc_time.o
AR libsoc.a
CC bootloader_start.o
CC flash_qio_mode.o
AR libmain.a
LD bootloader.elf
esptool.py v2.0-beta3
Building partitions from /Users/tedwood/lws-esp32-factory/partitions.csv...
CC esp_ota_ops.o
AR libapp_update.a
CC src/bootloader_flash.o
CC src/bootloader_random.o
CC src/efuse.o
CC src/esp_image_format.o
CC src/flash_encrypt.o
CC src/flash_partitions.o
CC src/secure_boot.o
CC src/secure_boot_signatures.o
AR libbootloader_support.a
CC bt.o
AR libbt.a
CC libcoap/src/address.o
CC libcoap/src/async.o
CC libcoap/src/block.o
CC libcoap/src/coap_time.o
[... 524 lines cut ...]
CC eri.o
CC trax.o
AR libxtensa-debug-module.a
CC adler32.o
CC compress.o
CC crc32.o
CC deflate.o
CC gzclose.o
CC gzlib.o
CC gzread.o
CC gzwrite.o
CC infback.o
CC inffast.o
CC inflate.o
CC inftrees.o
CC trees.o
CC uncompr.o
CC zutil.o
AR libzlib.a
CC main.o
AR libmain.a
LD lws-esp32-factory.elf
esptool.py v2.0-beta3
To flash all build output, run 'make flash' or:
python2 /Users/tedwood/RPK/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/cu.SLAB_USBtoUART --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /Users/tedwood/lws-esp32-factory/build/bootloader/bootloader.bin 0x10000 /Users/tedwood/lws-esp32-factory/build/lws-esp32-factory.bin 0x8000 /Users/tedwood/lws-esp32-factory/build/partitions.bin

lws-team, 2017-05-02:
It seems the make rule is disordered...

```
cp: /Users/tedwood/lws-esp32-factory/main/../build/lws-esp32-factory.bin: No such file or directory
stat: cannot stat '/Users/tedwood/lws-esp32-factory/main/../build/lws-esp32-factory.bin': No such file or directory
             Original length: 0x000000 (       0)
    After ROMFS + Build info: 0x02a92d (  174381)
```

This should happen right at the end, after it has done the build.

Can you try use GNU make if not doing so already?

ranlib is in esp-idf part, lws doesn't run it itself as far as I recall.

tedwoodt25, 2017-05-02:
Using GNU make V3.81

lws-team, 2017-05-02:
Hmmm  try this as a workaround

```
 $ make all pack.img flash monitor
```

tedwoodt25, 2017-05-02:
It still doesn't make pack.img at the end. It gets to the end of the build and says "pack.img is up to date"

lws-team, 2017-05-02:
Ehhh... :-) there shouldn't be a file pack.img.  If there is, delete it.

Can you edit this bit of `./components/libwebsockets/scripts/esp32.mk` around line 20

```
.PHONY: romfs.img
pack.img:
...
```

Add another .PHONY after the first

```
.PHONY: pack.img
```

see if that helps.

tedwoodt25, 2017-05-02:
No change. Still showing  "make: Nothing to be done for `pack.img'" at end of build

.

lws-team, 2017-05-02:
How about

```
 $ make all ; make pack.img
```

tedwoodt25, 2017-05-02:
It still doesn't mount ROMFS

make pack.img
cp: /Users/tedwood/lws-esp32-factory/main/../build/lws-esp32-factory.bin and /Users/tedwood/lws-esp32-factory/main/../build/lws-esp32-factory.bin are identical (not copied).
             Original length: 0x0cbb50 (  834384)
    After ROMFS + Build info: 0x0f647d ( 1008765)

Question: what is the purpose of that cp ? It always seems to copy to itself.

lws-team, 2017-05-02:
The cp is a bug, it's harmless.

Actually that result shows now we can work around the disordered make part of the problem.  Which considering I don't have your problem or platform, is not bad.

My working image is smaller than yours

```
 Original length: 0x0ba470 ( 762992) 
After ROMFS + Build info: 0x0e3589 ( 931209) 
```

How about

 - Hack `components/libwebsockets/scripts/esp32.mk` again removing line 76 or so

```
all: pack.img
```

 - delete ./build and retry

tedwoodt25, 2017-05-02:
Same size data as before

I have to say I'm impressed by your ability to debug this without having the platform in front of you.

lws-team, 2017-05-02:
It's bedtime where I am... the code walks the ELF sections at runtime to find the end of the original esp32 image to which the romfs is appended.  So please paste the startup log, the build log lines for that exact same image like

```
 Original length: 0x0ba470 ( 762992)
 After ROMFS + Build info: 0x0e3589 ( 931209)
```

and let me know the exact esp-idf commit you are using.  Any changes to sdkconfig?

Dunno if it's necessary but emailing the .bin to andy@warmcat.com might throw some light too.

tedwoodt25, 2017-05-02:
Really appreciate your help with this.

using esp-idf commit  07b61d54f7e502bb12527e4b582de121ffa6fa02
Original length: 0x0b1ed0 (  728784)
After ROMFS + Build info: 0x0daffb (  897019)

Hard resetting...
MONITOR
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2492
ho 0 tail 12 room 4
load:0x40078000,len:8920
ho 0 tail 12 room 4
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

E (50) boot: force_factory 0 bs.ota_info.offset 53248

I (1022) cpu_start: Pro cpu up.
I (1023) cpu_start: Starting app cpu, entry point is 0x40080dd8
0x40080dd8: call_start_cpu1 at /Users/tedwood/lws-esp32-factory/esp-idf/components/esp32/./cpu_start.c:173

I (0) cpu_start: App cpu up.
I (1032) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1055) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1075) heap_alloc_caps: At 3FFB9148 len 00026EB8 (155 KiB): DRAM
I (1096) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1118) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1139) heap_alloc_caps: At 40092658 len 0000D9A8 (54 KiB): IRAM
I (1160) cpu_start: Pro cpu start user code
I (1219) cpu_start: Starting scheduler on PRO CPU.
I (204) cpu_start: Starting scheduler on APP CPU.
I (234) wifi: wifi firmware version: ba7c292
I (234) wifi: config NVS flash: disabled
I (234) wifi: config nano formating: disabled
I (234) wifi: Init dynamic tx buffer num: 32
I (234) wifi: Init dynamic rx buffer num: 64
I (234) wifi: wifi driver task: 3ffc2004, prio:23, stack:3584
I (244) wifi: Init static rx buffer num: 10
I (244) wifi: Init dynamic rx buffer num: 0
I (254) wifi: Init rx ampdu len mblock:7
I (254) wifi: Init lldesc rx ampdu entry mblock:4
I (264) wifi: wifi power manager task: 0x3ffc73ac prio: 21 stack: 2560
E (264) wifi: wifi_set_chan_range 1694 ret=1
I (274) wifi: wifi timer task: 3ffc842c, prio:22, stack:3584
I (304) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 0
I (304) wifi: mode : sta (24:0a:c4:02:05:ec) + softAP (24:0a:c4:02:05:ed)
4: No SSL certs... using port 80
4: Initial logging level 63
4: Libwebsockets version: 2.2.0 tedwood@Teds-MacBook-Pro.local-v2.0.0-365-gff151d0
4: IPV6 not compiled in
4:  Threads: 1 each 30 fds
4:  mem: platform fd map:   120 bytes
4:  Compiled with OpenSSL support
4:  SSL disabled: no LWS_SERVER_OPTION_DO_SSL_GLOBAL_INIT
4:  mem: per-conn:          432 bytes + protocol rx buf
1: Failed to mount ROMFS at 0xc1ed4 0xc1ed4
4: setting mount default to factory

lws-team, 2017-05-02:
The ROMFS offset computation is actually correct... the partition starts at +0x10000, the length before it was 0xb1ed0 and there should be a 32-bit ROMFS size before the ROMFS itself.

When you flash it, it is actually flashing 897019 bytes?

What is in your image at that offset?  Here it (using my image size number) looks like

```
$ hexdump -s 762992  build/lws-esp32-factory.bin -n32 -C
000ba470  00 90 02 00 2d 72 6f 6d  31 66 73 2d 00 02 8e 00  |....-rom1fs-....|
000ba480  d4 fb 02 b7 72 6f 6d 20  35 39 30 38 38 63 32 30  |....rom 59088c20|
```
It shows the ROMFS length word (0x29000) and then the romfs image.

What does `hexdump -s 728784  build/lws-esp32-factory.bin -n32 -C` show you?

tedwoodt25, 2017-05-04:
Sorry - was at conference yesterday. 

Total number of bytes written to flash is actually 743600 

esptool.py v2.0-beta3
Connecting........____
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 11744 bytes to 7519...
Wrote 11744 bytes (7519 compressed) at 0x00001000 in 0.1 seconds (effective 842.1 kbit/s)...
Hash of data verified.
Compressed 728784 bytes to 452644...
Wrote 728784 bytes (452644 compressed) at 0x00010000 in 7.9 seconds (effective 738.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 110...
Wrote 3072 bytes (110 compressed) at 0x00008000 in 0.0 seconds (effective 9690.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

Hexdump looks like this:

Teds-MacBook-Pro:lws-esp32-factory tedwood$ hexdump -s 728784 build/lws-esp32-factory.bin -n32 -C
00b1ed0 e9 08 02 20 24 0e 08 40 00 00 00 00 00 00 00 00
00b1ee0 00 00 00 00 00 00 00 00 00 00 00 00 e8 ff 00 00
00b1ef0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

tedwoodt25, 2017-05-04:
Ah.

After a make all; make pack.img:

Teds-MacBook-Pro:lws-esp32-factory tedwood$ hexdump -n64  -s 728784  -C build/lws-esp32-factory.bin 
000b1ed0  00 90 02 00 2d 72 6f 6d  31 66 73 2d 00 02 8e 00  |....-rom1fs-....|
000b1ee0  d8 88 ee 95 72 6f 6d 20  35 39 30 62 30 37 36 34  |....rom 590b0764|
000b1ef0  00 00 00 00 00 00 00 49  00 00 00 20 00 00 00 00  |.......I... ....|
000b1f00  d1 ff ff 97 2e 00 00 00  00 00 00 00 00 00 00 00  |................|
000b1f10

After a make all flash
Teds-MacBook-Pro:lws-esp32-factory tedwood$ hexdump -n64  -s 728784  -C build/lws-esp32-factory.bin 
000b1ed0  e9 08 02 20 24 0e 08 40  00 00 00 00 00 00 00 00  |... $..@........|
000b1ee0  00 00 00 00 00 00 00 00  00 00 00 00 e8 ff 00 00  |................|
000b1ef0  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|

Make flash recreates lws-esp32-factory.bin from scracth each time.

If I do
make all
make pack.img

and then call the python loader script direct, it flashes the romfs and starts up OK

johsely1234, 2017-05-04:
I had a similiar problem, but it was my fault.
I compiled with make flash instead of make **all** flash. (on Ubuntu)
Just for people who might have the same problem.

lws-team, 2017-05-04:
Ok... so for mac build IIUI

 - makefile script needs gstat instead of stat... the underlying problem is the format string %s means 'size' in GNU stat and is undefined in bsd stat (size is %z there but in GNU stat, that means file date).

 - have to build with `make all; make pack.img flash monitor`

I pushed a patch on lws favouring gstat if it exists, plus added a note in the READMEs about the make workaround for mac, and updated -factory and -test-server-demos.  So I think that's it for this one.

closed 2017-05-04 by lws-team

lws-team, 2017-05-04:
Mm no I see I missed your point about make flash regenerating the file... how about hack in a stanza 

```
flash: pack.img

```

in `components/libwebsockets/scripts/esp32.mk`?

tedwoodt25, 2017-05-05:
That didn't work. It still rebuilds the image as part of make flash  and doesn't redo make pack.img afterwards....

lws-team, 2017-05-05:
I still don't understand what is different processing the aggregate makefiles under Linux and under bsd... both apparently with GNU make.  I am not the only person building this just fine.

Can you try a later GNU make?  My version is from Fedora 26 alpha, 4.2.1, but you don't need it to be that recent, since it built fine on F25 and Ubuntu reportedly.

tedwoodt25, 2017-05-05:
Thats it. gmake 4.2.1 builds it correctly

lws-team, 2017-05-05:
Ehhh lol... OK I will update the READMEs accordingly... thanks for your help debugging it

lws-team, 2017-05-11:
I just pushed an update to the build stuff that fixes problems with parallel build (make -j n) and should remove this make version requirement thing.