warmcat/lws-esp32-factory #3: Crashing in Factory Setup after reload, general instability

johsely1234, 2017-05-04:
I have a problem that the lws esp32 factory app is crashing. The easiest way to reproduce it is following:

I installed everything according to the guide. Everything compiles fine.
[image]

After download I connect from Laptop to esp the I start Browser and go to 192.168.4.1
It loads the site, then i try to reload the page and the esp crashes.

I dont know what the first error means.

`E (48) boot: force_factory 0 bs.ota_info.offset 53248

I (1077) cpu_start: Pro cpu up.
I (1078) cpu_start: Starting app cpu, entry point is 0x40080dd8
I (0) cpu_start: App cpu up.
I (1088) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1110) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1130) heap_alloc_caps: At 3FFB8868 len 00027798 (157 KiB): DRAM
I (1152) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1173) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1195) heap_alloc_caps: At 40099D24 len 000062DC (24 KiB): IRAM
I (1215) cpu_start: Pro cpu start user code
I (1275) cpu_start: Starting scheduler on PRO CPU.
I (205) cpu_start: Starting scheduler on APP CPU.
I (215) wifi: wifi firmware version: a8ef40a
I (215) wifi: config NVS flash: disabled
I (215) wifi: config nano formating: disabled
I (215) wifi: Init dynamic tx buffer num: 32
I (225) wifi: wifi driver task: 3ffc1598, prio:23, stack:3584
I (225) wifi: Init static rx buffer num: 10
I (235) wifi: Init dynamic rx buffer num: 0
I (235) wifi: Init rx ampdu len mblock:7
I (235) wifi: Init lldesc rx ampdu entry mblock:4
I (245) wifi: wifi power manager task: 0x3ffc6940 prio: 21 stack: 2560
E (245) wifi: wifi_set_chan_range 1669 ret=1
I (255) wifi: wifi timer task: 3ffc79c0, prio:22, stack:3584
I (315) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 0
I (315) wifi: mode : sta (24:0a:c4:04:26:dc) + softAP (24:0a:c4:04:26:dd)
4: No SSL certs... using port 80
4: Initial logging level 63
4: Libwebsockets version: 2.2.0 root@esp32dev-VirtualBox-v2.0.0-365-gff151d0
4: IPV6 not compiled in
4:  Threads: 1 each 30 fds
4:  mem: platform fd map:   120 bytes
4:  Compiled with OpenSSL support
4:  SSL disabled: no LWS_SERVER_OPTION_DO_SSL_GLOBAL_INIT
4:  mem: per-conn:          432 bytes + protocol rx buf
4: ROMFS length 164KiB

{
 "schema": "lws1",
 "model": "lws",
 "builder": "esp32dev-VirtualBox",
 "app": "lws-esp32-factory",
 "user": "root",
 "git": "2331748-dirty",
 "date": "Don Mai  4 10:14:15 CEST 2017",
 "unixtime": "1493885655",
 "file": "lws-esp32-factory-1493885655.bin",
 "factory": "y"
}
4: Creating Vhost 'ap' port 80, 3 protocols, IPv6 off
4:    mounting file:///ap to /
4:    mounting callback://esplws-scan to /factory
4:    mounting callback://esplws-ota to /otaform
4: setting mount default to factory
I (17845) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (17845) wifi: station: 9c:b7:0d:a0:1b:bc join, AID=1, g, 20
W (17865) wifi: buffer null

W (17885) wifi: buffer null

4: esp32_lws_fops_open: /ap/factory.html
4: esp32_lws_fops_open: /ap/factory.png
4: esp32_lws_fops_open: /ap/product.png
4: esp32_lws_fops_open: /ap/ssl-pub.png
4: esp32_lws_fops_open: /ap/warning.png
4: esp32_lws_fops_open: /ap/factory-backdrop.jpg
4: esp32_lws_fops_open: /ap/ssl-pri.png
4: esp32_lws_fops_open: /ap/favicon.ico
4: lws_header_table_detach: wsi 0x3ffcf01c: ah held 5s, ah.rxpos 0, ah.rxlen 0, mode/state 0 4,wsi->more_rx_waiting 0
4: lws_header_table_detach: wsi 0x3ffcf1d4: ah held 6s, ah.rxpos 0, ah.rxlen 0, mode/state 0 4,wsi->more_rx_waiting 0
4: esp32_lws_fops_open: /ap/factory.html
4: esp32_lws_fops_open: /ap/factory-backdrop.jpg
4: esp32_lws_fops_open: /ap/factory.png
4: esp32_lws_fops_open: /ap/product.png
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400e08e6  PS      : 0x00060530  A0      : 0x800e12ea  A1      : 0x3ffc13a0 
A2      : 0x00000000  A3      : 0x3ffcd3f4  A4      : 0x40118184  A5      : 0x3ffcd038 
A6      : 0x3ffcd618  A7      : 0x3ffb74fc  A8      : 0x800e08e4  A9      : 0x3ffc1380 
A10     : 0x00000000  A11     : 0x00000036  A12     : 0x3ffcd3f4  A13     : 0x3ffcd038 
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c 
EXCVADDR: 0x00000004  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff 

Backtrace: 0x400e08e6:0x3ffc13a0 0x400e12ea:0x3ffc1470 0x40092350:0x3ffc1490 0x40093c7a:0x3ffc14e0

Entering gdb stub now.
$T0b#e6
`

This is the output of the GDBstub:

#0  0x400e07b8 in hostap_input ()
#1  0x400e11ba in ap_rx_cb ()
#2  0x40092350 in ppRxPkt ()
#3  0x40093c7a in ppTask ()

To compile the debug version I had to fix this error:

components/libwebsockets/plugins/protocol_esp32_lws_ota.c:91:18: error: 'part' may be used uninitialized in this function [-Werror=maybe-uninitialized]

just initialize it with 0.

Any ideas on this topic?

lws-team, 2017-05-04:
I could reproduce it with esp-idf from a few days ago that I had.

Can't reproduce it with today's esp-idf 8f9707fd07571293954c3b36e2e7532b500dd758

Generally lws doesn't do anything special for AP mode.  If it's blowing up in AP mode, it's most likely esp-idf.  Please try updating that.

tedwoodt25, 2017-05-04:
I'm seeing this same error, using that same esp-idf commit ending 00dd758

johsely1234, 2017-05-04:
Yes it works better with that version of esp-idf.

Sometimes it shows 
1: ERROR on accept: 
but continues working.

Yes could really be an issue with esp-idf, I had some problems too with AP.

But it is still crashing after reloading some times.

lws-team, 2017-05-04:
@tedwoodt25 you see the same error if refreshing the browser?  Or also the first time?

It feels like some race regarding socket close.  But on my side I call close() there's not supposed to be something else to do because AP mode; this does not exist in station mode.  So I don't see what I should / can do about this.

tedwoodt25, 2017-05-05:
It seems to occur at the point where I try to join the AP, which as you say suggests a problem in esp-idf. 

It doesn't always happen - about 10% of attempts to connect to the AP are successful

lws-team, 2017-05-05:
Hm...

1) it ever dies the FIRST time you try to connect after reset?

2) It's exactly the same backtrace?

3) As I say I can no longer reproduce it here even camping on F5... it works.  So maybe related to exact AP type you have?  I dunno.

Either way, lws is not responsible for generic AP mode functionality... esp-idf is supposed to provide it... it can still be we need to do something different but I'm waiting for the clue as to what.

tedwoodt25, 2017-05-05:
1. Yes. It dies the first time you try to connect. Although, even if you do connect it eventually dies. I haven't capture the GDB out from this yet
2. Backtrace exactly the same each time
3. Understood

lws-team, 2017-05-05:
For 2) I meant exactly the same backtrace as OP told on this bug... or something else?

tedwoodt25, 2017-05-05:
No my back trace is different but repeatable. Addresses are all similar. I think he was using an older version of esp-idf Or we may have different configs I guess.

lws-team, 2017-05-05:
Aka "you may have some completely different problem".

Please send your BT when you are able to.

tedwoodt25, 2017-05-05:
I (199774) wifi: station: 00:bc:bf:0f:64:ce leave, AID = 1
I (199774) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (216834) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (216834) wifi: station: 00:bc:bf:0f:64:ce join, AID=1, n, 20
I (304954) wifi: station: 00:bc:bf:0f:64:ce leave, AID = 1
I (304954) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (312084) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (312094) wifi: station: 00:bc:bf:0f:64:ce join, AID=1, n, 20
I (350094) wifi: station: 00:bc:bf:0f:64:ce leave, AID = 1
I (350104) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (365264) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (365264) wifi: station: a0:99:9b:0b:b4:99 join, AID=1, g, 20
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400e0546  PS      : 0x00060930  A0      : 0x800e0f4a  A1      : 0x3ffc1f90  
0x400e0546: hostap_input at ??:?

A2      : 0x000000fa  A3      : 0x3ffcdbcc  A4      : 0x4011f6d8  A5      : 0x3ffcdc78  
0x4011f6d8: tcpip_adapter_ap_input at /Users/tedwood/RPK/esp-idf/components/tcpip_adapter/./tcpip_adapter_lwip.c:854

A6      : 0x3ffce224  A7      : 0x3ffb7ce4  A8      : 0x800e0544  A9      : 0x3ffc1f70  
A10     : 0x00000000  A11     : 0x0000004e  A12     : 0x3ffcdbcc  A13     : 0x3ffcdc78  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000fe  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400e0546:0x3ffc1f90 0x400e0f4a:0x3ffc2060 0x400f3388:0x3ffc2080 0x400f4a96:0x3ffc20d0
0x400e0546: hostap_input at ??:?
0x400e0f4a: ap_rx_cb at ??:?
0x400f3388: ppRxPkt at ??:?
0x400f4a96: ppTask at ??:?

tedwoodt25, 2017-05-05:
MONITOR
--- idf_monitor on /dev/cu.usbserial-DN02MUXN 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:2492
ho 0 tail 12 room 4
load:0x40078000,len:8920
ho 0 tail 12 room 4
load:0x40080000,len:252
0x40080000: _iram_start at ??:?

entry 0x40080034
0x40080034: _iram_start at ??:?

E (66) boot: force_factory 0 bs.ota_info.offset 53248

I (1038) cpu_start: Pro cpu up.
I (1039) cpu_start: Starting app cpu, entry point is 0x40080dd8
0x40080dd8: call_start_cpu1 at /Users/tedwood/RPK/esp-idf/components/esp32/./cpu_start.c:173

I (0) cpu_start: App cpu up.
I (1048) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1071) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1091) heap_alloc_caps: At 3FFB9150 len 00026EB0 (155 KiB): DRAM
I (1112) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1134) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1155) heap_alloc_caps: At 40092658 len 0000D9A8 (54 KiB): IRAM
I (1176) cpu_start: Pro cpu start user code
I (1235) cpu_start: Starting scheduler on PRO CPU.
I (204) cpu_start: Starting scheduler on APP CPU.
I (224) wifi: wifi firmware version: ba7c292
I (224) wifi: config NVS flash: enabled
I (224) wifi: config nano formating: disabled
I (244) wifi: Init dynamic tx buffer num: 32
I (244) wifi: Init dynamic rx buffer num: 50
I (244) wifi: wifi driver task: 3ffc0bb0, prio:23, stack:3584
I (244) wifi: Init static rx buffer num: 10
I (254) wifi: Init dynamic rx buffer num: 50
I (254) wifi: Init rx ampdu len mblock:7
I (254) wifi: Init lldesc rx ampdu entry mblock:4
I (264) wifi: wifi power manager task: 0x3ffc7494 prio: 21 stack: 2560
E (264) wifi: wifi_set_chan_range 1694 ret=1
I (274) wifi: wifi timer task: 3ffc84fc, prio:22, stack:3584
W (274) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (504) phy: phy_version: 350, Mar 22 2017, 15:02:06, 1, 2
I (504) wifi: mode : sta (24:0a:c4:04:ad:d8) + softAP (24:0a:c4:04:ad:d9)
4: No SSL certs... using port 80
4: Initial logging level 63
4: Libwebsockets version: 2.2.0 tedwood@Teds-MacBook-Pro.local-v2.0.0-365-gff151d0
4: IPV6 not compiled in
4:  Threads: 1 each 30 fds
4:  mem: platform fd map:   120 bytes
4:  Compiled with OpenSSL support
4:  SSL disabled: no LWS_SERVER_OPTION_DO_SSL_GLOBAL_INIT
4:  mem: per-conn:          432 bytes + protocol rx buf
4: ROMFS length 164KiB
{
 "schema": "lws1",
 "model": "RPK",
 "builder": "Teds-MacBook-Pro.local",
 "app": "lws-esp32-factory",
 "user": "tedwood",
 "git": "2331748-dirty",
 "date": "Fri  5 May 2017 12:22:25 BST",
 "unixtime": "1493983345",
 "file": "lws-esp32-factory-1493983345.bin",
 "factory": "y"
}
4: Creating Vhost 'ap' port 80, 3 protocols, IPv6 off
4:    mounting file:///ap to /
4:    mounting callback://esplws-scan to /factory
4:    mounting callback://esplws-ota to /otaform
4: setting mount default to factory
I (24824) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (24824) wifi: station: a0:99:9b:0b:b4:99 join, AID=1, g, 20
W (24864) wifi: buffer null

4: esp32_lws_fops_open: /ap/factory.html
4: esp32_lws_fops_open: /ap/product.png
4: esp32_lws_fops_open: /ap/factory.png
4: esp32_lws_fops_open: /ap/ssl-pub.png
4: esp32_lws_fops_open: /ap/warning.png
W (31014) wifi: buffer null

W (31014) wifi: buffer null

4: esp32_lws_fops_open: /ap/ssl-pri.png
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400e0548  PS      : 0x00060f30  A0      : 0x800e0f4a  A1      : 0x3ffc22c0  
0x400e0548: hostap_input at ??:?

A2      : 0x3ffcd2bc  A3      : 0x3ffcdf7c  A4      : 0x00000000  A5      : 0x3ffcd2bc  
A6      : 0x3ffce218  A7      : 0x3ffb7ce4  A8      : 0x800e0544  A9      : 0x3ffc22a0  
A10     : 0x00000000  A11     : 0x000000d9  A12     : 0x3ffcdf7c  A13     : 0x3ffcd2bc  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400e0548:0x3ffc22c0 0x400e0f4a:0x3ffc2390 0x400f3388:0x3ffc23b0 0x400f4a96:0x3ffc2400
0x400e0548: hostap_input at ??:?

0x400e0f4a: ap_rx_cb at ??:?

0x400f3388: ppRxPkt at ??:?

0x400f4a96: ppTask at ??:?

lws-team, 2017-05-05:
Hm it says it blows in 0x4011f6d8: tcpip_adapter_ap_input at /Users/tedwood/RPK/esp-idf/components/tcpip_adapter/./tcpip_adapter_lwip.c:854

But I doubt it's true since it's the end of tcpip_adapter_get_sta_list() and nothing calls that from inside esp-idf AFAICT.

tedwoodt25, 2017-05-05:
The addresses might not match but the call stack looks similar to the original report

lws-team, 2017-05-05:
Yes it's the same problem.  But the backtrace source stuff seems wrong, so I don't have anywhere to go with it.

lws-team, 2017-05-06:
Today I am bringing up some new PCBs and spending more time with AP mode... looking at the PC address in the map file, it is actually dying in hostap_input as it says, which lives in their object-only library libnet80211.a.

The wrongly-reported source for the BT address can be looked up in the map file manually, it is the caller of hostap_input at tcpip_adapter_ap_input(), tcpip_adapter_lwip.c:809.

Dumping the args on that doesn't give any clue when it blows

```
E (101995) tcpip_adapter: buffer 0x3ffdf9dc, len 54, eb 0x3ffd8e0c
E (102005) tcpip_adapter: buffer 0x3ffe0910, len 54, eb 0x3ffd8e60
E (102015) tcpip_adapter: buffer 0x3ffdea10, len 54, eb 0x3ffd8284
Guru Meditation Error...
```

basically in blows in espressif's closed source 802111 library blob.

tedwoodt25, 2017-05-10:
This appears to be fixed in latest esp-idf

lws-team, 2017-05-10:
Great... I meddled with sdkconfig and it went away here a few days ago.  But I also updated esp-idf... at any rate it has been happy here.

closed 2017-05-11 by lws-team