warmcat/lws-esp32-factory #4: Unable to update submodule

Scalpel78, 2017-05-09:
Hi, wanting to test the factory image I've done a git clone of this repository.

    $ git clone https://github.com/warmcat/lws-esp32-factory.git

However, when I run this:

    $ git submodule update --init --recursive

I get this error:

    frode@Mint18 ~/Temp/lws-esp32-factory $ git submodule update --init --recursive
    fatal: reference is not a tree: f79ba408276c6e5485a6f23e3e42e887aab85fa5
    Unable to checkout 'f79ba408276c6e5485a6f23e3e42e887aab85fa5' in submodule path 
    'components/libwebsockets'

Googling this seems to indicate that there is a missing commit for the submodule.

lws-team, 2017-05-09:
It should be fixed now.

Scalpel78, 2017-05-09:
It is, thanks!

closed 2017-05-09 by Scalpel78