warmcat/lws-esp32-factory #5: Unable to load /factory.html

Scalpel78, 2017-05-09:
Hi, 
I got the image working for a little while, but now it has stopped responding to /factory.html.

The output from 'make monitor' is:

E (50) boot: force_factory 0 bs.ota_info.offset 53248

I (1027) cpu_start: Pro cpu up.
I (1027) cpu_start: Single core mode
I (1028) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (1041) heap_alloc_caps: At 3FFAE2A0 len 00001D60 (7 KiB): DRAM
I (1062) heap_alloc_caps: At 3FFB8AF0 len 00027510 (157 KiB): DRAM
I (1083) heap_alloc_caps: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (1104) heap_alloc_caps: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1126) heap_alloc_caps: At 4009272C len 0000D8D4 (54 KiB): IRAM
check b=0x3ffae2ac size=7496 ok
check b=0x3ffafff4 size=0 ok
check b=0x3ffb8afc size=161016 ok
check b=0x3ffdfff4 size=0 ok
check b=0x3ffe044c size=15272 ok
check b=0x3ffe3ff4 size=0 ok
check b=0x3ffe435c size=113816 ok
check b=0x3ffffff4 size=0 ok
check b=0x40092738 size=55484 ok
I (1225) cpu_start: Pro cpu start user code
I (1283) cpu_start: Starting scheduler on PRO CPU.
I (1361) wifi: wifi firmware version: ba7c292
I (1361) wifi: config NVS flash: disabled
I (1361) wifi: config nano formating: disabled
I (1371) wifi: Init dynamic tx buffer num: 20
I (1371) wifi: Init dynamic rx buffer num: 20
I (1371) wifi: wifi driver task: 3ffc13d8, prio:23, stack:3584
I (1381) wifi: Init static rx buffer num: 20
I (1381) wifi: Init dynamic rx buffer num: 20
I (1391) wifi: Init rx ampdu len mblock:7
I (1391) wifi: Init lldesc rx ampdu entry mblock:4
I (1391) wifi: wifi power manager task: 0x3ffca8a8 prio: 21 stack: 2560
E (1401) wifi: wifi_set_chan_range 1694 ret=1
I (1411) wifi: wifi timer task: 3ffcb978, prio:22, stack:3584
I (1431) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 0
I (1431) wifi: mode : sta (24:0a:c4:03:cd:60) + softAP (24:0a:c4:03:cd:61)
4: No SSL certs... using port 80
4: Initial logging level 63
4: Libwebsockets version: 2.2.0 frode@Mint18-v2.0.0-380-ga7def3c
4: IPV6 not compiled in
4:  LWS_WITH_STATS        : on
4:  Threads: 1 each 30 fds
4:  mem: platform fd map:   120 bytes
4:  Compiled with OpenSSL support
4:  SSL disabled: no LWS_SERVER_OPTION_DO_SSL_GLOBAL_INIT
4:  mem: per-conn:          456 bytes + protocol rx buf
1: Failed to mount ROMFS at 0xc2f14 0xc2f14
4: setting mount default to factory
I (2071) wifi: n:6 1, o:6 0, ap:6 1, sta:0 0, prof:6
I (2071) wifi: station: e0:94:67:f6:fc:ac join, AID=1, n, 40U

Going to http://192.168.4.1/factory.html just gives a 404. My computer is attached to the AP, as I can see it list my mac address in the 'make monitor' output, but the webserver does not respond.

The problem _might_ have occured after I uploaded wrong files for the SSL certificates, but I seem to be unable to recorver, even after doing 'make erase_flash' and also flashing other binary images, and back again to this lws-esp32-factory image.

Is there anything other than 'make erase_flash' that can be run to wipe everything?

lws-team, 2017-05-09:
> 1: Failed to mount ROMFS at 0xc2f14 0xc2f14

Take a look in `./build/`, you should see something like

```
-rw-rw-r--. 1 agreen agreen  901849 May 10 02:13 lws-esp32-factory.bin
```

When you flash it, are you flashing 901849 (or whatever it is thereabouts) bytes?

How old's your `make` version?

Scalpel78, 2017-05-10:
Yes, I have, amongst others these (after a 'make clean' and 'make all').

drwxr-xr-x 7 frode frode 4,0K May 10 07:29 lwip
-rw-r--r-- 1 frode frode 881K May  9 20:21 lws-esp32-factory-1494354097.bin
-rw-r--r-- 1 frode frode 881K May  9 20:25 lws-esp32-factory-1494354342.bin
-rw-r--r-- 1 frode frode 165K May  9 20:27 lws-esp32-factory-1494354421.bin
-rw-r--r-- 1 frode frode 881K May  9 20:28 lws-esp32-factory-1494354537.bin
-rw-r--r-- 1 frode frode 1,1M May  9 20:39 lws-esp32-factory-1494355189.bin
-rw-r--r-- 1 frode frode 881K May  9 20:41 lws-esp32-factory-1494355279.bin
-rw-r--r-- 1 frode frode 165K May  9 20:45 lws-esp32-factory-1494355520.bin
-rw-r--r-- 1 frode frode 165K May  9 20:49 lws-esp32-factory-1494355750.bin
-rw-r--r-- 1 frode frode 165K May 10 07:29 lws-esp32-factory-1494394152.bin
-rw-r--r-- 1 frode frode 716K May 10 07:29 lws-esp32-factory.bin
-rwxr-xr-x 1 frode frode 3,6M May 10 07:29 lws-esp32-factory.elf
-rw-r--r-- 1 frode frode 2,5M May 10 07:29 lws-esp32-factory.map
drwxr-xr-x 2 frode frode 4,0K May 10 07:29 main

The flash process says: 

Connecting....
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 11744 bytes to 7510...
Wrote 11744 bytes (7510 compressed) at 0x00001000 in 0.1 seconds (effective 845.7 kbit/s)...
Hash of data verified.
Compressed 732976 bytes to 456520...
Wrote 732976 bytes (456520 compressed) at 0x00010000 in 8.6 seconds (effective 685.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 110...
Wrote 3072 bytes (110 compressed) at 0x00008000 in 0.0 seconds (effective 9129.3 kbit/s)...
Hash of data verified.
Leaving...
Hard resetting...

My make version is 4.1, but note that I got everything a few minutes before it stopped responding. Did not notice any errors during make, and was able to connect to the AP and see page where I could upload the SSL files. I did some simple stability tests by sitting on F5, and everything worked perfectly. Then I, mistankenly, uploaded from SSL files, and shortly after that I ended up where I am now, where /factory.html no longer responds.

I have also tried now to delete the entire build directory, and have done a new 'make all', but still not able to access. When I connect or disconnect my computer to the AP I see that the ESP responds to those events:

I (2737) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:6
I (2737) wifi: station: e0:94:67:f6:fc:ac join, AID=1, g, 20
I (355547) wifi: station: e0:94:67:f6:fc:ac leave, AID = 1
I (355547) wifi: n:6 0, o:6 0, ap:6 1, sta:0 0, prof:

lws-team, 2017-05-10:
This is a quite specific problem.... I dunno what the cause is yet but the reason is not ambiguous

> -rw-r--r-- 1 frode frode 716K May 10 07:29 lws-esp32-factory.bin

the app should be around 900KB.  The build itself is like this, ~700KB but then the build process concatenates on top the ROMFS, and a manifest, bringing it to 900KB.  So if you are only flashing ~700KB, there has been no romfs added, and it has nothing to serve, so there is no mystery about the symptom end of it.

Are you literally building with `make all flash monitor` or something else?

Scalpel78, 2017-05-10:
Yes, I'm literally doing 

    make all flash monitor -j5

and have also done 

    make clean

and 

    make erase_flash

Any tips as to where I could check to see why the .bin isn't as big as it should? Is there a step somewhere that I can run manually to see what happens when appending the romfs part?

lws-team, 2017-05-10:
How about lose the -j5?

lws-team, 2017-05-10:
The concatenation bit is done by 'pack.img' target

https://github.com/warmcat/libwebsockets/blob/master/scripts/esp32.mk#L20

Scalpel78, 2017-05-10:
Ah! Dropping -j5 fixed it. Now it creates a .bin file of 881K, and I'm able to open /factory.html.

    -rw-r--r-- 1 frode frode 881K May 10 20:43 lws-esp32-factory.bin

My guess is that -j5 causes the parts to be built in wrong order.

closed 2017-05-10 by Scalpel78

lws-team, 2017-05-11:
The latest -factory should work fine with -j n... `make all flash monitor` or `make flash monitor` should both be fine now too.