warmcat/lws-esp32-factory #8: simultaneous_ssl_restriction problem

johsely1234, 2017-06-08:
Hi I tried to change the maximum of the ssl connections from 3 to 4 with this change in the code (I tried it with the factory app):

lws_esp32_set_creation_defaults(&info);
info.simultaneous_ssl_restriction = 4;

It still only works to connect 3 Clients. This would not be a problem for me. But this change also causes another problem. If I disconnect all Clients, the wsi cleanup gets called, but I can't connect no client anymore. If I don't change the simultaneous_ssl_restriction this works without problems.

WHen all Clients (3) are connected:
`4: LWS internal statistics dump ----->
4: LWSSTATS_C_CONNECTIONS:                            3
4: LWSSTATS_C_API_CLOSE:                              0
4: LWSSTATS_C_API_READ:                               3
4: LWSSTATS_C_API_LWS_WRITE:                          3
4: LWSSTATS_C_API_WRITE:                              3
4: LWSSTATS_C_WRITE_PARTIALS:                         0
4: LWSSTATS_C_WRITEABLE_CB_REQ:                       0
4: LWSSTATS_C_WRITEABLE_CB_EFF_REQ:                   0
4: LWSSTATS_C_WRITEABLE_CB:                           0
4: LWSSTATS_C_SSL_CONNECTIONS_FAILED:                 0
4: LWSSTATS_C_SSL_CONNECTIONS_ACCEPTED:               3
4: LWSSTATS_C_SSL_CONNS_HAD_RX:                       3
4: LWSSTATS_C_TIMEOUTS:                               0
4: LWSSTATS_C_SERVICE_ENTRY:                       2178
4: LWSSTATS_B_READ:                                1581
4: LWSSTATS_B_WRITE:                                387
4: LWSSTATS_B_PARTIALS_ACCEPTED_PARTS:                0
4: LWSSTATS_MS_SSL_CONNECTIONS_ACCEPTED_DELAY:     5571ms
4:   Avg accept delay:                             1857ms
4: LWSSTATS_MS_SSL_RX_DELAY:                        134ms
4:   Avg accept-rx delay:                            44ms
4: LWSSTATS_MS_WRITABLE_DELAY:                        0us
4: LWSSTATS_MS_WORST_WRITABLE_DELAY:                  0us
4: Simultaneous SSL restriction:                      3/4/0
4: Live wsi:                                          4
4:   Listen port 443 actual POLLIN:   1
4: PT 1
4:   AH in use / max:                  0 / 3
4:   AH wait list count / actual:      0 / 0
4: 
1: SSL_new failed: 99 (errno 0)
1: lws_adopt_descriptor_vhost: fail ssl negotiation
`

All Disconnected:
`4: LWSSTATS_C_CONNECTIONS:                            4
4: LWSSTATS_C_API_CLOSE:                              4
4: LWSSTATS_C_API_READ:                               7
4: LWSSTATS_C_API_LWS_WRITE:                          6
4: LWSSTATS_C_API_WRITE:                              6
4: LWSSTATS_C_WRITE_PARTIALS:                         0
4: LWSSTATS_C_WRITEABLE_CB_REQ:                       3
4: LWSSTATS_C_WRITEABLE_CB_EFF_REQ:                   3
4: LWSSTATS_C_WRITEABLE_CB:                           0
4: LWSSTATS_C_SSL_CONNECTIONS_FAILED:                 0
4: LWSSTATS_C_SSL_CONNECTIONS_ACCEPTED:               3
4: LWSSTATS_C_SSL_CONNS_HAD_RX:                       3
4: LWSSTATS_C_TIMEOUTS:                               0
4: LWSSTATS_C_SERVICE_ENTRY:                      28868
4: LWSSTATS_B_READ:                                1633
4: LWSSTATS_B_WRITE:                                387
4: LWSSTATS_B_PARTIALS_ACCEPTED_PARTS:                0
4: LWSSTATS_MS_SSL_CONNECTIONS_ACCEPTED_DELAY:     5571ms
4:   Avg accept delay:                             1857ms
4: LWSSTATS_MS_SSL_RX_DELAY:                        134ms
4:   Avg accept-rx delay:                            44ms
4: LWSSTATS_MS_WRITABLE_DELAY:                        0us
4: LWSSTATS_MS_WORST_WRITABLE_DELAY:                  0us
4: Simultaneous SSL restriction:                      0/4/0
4: Live wsi:                                          1
4:   Listen port 443 actual POLLIN:   1
4: PT 1
4:   AH in use / max:                  0 / 3
4:   AH wait list count / actual:      0 / 0
`

Best Regards

lws-team, 2017-06-08:
I think you find you're just out of memory.

That's why I introduced the limitation, to make multiple connections queue sequentially.

For the lws test server, it opens 4 ws connections and the limit must be 4.  But for a more typical situation with less ws connections 3 should be enough.

You can turn up mbedtls and / or openssl emulation debug level in sdkconfig to get more info.

johsely1234, 2017-06-12:
Yes it is a memory problem on my side.

I used just the lws-factory-app for testing multiple connections.

The debug infos did not really help me for this, but I let them turned on for now.

One problem that I found on lws-side is probably a small memory leak. I put a heap output in the main loop:
```
while (!lws_service(context, 10)) {
         taskYIELD();
	printf("free heap: %d\n", xPortGetFreeHeapSize());
}
```

and connected and disconnected one client sequentially for multiple times. The outputs I got (memory after disconnect):

> 168828
> 166944
> 166332
> 166232
> 166132
> 166032

As it seems there are 100 Bytes which are not freed (after the 3rd disconnect). Even not if I let the program run longer.

lws-team, 2017-06-12:
It's a bit difficult to use this information to point the finger.  After lws starts in factory, there will be various allocations (larger than 100 bytes) for context, vhost and per-vhost protocol that won't get returned, so I assume your starting point is after this.

But generally per-connection stuff is definitely handled well in lws, since we check it via valgrind on Linux, where it is completely clean.

If it's 100 bytes per connection, or otherwise grows, somebody should take care about it (dunno how though, there are no tools afaik).  If it's 100 bytes, always one time, well...

johsely1234, 2017-06-12:
Ok at some point it stabilizes, so its ok!

lws-team, 2017-06-12:
+/- 100 bytes is not worth my time to try to track it down.  LWS itself is known to not have this problem.

100 bytes per connection, or 100 bytes per second ^^ would be bad for me too.

But 100 bytes one time, is something for contributors to send fixes for... right?

closed 2017-06-12 by lws-team

johsely1234, 2017-06-12:
Yes it is only a few times the until the 3rd or 4th connection attempt, then it is stable on the same level of memory usage. So it is no problem.

Yeah, but I have to get other things done as well ;-)

Thanks for your great support anyway!