warmcat/lws-esp32-factory #10: genromFS issue

khoogheem, 2017-07-09:
looks like when I try to do a make flash monitor genromfs is complaining due to trying to use a -c option which is not available.

~~~
esptool.py v2.0
stat: illegal option -- c
usage: stat [-FlLnqrsx] [-f format] [-t timefmt] [file ...]
stat: illegal option -- c
usage: stat [-FlLnqrsx] [-f format] [-t timefmt] [file ...]
             Original length: 0x000000 (       0)
/bin/bash: % 256 : syntax error: operand expected (error token is "% 256 ")
/bin/bash: (  / 256 ) % 256 : syntax error: operand expected (error token is "/ 256 ) % 256 ")
/bin/bash: (  / 65536 ) % 256 : syntax error: operand expected (error token is "/ 65536 ) % 256 ")
/bin/bash: (  / 16777216 ) % 256 : syntax error: operand expected (error token is "/ 16777216 ) % 256 ")
stat: illegal option -- c
usage: stat [-FlLnqrsx] [-f format] [-t timefmt] [file ...]
stat: illegal option -- c
usage: stat [-FlLnqrsx] [-f format] [-t timefmt] [file ...]
/bin/bash: % 256 : syntax error: operand expected (error token is "% 256 ")
/bin/bash: (  / 256 ) % 256 : syntax error: operand expected (error token is "/ 256 ) % 256 ")
/bin/bash: (  / 65536 ) % 256 : syntax error: operand expected (error token is "/ 65536 ) % 256 ")
/bin/bash: (  / 16777216 ) % 256 : syntax error: operand expected (error token is "/ 16777216 ) % 256 ")
stat: illegal option -- c
usage: stat [-FlLnqrsx] [-f format] [-t timefmt] [file ...]
    After ROMFS + Build info: 0x000000 (       0)
~~~

lws-team, 2017-07-09:
Not `genromfs`, but `stat` is complaining.

Normally on Linux, gnu `stat` -c lets you define the format

https://linux.die.net/man/1/stat

This is on a Mac?  You should be able to install gnu stat (gstat) and lws will know to use it instead of stat automatically.

khoogheem, 2017-07-09:
Yes.. It does look like it was stat issue on OSX.

maybe a note in the Readme for that section for OSX should install gstat 
~~~
brew install coreutils
~~~

Thanks -Kevin

lws-team, 2017-07-10:
I added a note in the build prerequisites section of the README.

closed 2017-07-10 by lws-team