warmcat/lws-esp32-factory #13: JTAG Debugging does not work

johsely1234, 2017-08-24:
Hello,

I got an issue debugging the lws-template app with JTAG.
I used the description from esp-idf:
https://esp-idf.readthedocs.io/en/latest/api-guides/jtag-debugging/index.html

Did you try to debug this application with JTAG?

I used this commit:
> commit d4a1be4f38f790df47d3bb5eb4273922931fad06
> Author: Andy Green <andy@warmcat.com>
> Date:   Sun Aug 13 07:47:05 2017 +0800
> 
>     esp32: document gpio may conflict with jtag

I used the esp-idf v2.1-rc1, because I did not update already.

```
Info : Target halted. PRO_CPU: PC=0x00000000             APP_CPU: PC=0x00000000 (active)
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : esp32: Failed writing 4 bytes at address 0x3FF5F064, data - a1, 3a, d8, 50, a1, 3a, d8, 50
embedded:startup.tcl:21: Error: 
in procedure 'init' 
in procedure 'ocd_bouncer' 
in procedure 'flash' 
in procedure 'ocd_bouncer' 
in procedure 'esp_core_halt' 
in procedure 'esp32' called at file "share/openocd/scripts/target/esp32.cfg", line 46
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21

Info : esp32: Debug controller was reset (pwrstat=0xFF, after clear 0xFF).
Info : esp32: Core was reset (pwrstat=0xFF, after clear 0xFF).
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: esp32_fetch_all_regs (line 163): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: esp32_fetch_all_regs (line 190): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Error: Exception reading pc!

Info : Target halted. PRO_CPU: PC=0x00000000             APP_CPU: PC=0x00000000 (active)
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: xtensa_write_memory (line 696): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : esp32: Failed writing 4 bytes at address 0x3FF5F064, data - a1, 3a, d8, 50, a1, 3a, d8, 50
embedded:startup.tcl:21: Error: 
in procedure 'esp_core_halt' 
in procedure 'esp32' called at file "share/openocd/scripts/target/esp32.cfg", line 46
in procedure 'ocd_bouncer' 
at file "embedded:startup.tcl", line 21

esp32: target state: halted
Info : accepting 'gdb' connection on tcp/3333
Info : Use core1 of target 'esp32'
Error: cpu0: xtensa_read_memory (line 588): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: xtensa_read_memory (line 588): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: xtensa_read_memory (line 588): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : cpu0: Failed reading 7416 bytes at address 0x40090000
Error: no working area available, can't alloc space for stub code!
Error: Failed to load stub (-308)!
Error: Algorithm run faied (-308)!
Info : Auto-detected flash size 0 KB
Error: Failed to probe flash, size 0 KB
Info : Set current thread to 0x00000000, old= 0x00000000
Error: cpu0: xtensa_read_memory (line 588): DSR (FFFFFFFF) indicates target still busy!
Error: cpu0: xtensa_read_memory (line 588): DSR (FFFFFFFF) indicates DIR instruction generated an exception!
Error: cpu0: xtensa_read_memory (line 588): DSR (FFFFFFFF) indicates DIR instruction generated an overrun!
Warn : cpu0: Failed reading 4 bytes at address 0x3FFB3178
Error: Could not read FreeRTOS thread count from target
```

lws-team, 2017-08-24:
> Did you try to debug this application with JTAG?

No.  I don't use JTAG or GDB even, since that has never worked on my box (complains about timeouts).  I'm happy debugging by logs.

What did you do about JTAG IO GPIO14, which I use as GPIO by default?  Notice that the bootloader uses GPIO14.

johsely1234, 2017-08-24:
In the file components/bootloader/Kconfig.projbuild
I set the default to pin 16.

```
config BOOTLOADER_FACTORY_BUTTON_GPIO
	int "GPIO for Force Factory Boot"
	default 16
```

It worked nice then to generally connect openocd with jtag.

closed 2017-08-24 by lws-team