warmcat/lws-esp32-factory #15: esp32-rtf doesn't work

nodi-andy, 2017-10-16:
i call https://192.168.local.ip/esp32-rtf or i click the button "Reset to configuration mode" but can not get the initial version to upload new FW or use OTA.

Is there a known bug?
Andy

lws-team, 2017-10-16:
Are you using https://github.com/warmcat/lws-esp32-factory/commit/6bbf538e233022cfe11174c692e1f9924011b06a ?

Which browser are you using?

nodi-andy, 2017-10-16:
no, not so fresh. i have this one:
commit d4969b877502fb7b905aa537525c4b29ba9e8bc8
Date:   Mon Sep 25 17:10:41 2017 +0800
with current chrome (Version 61.0.3163.100)

is it insecure to reset using sockets?
If i could find the command in source, i would call using any other way but couldn't figure it out

nodi-andy, 2017-10-19:
I can't make an update because of bug #17

nodi-andy, 2017-10-20:
with these settings
d4969b8  and https://github.com/warmcat/libwebsockets/tree/5ff9db9c6b520e2b16dad3fc6f6c84e20d13b1e4 
works everything fine!

lws-team, 2017-11-06:
Latest -factory, pressing the reset button works fine for me, restarts to whatever is in OTA.

Latest -test-server ota, pressing the reset to configuration button resets to factory... I think no problem...

For security, you don't have to include that plugin.  Just don't add it to the list of protocols by commenting this in main/main.c

```
        LWS_PLUGIN_PROTOCOL_ESPLWS_RTF, /* helper protocol to allow reset to factory */
```

closed 2017-11-06 by lws-team