warmcat/lws-esp32-factory #17: Ota upload can not be completed

nodi-andy, 2017-10-19:
i have regeneratable bug with ESP32.

If i use current code with latest libwebsocket than the upload interrupts about 60% up to 90% (randomly) and cpu crashes

but i use d4969b877502fb7b905aa537525c4b29ba9e8bc8 and https://github.com/warmcat/libwebsockets/tree/5ff9db9c6b520e2b16dad3fc6f6c84e20d13b1e4 than works fine.

It is very difficult to set this combination of versions because git doesnt allows it but i can copy the files manually under /components

if any other tester, use this line to upload without browser "curl --form "ota=@/path/to/esp/build/lws-esp32-test-server-demos.bin" --insecure https://192.168.local.ip/otaform"

lws-team, 2017-10-19:
You have some more info than "cpu crashes", like a backtrace?

nodi-andy, 2017-10-20:
yes. The specific error is "4: lws_spa_destroy: destroy spa 0x3ffd66d0<\r><\n>
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.<\r><\n>"

and here is the core dump:

4: writing 0x2c5e49... 0x2c6e48<\r><\n>
4: LWS_H2_PPS_UPDATE_WINDOW: sid 1: add 65536<\r><\n>
4: LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 65536<\r><\n>
4: writing 0x2c6e48... 0x2c7e47<\r><\n>
4:  heap :113648 (+776)<\r><\n>
4: ah excessive hold: wsi 0x3ffdfda4<\r><\n>
  peer address: 192.168.178.21<\r><\n>
  ah rxpos 0, rxlen 0, pos 317<\r><\n>
4:    post  = /otaform<\r><\n>
4:    origin: = https://192.168.178.51<\r><\n>
4:    accept: = */*<\r><\n>
4:    accept-encoding: = gzip, deflate, br<\r><\n>
4:    accept-language: = de,en;q=0.8,tr;q=0.6<\r><\n>
4:    content-length: = 2249798<\r><\n>
4:    content-type: = multipart/form-data; boundary=----WebKitFormBoundaryteDppABNkHLZf8fR<\r><\n>
4:    referer: = https://192.168.178.51/<\r><\n>
4:    :authority = 192.168.178.51<\r><\n>
4:    :method = POST<\r><\n>
4:    :path = /otaform<\r><\n>
4:    :scheme = https<\r><\n>
4:    user-agent: = Mozilla/5.0 (Windows NT 6.1; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.100 Safari/537.36<\r><\n>
4: lws_spa_destroy: destroy spa 0x3ffd66d0<\r><\n>
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.<\r><\n>
Register dump:<\r><\n>
PC      : 0x4011d66c  PS      : 0x00060f30  A0      : 0x801282e1  A1      : 0x3ffc2710  <\r><\n>
A2      : 0x99400099  A3      : 0x993ffd99  A4      : 0x00000000  A5      : 0x3ffdfda4  <\r><\n>
A6      : 0x00000009  A7      : 0x3ffd4478  A8      : 0x8011d722  A9      : 0x3ffc26d0  <\r><\n>
A10     : 0x40120cd0  A11     : 0xffffffff  A12     : 0x00000005  A13     : 0x00008f29  <\r><\n>
A14     : 0x3ffdffa4  A15     : 0x3ffdfda4  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  <\r><\n>
EXCVADDR: 0x99400157  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff5  <\r><\n>
<\r><\n>
Backtrace: 0x4011d66c:0x3ffc2710 0x401282de:0x3ffc2870 0x40128379:0x3ffc2ae0 0x4011dc0d:0x3ffc2b00 0x40100cb0:0x3ffc2b20 0x400d0e68:0x3ffc2b60<\r><\n>
<\r><\n>
Entering gdb stub now.<\r><\n>
$T0b#e6

i have to also mention about randomness, that sometimes there is not a total crash but upload starts from beginning but ends altough with a crash

lws-team, 2017-11-05:
Please retry with current -factory... notice it's built against a specific esp-idf commit mentioned in the README.

lws-team, 2017-11-06:
I can complete uploads just fine with current stuff.  If you still have a problem, let me know.

closed 2017-11-06 by lws-team

nodi-andy, 2017-11-07:
I cant still upload. May be it is about my network connection but here is my script, if you find any error and may be you can use the script for documentation

"bash..."
export IDF_PATH=/home/andy/esp/esp-idf
printenv IDF_PATH
export PATH=$PATH:/home/andy/esp32/xtensa-esp32-elf/bin
printenv PATH
git clone https://github.com/espressif/esp-idf
cd esp-idf
git reset --hard 0c50b65a34cd6b3954f7435193411a88adb49cb0
cd ..
git clone https://github.com/warmcat/lws-esp32-factory
cd lws-esp32-factory/
git submodule init
git submodule update --recursive
make menuconfig
make
cd ..
git clone https://github.com/warmcat/lws-esp32-test-server-demos
cd lws-esp32-test-server-demos
git submodule init
git submodule update --recursive
make menuconfig
make

lws-team, 2017-11-07:
I dunno if it's the cause of the different behaviour, but esp-idf has submodules too.  After forcing the checkout with git reset --hard, you must run something like git submodule update --recursive there too to align its submodules to the older checkout.