warmcat/lws-esp32-factory #19: Compilation error

speedy2day, 2017-11-17:
Hey there,

I tried to build as suggested in README.mk but it fails on `main.o` with the following error:
```
In file included from /home/username/esp-lws/lws-esp32-factory/main/./main.c:21:0:
/home/username/esp-lws/lws-esp32-factory/components/libwebsockets/lib/libwebsockets.h:36:24: fatal error: lws_config.h: No such file or directory
```

Could you please give me a hint, what to do? Cheers!

lws-team, 2017-11-17:
Are you building with literally `make flash monitor`?

speedy2day, 2017-11-17:
Yes. I do exactly the steps as suggested in the README.md. I am on a Xubuntu 16.10 machine. It seems like the include dirs are somehow mixed up, as it includes libwebsocket.h from .../lib/ but searches for lws_config.h when there is none in the same path. lws_config.h is found only in .../build/components/libwebsockets (if I remember right)

lws-team, 2017-11-17:
Ugh... sorry it's my fault I forgot to git add a file in lws.

Please update -factory and do the git submodule update thing.

speedy2day, 2017-11-20:
Perfect, now it compiles. The ESP boots up and generates the certificates.

Unfortunately I got stuck now with the point "Using the Factory Config"...
I find a WiFi-network, SSID: "ESP_03EB89", where I can connect to. None of config-lws-SERIAL btw.
But trying to open `http://192.168.4.1/factory.html` will give me a "site can't be reached" in chrome as well as in firefox. `https://192.168.4.1/` is the same.
Any ideas? Cheers!

Here is the log from the ESP, if it may help:
```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0020,len:4
load:0x3fff0024,len:3432
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:10964
entry 0x40078c40
4: gapss from LWS_GAPSS_INITIAL to LWS_GAPSS_SCAN
I (126) wifi: wifi firmware version: 59e5fab
I (126) wifi: config NVS flash: disabled
I (126) wifi: config nano formating: disabled
I (136) wifi: Init dynamic tx buffer num: 20
I (136) wifi: Init data frame dynamic rx buffer num: 20
I (146) wifi: Init management frame dynamic rx buffer num: 20
I (146) wifi: wifi driver task: 3ffc61bc, prio:23, stack:4096
I (156) wifi: Init static rx buffer num: 20
I (156) wifi: Init dynamic rx buffer num: 20
I (156) wifi: wifi power manager task: 0x3ffceccc prio: 21 stack: 2560
W (166) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
I (336) wifi: mode : sta (24:0a:c4:03:eb:88) + softAP (24:0a:c4:03:eb:89)
4: SYSTEM_EVENT_STA_DISCONNECTED
4: gapss from LWS_GAPSS_SCAN to LWS_GAPSS_SCAN
4:  mem: platform fd map:   120 bytes
4: ROMFS length 165KiB
{
 "schema": "lws1",
 "model": "lws",
 "builder": "mymachine",
 "app": "lws-esp32-factory",
 "user": "myname",
 "git": "3868f48-dirty",
 "date": "Mo 20. Nov 09:05:10 CET 2017",
 "unixtime": "1511165110",
 "file": "lws-esp32-factory-1511165110.bin",
 "factory": "y"
}
4: Creating Vhost 'ap' port 443, 4 protocols, IPv6 off
4:  Using SSL mode
4: Ignoring missing ap-cert.pem or ap-key.pem
4: no client cert required
4:  HTTP2 / ALPN enabled
4: created client ssl context for ap
4: lws_esp32_selfsigned: ap
4: lws_esp32_selfsigned: creating selfsigned initial certs
4: lws_esp32_selfsigned: generating 2048-bit RSA keypair... this may take a minute or so...
4: SYSTEM_EVENT_SCAN_DONE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
4: lws_esp32_selfsigned: keys done
4: lws_plat_write_cert: wrote ap-cert.pem
4: lws_plat_write_cert: wrote ap-key.pem
4: lws_esp32_selfsigned: cert creation complete
4: alloc_file: nvs: read ap-cert.pem, 1208 bytes
parsing as der
4: alloc_file: nvs: read ap-key.pem, 1675 bytes
1: lws_protocol_init: protocol lws-acme-client failed init
4:  heap :180136 (-13820)
4:  heap :182160 (+2024)
4:  heap :181920 (-240)
4:  heap :181680 (-240)
4: SYSTEM_EVENT_SCAN_DONE
4: scan_finished: count 10
4:  heap :182560 (+880)
4:  heap :182576 (+16)
...
```

lws-team, 2017-11-21:
I think there's no problem any more... the trace is normal.

The thing that is missing is there is no sign of any client connecting to the device as an AP.  It appears at the minute as something like ESP_ABCDEF.  You must do this otherwise there is no path to 192.168.4.1.

speedy2day, 2017-11-22:
Ok, thank you! There where clients connecting in the trace after the ...
But I think I didn't quite understood the whole point of the -factory, anyway. I guess I have to read the docs more carefully.

closed 2017-11-22 by speedy2day