warmcat/lws-esp32-factory #22: mbedTLS handshake fails: no usable ciphersuites

mile4712, 2017-12-13:
# System
I have a project using lws as server, which is built on the lws-esp32-factory.

My framework version is 2b92119840c41156451f75effe7a328273ce3e7f from Dec, 7 as this is the latest version which lws-esp32-factory was tested with.

# Problem
I have problems trying to connect from my Desktop PC with the TLS Handshake after updating from idf v2.1.
In the esp log i get the following lines (from lws): 
` ssl_pm_handshake: mbedtls_ssl_handshake() returned -0x6980 `

I enabled all key exchange methods in menuconfig and played a bit around with different hw-acceleration configurations (as this seemed to cause problems in the past) but I could not connect anyway.

### Client
To test if the key exchange works I tried to connect to the esp via firefox, chrome (this worked with v2.1 -> keys were exchanged)  and my own Application. 
Firefox Errorcode: `SSL_ERROR_NO_CIPHER_OVERLAP`

## Reproducibility
I cloned the lws-esp32-factory and exchanged the (main-) application with mine, so i don't know if this is easily reproducible. 

## Conclusion
I assume that the problem is somewhere in my config or code, because nobody else seems to have this problem. 

I noticed the info ``[1970/01/01 00:00:04:8891] NOTICE: set SN failed`` in  the log output and tried to find the reason but I couldn't. It comes from lws_self_signed but I didn't change anything there so I assumed it could be a too small buffer or something like that.

## Additional Information
I also examined the handshake failure with wireshark but I didn't see anything usefull there.
(only Client Hello -> Handshake Failure)
### Complete Output from  lws_esp32_init
```
[1970/01/01 00:00:02:1668] NOTICE: SYSTEM_EVENT_STA_DISCONNECTED
[1970/01/01 00:00:02:1676] NOTICE: gapss from LWS_GAPSS_INITIAL to LWS_GAPSS_SCAN
[1970/01/01 00:00:02:1899] NOTICE:  mem: platform fd map:   120 bytes
[1970/01/01 00:00:02:1912] NOTICE: ROMFS length 178KiB
{
 "schema": "lws1",
 "model": "lws",
 "builder": "esp32user",
 "app": "lws_server",
 "user": "root",
 "git": "8af9ffd-dirty",
 "date": "Die Dez 12 14:27:56 CET 2017",
 "unixtime": "1513085276",
 "file": "lws_server-1513085276.bin",
 "factory": "y"
}
[1970/01/01 00:00:02:2214] NOTICE: Creating Vhost 'ap' port 443, 1 protocols, IPv6 off
[1970/01/01 00:00:02:2299] NOTICE:  Using SSL mode
[1970/01/01 00:00:02:2355] NOTICE: no client cert required
[1970/01/01 00:00:02:2415] NOTICE:  HTTP2 / ALPN enabled
[1970/01/01 00:00:02:2476] NOTICE: created client ssl context for ap
[1970/01/01 00:00:02:2571] NOTICE: lws_esp32_selfsigned: ap
[1970/01/01 00:00:02:2611] NOTICE: lws_esp32_selfsigned: creating selfsigned initial certs
[1970/01/01 00:00:02:2707] NOTICE: lws_esp32_selfsigned: generating 2048-bit RSA keypair... this may take a minute or so...
[1970/01/01 00:00:04:8880] NOTICE: lws_esp32_selfsigned: keys done
[1970/01/01 00:00:04:8891] NOTICE: set SN failed
[1970/01/01 00:00:04:8900] NOTICE: lws_tls_server_certs_load: no paths
```
### My menuconfig settings
[image]
I already tried enabling everything in the following option but as it didn't work I set it back to the defaults.
[image]

All Key exchange Methods and Elliptic Curve ciphers are enabled.

### Errorcode in mbedTLS

[image]

lws-team, 2017-12-13:
Hm

```
[1970/01/01 00:00:04:8900] NOTICE: lws_tls_server_certs_load: no paths
```

basically it never loaded the generated cert.

It loads then using `info->ssl_cert_filepath` and `info->ssl_private_key_filepath` during the call to `lws_esp32_init`.

In -factory it has like this

```
        info.ssl_cert_filepath = "ap-cert.pem";
        info.ssl_private_key_filepath = "ap-key.pem";
        info.simultaneous_ssl_restriction = 3;

        nvs_flash_init();
        lws_esp32_wlan_config();

        ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL));

        lws_esp32_wlan_start_ap();
        /* this configures the LED timer channel 0 and starts the fading cb */
        context = lws_esp32_init(&info, &vh);
```

I guess you started with an older factory before the last changes to always TLS... you need to adapt your code to follow that part of what is in -factory.

mile4712, 2017-12-13:
Thank you very much, it works now!

closed 2017-12-13 by mile4712