warmcat/lws-esp32-factory #24: lws-esp32-factory does not compile - API to mdns have changed upstream

FredrikFornstad, 2018-01-28:
I am new to lws, but I think it looks really interesting. However I fail to compile it (msys32 on Windows) with the following message:
`-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys32/home/fredrik/esp/lws-esp32-factory/build/libwebsockets
[  2%] Building C object CMakeFiles/websockets.dir/lib/misc/base64-decode.c.o
In file included from C:/msys32/home/fredrik/esp/lws-esp32-factory/components/libwebsockets/lib/private-libwebsockets.h:262:0,
                 from C:/msys32/home/fredrik/esp/lws-esp32-factory/components/libwebsockets/lib/misc/base64-decode.c:43:
C:/msys32/home/fredrik/esp/lws-esp32-factory/components/libwebsockets/lib/libwebsockets.h:576:2:
error: unknown type name 'mdns_server_t'
  mdns_server_t *mdns;
  ^`

See here for more info: [https://github.com/espressif/esp-idf/issues/1545](url)

I guess I can try to use an older version of esp-idf but have not done that yet.

FredrikFornstad, 2018-01-28:
Tested now with an old esf-idf Environment (from Dec 2017 as specified in the LWs readme). It then compiles fine, but "make monitor" crashes after a while (after configuring the first AP-slot) due to an Windows 10 problem(?) with make monitor which has been changed in a later revision of esp-idf... 
See: https://github.com/espressif/esp-idf/issues/1136 

So an lws update for esp32 to handle the new mdns API would really be nice.

lws-team, 2018-01-29:
> So an lws update for esp32 to handle the new mdns API would really be nice.

Thanks for the heads-up, I updated lws to the new apis and then both -factory and -test-server-demos with current lws.

closed 2018-01-29 by lws-team