warmcat/lws-esp32-factory #27: Watchdog not reset in time

FredrikFornstad, 2018-02-02:
Not sure if this issue belongs to lws-esp32-factory or the lws-esp32-test-demo. Sorry if I post it in the wrong place.

When connecting to esp32 running the test-demo with a second PC (first PC is Win10-Edge browser, second pc is Win7 with IE10) it takes a long time to load the page and the watchdog triggers frequently during the loading, requiring a refresh of the browser. See log:

?[35;1m[2018/02/02 17:05:58:0060] NOTICE:  heap :130420 (-164)
?[35;1m[2018/02/02 17:05:59:0060] NOTICE:  heap :130260 (-160)
?[35;1m[2018/02/02 17:05:59:7702] NOTICE: SYSTEM_EVENT__SCAN_DONE
?[35;1m[2018/02/02 17:05:59:7713] NOTICE: scan_finished: count 6
?[35;1m[2018/02/02 17:05:59:7799] NOOTICE: lws_esp32_get_image_info: bad image header magic
?[35;1m[2018/02/02 17:06:00:0062] NOTICE:  heap :121736 (-8524)
?[35;1m[2018/02/02 17:06:00::3426] NOTICE: _realloc: size 552: new server wsi (free heap 120460)
?[35;1m[[2018/02/02 17:06:00:3462] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/02 17:06:00:3469] NOTICE: mbedtls_handshake: ssl ret -0 state 1
?[35;1m[2018/02/02 17:06:00:3542] NOTICE: mbedtls_handshake: ssl ret -6900 state 1
?[35;1m[2018/02/02 17:06:00:3614] NOTICE: _realloc: size 996: ah struct (free heap 81912)
?[35;1m[2018/02/02 117:06:00:3707] NOTICE: _realloc: size 4096: ah data (free heap 77572)
?[35;1m[2018/02/02 17:06:00:3783] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/02 17:06:00:3846] NOTICE: mbedtls_handdshake: ssl ret -0 state 2
?[35;1m[2018/02/02 17:06:00:3939] NOTICE: mbedtls_handshake: ssl ret -0 state 3
?[35;1m[2018/02/02 17:06:00:4015] NOTICE: mbedtls_handshake: ssl ret  -0 state 4
?[35;1m[20118/02/02 17:06:00:8163] NOTICE: mbedtls_handshake: ssl ret -0 state 5
?[35;1m[[2018/02/02 17:06:00:8172] NOTICE: mbedtls_handshake: ssl ret -0 state 6
?[35;1m[2018/02/02 17:06:00:8253] NOTICCE: mbedtls_handshake: ssl ret -0 state 7
?[35;1m[2018/02/02 17:06:00:8286] NOTICE: mbedtls_handshake: ssl ret -0 state 8
?[35;1m[2018/02/02 17:06:00:8361] NOTICE: mbedtls_handshake:  ssl ret -6900 state 8
?[35;1m[2018/02/02 17:06:00:9651] NOTICE: ssl_pm_handshake
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
TTasks currently running:
CPU 0: main
CPU 1: IDLE
?[35;1m[2018/02/02 17:06:09:7684] NOTICE: SYSTEM_EVENT_SCAN_DONE
?[35;1m[2018/02//02 17:06:09:7697] NOTICE: scan_finished: count 6
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
TTasks currently running:
CPU 0: main
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE ((CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
?[35;1m[2018/02/02 17:06:19:7683] NOTICE: SYSTEM_EVENT_SCAN_DONE
?[35;1m[2018//02/02 17:06:19:7695] NOTICE: scan_finished: count 5
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Taskss currently running:
CPU 0: main
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Taskss currently running:
CPU 0: main
CPU 1: IDLE
?[35;1m[2018/02/02 17:06:29:7695] NOTICE: SYSTEM_EVENT_SCAN_DONE
?[35;1m[20018/02/02 17:06:29:7707] NOTICE: scan_finished: count 6
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently runnning:
CPU 0: main
CPU 1: IDLE
Task watchdog got triggered. The following tasks did not reset the watchdog in time:

lws-team, 2018-02-03:
Please confirm this still exists on the current -test-server-demos

FredrikFornstad, 2018-02-19:
I have not been able to trigger the watchdog with two broswers connected.

However, during startup of esp32 and only one browser connected this procedure seem to be possible to reproduce every time::
1. One Edge-browser connected
2. Restarted esp32 by issuing a "make monitor" command
3. About 10 seconds after esp32 got an IP and the setup of the connection to the browser was ongoing I got a watchdog trigger:

?[35;1m[2018/02/19 17:50:46:6397] NOTICE: ssl_pm_handshakke
?[35;1m[2018/02/19 17:50:46:6422] NOTICE: mbedtls_handshake: ssl ret -0 state 1
?[35;1m[2018/02/19 17:50:46:6532] NOTICE: mbedtls_handshake: ssl ret -0 state 2
?[35;1m[2018/02/19 17:50:46:6620] NOTICE: mbedtls_handshake: ssl ret -0 state 3
?[35;1m[2018/02/19 17:50:46:6703] NOTICE: mbedtls_handshake: ssl ret -0 state 4
Task watchdog got triggered. The followingg tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: wifi

CPU 1: IDLE
?[35;1m[2018/02/19 17:50:47:7509] NOTICE: mbedtls_handshake: ssl ret -0 state 5
?[35;1m[2018/02/19 17:50:47:7569] NOTICE: mbedtls_handshake: ssl ret -0 state 6

I guess the startup is is an extra stressful moment for the esp32, so maybe not so strange it is happening here. Not sure if it is possible to add an extra yield-statement or two during the setup-phase. If not I guess this can be left without action as it seems Everything continues to startup the Connection anyway.

lws-team, 2018-02-19:
Yes... this is something more reasonable than whatever the original thing was.  It is calling into mbedtls and sitting there doing the crypto for the best part of a second.  The watchdog is right we blocked everything, but it's not because something failed, it's the normal operation and just something cosmetic.

FredrikFornstad, 2018-02-21:
Before I close this issue: I connected two browsers (one Edge and the other Firefox on Windows10) to esp32 running lws-test-demo (yes I know, that issue should not be listed here....) and I noticed this:

With both browsers on the "dumb increment counter" page, when I press "Ctrl-F5" in the Edge browser I can see one watchdog warning (the counting goes quite slow on FireFox as expected).
However, if I reload the page in the Firefox browser, there will NEVER be any watchdog warning. The counting slowdown in the Edge browser is similar as I could see in the FireFox browser in the first case.

lws-team, 2018-02-21:
Dunno... this watchdog is just there to tell us the core didn't return to the RTOS event loop for some time.  For some things we do like mbedtls handshake computation or flash erase, it's true it's possible we just sit there for over a second.  It's not an error and it doesn't create problems.

Tls negotiates some parameters on each side, it's not an error if different browsers negotiated different crypto either.  So I don't see what to do about that except maybe increase the watchdog timeout.

closed 2018-02-22 by FredrikFornstad