warmcat/lws-esp32-factory #28: Upload file does not seem to work

FredrikFornstad, 2018-02-02:
In the readme is says that Before "make flash_ota" can be used, one need to upload/flash the ota-application via the web-interface (or auto-update).

However for me the web-upload does NOT work, but I have no problem to run the ota-application after doing "make flash_ota".

Below is the log after trying to upload the test-demo via the web-interface. It takes about a minute, but after the web-page is served again to the client, the slot is still empty. I also tried  to load Another version of the factory-app and that did not work either. Below is part of the log. not sure if it helps (it seems  to me that it starts to write at the same position several times Before giving up, or?):

?[35;1m[2018/02/02 20:18:17:00666] NOTICE:  heap :126792 (+320)
?[35;1m[2018/02/02 20:18:23:3005] NOTICE: _realloc: size 552: new server wsi (free heap 125488)
?[35;1m[2018/02/02 20:18:23:3040] NOTICE: ssl_pm_handsshake
?[35;1m[2018/02/02 20:18:23:3047] NOTICE: mbedtls_handshake: ssl ret -0 state 1
?[35;1m[2018/02/02 20:18:23:3130] NOTICE: mbedtls_handshake: ssl ret -0 state 2
?[35;1m[2018/02/02 20:18:23:3225] NOTICE: mbedtls_handshake: ssl ret -0 state 3
?[35;1m[20018/02/02 20:18:23:3302] NOTICE: mbedtls_handshake: ssl ret -0 state 4
?[35;1m[2018/02/02 20:18:24:3530] NOTICE: mbedtls_hanndshake: ssl ret -0 state 5
?[35;1m[2018/02/02  20:18:24:3549] NOTICE: mbedtls_handshake: ssl reet -0 state 6
?[35;1m[2018/02/02 20:18:24:3616]  NOTICE: mbedtls_handshake: ssl ret -0 state  7
?[35;1m[2018/02/02 20:18:24:3659] NOTICE: mbedtls__handshake: ssl ret -0 state 8
?[35;1m[2018/02/02 20:18:24:3735] NOTICE: mbedtls_handshake: ssl ret -6900 state 8
?[35;1m[2018/02/02 20:18:24:3810] NOTICE: _realloc: size 996: ah struct (free heap  81992)
?[35;1m[2018/02/02 20:18:24:3886] NOTICE: _realloc: size 4096: ah data (free heap 77892)
?[35;1m[2018/02/02 20:18:24:3974] NOTICE:  heap :77888 (-48904)
?[35;1m[2018/02/02 20:18:24:4032] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/02 20:18:25:0963] NOTICE: mbedtls_handshake: ssl ret -0 state 9
?[35;1m[2018/02/02 20:18:25:0971] NOTICE: mbedtls_handshake: ssl ret -0 state 10
?[35;1m[2018/02/02 20:18:25:1013] NOTICE: mbedtls_handshake: ssl ret -0 state 11
?[35;1m[2018/02/02 20:18:25:1106] NOTICE: mbedtls_handshake: ssl ret --0 state 12
?[35;1m[2018/02/02 20:18:25:1176] NOTICE: mbedtls_handshake: ssl ret -0 state 13
?[35;1m[2018/02/02 20:18:25:1269] NOTICE: mbedtls_handshake: ssl ret -0 state 14
?[35;1m[2018/02/02 20:18:25:1304] NOTICE: mbedtls_handshake: ssl ret -0 state 15
?[35;1m[2018/02/02 20:18:25:1385] NOTICE: mbedtls_handshake: ssl ret -0 state 16
?[35;1m[2018/02/02 20:18:25:1454] NOTICE: _realloc: size 208: h2n (free heap 80340)
?[35;1m[2018/02/02 20:18:25:1529] NOTICE: _reealloc: size 780: dynamic table entries (free heap 79560)
?[35;1m[2018/02/02 20:18:25:1650] NOTICE:  heap :79556 (+1668)
?[35;1m[2018/02/02 20:18:25:1735] NOTICE: _realloc: size 512: h2 rx scratch (free heap 83904)
?[35;1m[2018/02/02 20:18:25:1792] NOTIICE: _realloc: size 48: pps (free heap 84220)
?[35;1m[2018/02/02 20:18:25:1847] NOTICE: _realloc: size 48: pps (free heap 82648)
?[35;1m[2018/02/02 20:18:25:2013] NOTICE: _realloc: size 552: new server wsi (free heap 78976)
?[35;1m[2018/022/02 20:18:25:2021] NOTICE: _realloc: size 996: ah struct (free heap 77976)
?[35;1m[2018/02/02 20:18:25:2085] NOTICE: _realloc: size 4096: ah data (free heap 73876)
?[35;1m[2018/02/02 20:18:25:2174] NOTICE: _realloc: size 9: hpack dyn (free heap 733863)
?[35;1m[2018/02/02 20:18:25:2248] NOTICE: _realloc: size 13: hpack dyn (free heap 73843)
?[35;1m[2018/02/02 20:18:25:2328] NOTICE: _realloc: size 4: hpack dyn (free heap 73836)
?[35;1m[2018/02/02 20:18:25:2411] NOTICE: _realloc: size 21: hpack dyn (free heap 738119)
?[35;1m[2018/02/02 20:18:25:2490] NOTICE: _realloc: size 22: hpack dyn (free heap 73794)
?[35;1m[2018/02/02 20:18:25:2571] NOTICE: _realloc: size 15: hpack dyn (free heap 73777)
?[35;1m[2018/02/02 20:18:25:2656] NOTICE: _realloc: size 130: hpack dyn (free heap 73646)
?[35;1m[2018/02/02 20:18:25:2735] NOOTICE: _realloc: size 71: hpack dyn (free heap 73573)
?[35;1m[2018/02/02 20:18:25:2815] NOTICE: _realloc: size 18: hpack dyn (free heap 73550)
?[35;1m[2018/02/02 20:18:25:2897] NOTICE: _realloc: size 8: hpack dyn (free heap 73540)
?[35;1m[2018/02/02 20:18:25:2977] NOTICE: _realloc: size 9: hpack dyn (free heap 73527)
?[35;1m[2018/02/02 20:18:25:3072] NOTICE: _realloc: size 596: user space (free heap 72928)
?[35;1m[2018/02/02 20:18:25:3139] NOTICE: lws_http_action: 0x3ffd7c6c: LWSS_HTTP_BODY state set
?[35;1m[2018/02/02 20:18:25:3639] NOTICE: _realloc: size 44: spa (free heap 74888)
?[35;1m[2018/02/02 20:18:25:3646] NOTIICE: _realloc: size 4096: spa (free heap 70792)
?[35;1m[2018/02/02 20:18:25:3688] NOTICE: _realloc: size 4: spa params (free heap 70784)
?[35;1m[2018/02/02 20:18:25:3769] NOTICE: _realloc: size 552: stateful urldecode (free heap 70232)
?[35;1m[2018/02/02 20:18:25:3859] NOTICE: boundary '
------------------------------7e211e18905f4'
?[35;1m[2018/02/02 20:18:25:3949] NOTICE: _realloc: size 4: spa param len (free heap 70224)
?[35;1m[2018/02/02 20:18:25:4035] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[2018/02/02 20:18:25:4134] NOTICE: Directing OTA to part type 0/16 start 0x110000
?[35;1m[2018/02/02 20:18:26:7294] NOTICE: writing 0x1110000...
?[35;1m[2018/02/02 20:18:26:7875] NOTICE:  heap :70072 (-9484)
?[35;1m[2018/02/02 20:18:26:8641] NOTICE: SYSTEM_EVENT_SCAN_DONE
?[35;1m[2018/02/02 20:18:26:8654] NOTICE: scan_finished: count 2
?[35;1m[2018/02/02 20:18:27:0104] NOTICE:  heap :68744 (-1328)
?[35;1m[2018/02/02 20:18:28:0062] NOTICE:  heap :76384 (+7640)
?[35;1m[2018/02/02 20:18:35:0061] NOTICE:  heap :76228 (-1566)
?[35;1m[2018/02/02 20:18:36::0061] NOTICE: lws_spa_destroy: destroy spa 0x3fffdd4d8
?[35;1m[2018/02/02 20:18:36:0115] NOTICEE:  heap :125960 (+49732)
[... 101 lines cut ...]
?[35;1m[2018/02/02 20:18:50:9454] NOTICE: _realloc: size 4: hpack dyn (free heap 76984)
?[35;1m[2018/02/02 20:18:50:9537] NOTICE:: _realloc: size 21: hpack dyn (free heap 76963)
?[35;1m[2018/02/02 20:18:50:9617] NOTICE: _realloc: size 22: hpack dyn (free heap 76942)
?[35;1m[2018/02/02 20:18:50:9698] NOTICE: _realloc: size 15: hpack dyn (free heap 76925)
?[35;1m[20018/02/02 20:18:50:9782] NOTICE: _realloc: size 130: hpack dyn (free heap 76794)
?[35;1m[2018/02/02 20:18:50:9862] NOTICE: _realloc: size 71: hpack dyn (free heap 76721)
?[35;1m[2018/02/02 20:18:50:9941] NOTICE: _realloc: size 18: hpack dyn (free heap 76698)
?[35;1m[2018/02/02 20:18:51:0026] NOTICE: _realloc: size 8: hpack dyn (free heap 76688)
?[35;1m[2018/02/02 20:18:51:0103] NOTICE: _realloc: size 9: hpack dyn (free heap 76675)
?[35;1m[2018/02/02 20:18:511:0191] NOTICE:  heap :76672 (-724)
?[35;1m[2018/02/02 20:18:51:0265] NOTICE: _realloc: size 596: user space (free heap 76076)
?[35;1m[2018/02/02 20:18:51:0324] NOTICE: lws_http_action: 0x3ffd7c6c: LWSS_HTTP_BODY state set
?[35;1m[2018/02/02 20:18:51:08433] NOTICE: _realloc: size 44: spa (free heap 74512)
?[35;1m[2018/02/02 20:18:51:0850] NOTICE: _realloc: size 4096: spa (free heap 70416)
?[35;1m[2018/02/02 20:18:51:0893] NOTICE: _realloc: size 4: spa params (free heap 70408)
?[35;1m[2018/02/02 20:18:51:0973] NOTICE: _realloc: size 5552: stateful urldecode (free heap 69856)
?[35;1m[2018/02/02 20:18:51:1064] NOTICE: boundary '
-----------------------------7e211e18905f4'
?[35;1m[2018/02/02 20:18:51:1153] NOTICE: _realloc: size 4: spa param len (freee heap 69848)
?[35;1m[2018/02/02 20:18:51:1239] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[20188/02/02 20:18:51:1340] NOTICE: Directing OTA to part type 0/16 start 0x110000
?[35;1m[2018/02/02 20:18:52::4136] NOTICE: writing 0x110000...
?[35;1m[2018/022/02 20:18:52:4743] NOTICE:  heap :69604 (-70688)
?[35;1m[2018/02/02 20:18:53:0061] NOTICE:  heap :75844 (+62440)
?[35;1m[2018/02/02 20:18:55:0060] NOOTICE:  heap :75680 (-164)

lws-team, 2018-02-03:
Please retry with current -factory... you will need to do `git submodule update --init --recursive` after the pull for -factory.

FredrikFornstad, 2018-02-03:
Impressive turnaround time from your side!
Partial progress, but it does not quite work yet with my setup (Windows 10, Edge browser). These were my steps:
For both lws-esp32-factory and lws-esp32-test-demo
  git fetch origin
  git reset --hard origin/master
  git submodule update --init --recursive
  make menuconfig --> change to my serial COM7-port

make erase_flash
cd to lws-esp32-factory
make flash monitor
When up and running, surf to the ESP-xxxxx AP with my Android phone, select my home router and password. Then enter Group and Role under settings.

With my Win10 PC, surf to esp32 new IP on my router LAN
Upload lws-esp32-test-demo.bin
I can see that the upload starts, but it does not seem to finish as it should. After a few tries it give up. See log below:

?[35;1m[2018/02/03 11:23:18:0099] NOTICE:  heap :145048 (-152)
?[35;1m[2018/02/03 11:23:19:0098] NOTICE:  heap :1448888 (-160)
?[35;1m[2018/02/03 11:23:19:8452] NOTICE: SYSTEM_EVENT_SCAN_DONE
?[35;1m[20188/02/03 11:23:19:8463] NOTICE: scan_finished: count 5
?[35;1m[2018/02/03 11:23:19:8546] NOTICE: lws_esp32_get_image_info: bad image header magic
?[35;1m[2018/02/03 11:23:19:8639] NOTICE: lws_esp32_get_imagee_info: bad image header magic
?[35;1m[2018/02/03 11:23:20:0104] NOTICE:  heap :137388 (-7500)
?[35;1m[2018/02/03 11:23:21:0098] NOTICE:  heap :1452004 (+7816)
?[35;1m[2018/02/03 11:23:21:6064] NOTICE: _realloc: size 552: new server wsi (free heap 143812)
?[35;1m[2018/02/03 11:223:21:6112] NOTICE: lws_gate_accepts: on = 0
?[35;1m[2018/02/03 11:23:21:6119] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/03 11:23:21:6165] NOTICE: mbedtls_handshake: ssl ret -0 state 1
?[35;1m[2018/02/03 11:23:21:6252] NOTICE: mbedtls_handshake: ssl ret -0 state 2
?[35;1m[2018/02/03 11:23:21:6353] NOTICE: mbedtls_handshake: ssl ret -0 state  3
?[35;1m[2018/02/03 11:23:21:6413] NOTICE: mbedtls_handshake: ssl ret -0 state 4
?[35;1m[2018/02/03 11:23:22:8248] NOTICE: mbbedtls_handshake: ssl ret -0 state 5
?[35;1m[2018/02/03 11:23:22:8256] NOTICE: mbedtls_handshake: ssl ret -0 state 6
?[35;1m[2018/02/03 11:23:222:8341] NOTICE: mbedtls_handshake: ssl ret -0 state 7
?[35;1m[2018/02/03 11:23:22:8384] NOTICE: mbedtls_handshake: ssl ret -0 state 8
?[35;1m[2018/02/03 11:23:22:8445] NOTICE: mbedtls_handshake: ssl ret -6900 state 8
?[35;1m[2018/02/03 11:23:22:8516] NOTICE: _realloc: size 9966: ah struct (free heap 130596)
?[35;1m[2018/02/03 11:23:22:8597] NOTICE: _realloc: size 4096: ah data (free heap 126496)
?[35;1m[2018/02/03 11:23:22:8687] NOTICE:  heap :126256 (-18948)
?[35;1m[2018/02/03 11:23:22:8745] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/03 11:23:23:7500] NOTICE: mbedtls_handshake: ssl ret -0 state 99
?[35;1m[2018/02/03 11:23:23:7507] NOTICE: mbedtls__handshake: ssl ret -0 state 10
?[35;1m[2018/02/03 11:23:23:7550] NOTICE: mbedtls_handshake: ssl ret -0 state 11
?[35;1m[2018/02/03 11:23:23:7650] NOTICE: mbedtls_handshakke: ssl ret -0 state 12
?[35;1m[2018/02/03 11:23:23:7739] NOTICE: mbedtls_handshake: ssl ret -0 state 13
?[35;1m[2018/02/03 11:23:23:7829] NOTICE: mbedtls_handshake: ssl ret -0 state 14
?[35;1m[2018/02/03 11:23:23:7840] NOTICE: mbedtls_handshake: ssl ret -0 state 15
?[35;1m[2018/02/03 11:23:23:7933] NOTICE: mbedtls_handshake: ssl ret -0 state 16
?[35;1m[2018/02/03 11:23:23:8015] NOTICE: _realloc: size 208: h2n (free heap 127156)
?[35;1m[2018/02/03 11:23:23:8067] NOTICE: _realloc: size 780: dynamiic table entries (free heap 126372)
?[35;1m[2018/02/03 11:23:23:8163] NOTICE:  heap :126368 (+112)
?[35;1m[2018/02/03 11:23:23:8225] NOTICE: _realloc: size 512: h2 rx scratch (free heap 130956)
?[35;1m[2018/02/03 11:23:23:8321] NOTICE: _realloc: size 48: pps (free heap 131076)
?[35;1m[2018/02/03 11:23:23:8386] NOTICE: _realloc: size 48: pps (free heap 131028)
?[35;1m[2018/02/03 11:23:23:8580] NOTICE: _realloc: size 552: new server wsi (free heap 127744)
?[35;1m[2018/02/03 11:223:23:8588] NOTICE: _realloc: size 996: ah struct (free heap 128300)
?[35;1m[2018/02/03 11:23:23:8653] NOTICE: _realloc: size 4096: ah data (free heap 125764)
?[35;1m[2018/02/03 11:23:23:8741] NOTICE: _realloc: size 9: hpack dyn (free heap 125751)
?[35;1m[2018/02/03 11:23:23:8814] NOTICE: _realloc: size 13: hpack dyn (free heap 125731)
?[35;1m[2018/02/03 11:23:23:8897] NOTICE: _realloc: size 4: hpack dyyn (free heap 125720)
?[35;1m[2018/02/03 11:23:23:8985] NOTICE: _realloc: size 21: hpack dyn (free heap 125699)
?[35;1m[2018/02/03 11:23:23:9059] NOTICE: _realloc: size 22: hpack dyn (free heap 125674)
?[35;1m[2018/022/03 11:23:23:9141] NOTICE: _realloc: size 15: hpack dyn (free heap 125653)
?[35;1m[2018/02/03 11:23:23:9226] NOTICE: _realloc: size 130: hpack dyn (free heap 125518)
?[35;1m[2018/02/03 11:23:23:9309] NOTICE: _realloc: size 71: hpack dyn (free heap 125441)
?[35;1m[2018/02/03 11:23:23:9388] NOTICE: _realloc: size 18: hpack dyn (free heap 125418)
?[35;1m[2018/02/03 11:23:23:9470] NOTICE: _realloc: size 8: hpack dyn (free heap 125404)
?[35;1m[2018/02/03 11:23:23:9551] NOTICE: _realloc: size 9: hpack dyn (free heap 125395)
?[35;1m[2018/02/03 11:23:23:9666] NNOTICE: _realloc: size 596: user space (free heap 124792)
[... 600 lines cut ...]
?[35;1m[2018/02/03 11:28:08:0244] NOTICE: _realloc: size 18: hpack dyn (free heap 121802)
?[35;1m[2018/02/03 11:28:08:0326] NOTICE: _realloc: size 17: hpack dyn (free heap 1217883)
?[35;1m[2018/02/03 11:28:08:0415] NOTICE:  heap :121776 (-8180)
?[35;1m[2018/02/03 11:28:08:0474] NOTICE: esp32_lws_fops_open: /ap/Gudea-Regular.ttf
[... cut ...]
?[35;1m[2018/02/03 11:28:08:9451] NOTICE: callback_esplws_scan: ESTABLISHED
?[35;1m[20118/02/03 11:28:09:0096] NOTICE:  heap :117712 (-4064)
?[35;1m[2018/02/03 11:28:09:9080] NOTICE: SYSTEM_EVENT_SCAN_DONE

?[35;1m[2018/02/03 11:28:09:9092] NOTICE: scan_finished: count 5
?[35;1m[2018/02/03 11:28:10:0098] NOTICE:  heap :122508 (+4796)
?[35;1m[2018/02/03 11:28:11:0098] NOTICEE:  heap :132712 (+10204)
?[35;1m[2018/02/03 11:28:118:0099] NOTICE:  heap :132552 (-160)
?[35;1m[2018/02/03 11:28:19:0097] NOTICE:  hheap :132392 (-160)

lws-team, 2018-02-03:
> make menuconfig --> change to my serial COM7-port

At least on Linux, you don't need to touch the config.  You can override ESPPORT in your shell env

```
$ export ESPPORT=/dev/ttyUSB0
```

Does this change anything?  Apply to components/libwebsockets

```
diff --git a/lib/http2/http2.c b/lib/http2/http2.c
index 520c01b25..4f341449b 100644
--- a/lib/http2/http2.c
+++ b/lib/http2/http2.c
@@ -813,7 +813,7 @@ lws_h2_parse_frame_header(struct lws *wsi)
                h2n->swsi->h2.peer_tx_cr_est -= h2n->length;
                lwsl_debug("   peer_tx_cr_est %d\n",
                           h2n->swsi->h2.peer_tx_cr_est);
-               if (h2n->swsi->h2.peer_tx_cr_est < h2n->length + 265536) {
+               if (h2n->swsi->h2.peer_tx_cr_est < (int32_t)h2n->length + 265536) {
                        h2n->swsi->h2.peer_tx_cr_est += h2n->length + 265536;
                        pps = lws_h2_new_pps(LWS_H2_PPS_UPDATE_WINDOW);
                        if (!pps)
```

FredrikFornstad, 2018-02-03:
I will test your modified version in a few minutes. but before that I should just add that after the above run, I tried to do a make flash_ota via mingw32-command line. That worked, but it does not boot (which is according to the readme, but not how it worked yesterday...)

Anyway, I then tried to upload via the IE-browser from the same PC (while the Edge browser was still connected). The result was slightly different with some more DEBUG messages, and triggered the watchdog once and did not repeat the write-try as the Edge browser did. However, it stopped at about the same Place so the end result was still not ok.

?[35;1m[2018/02/03 12:19::49:0086] NOTICE:  heap :146152 (+400)
?[35;1m[2018/02/03  12:19:52:1712] NOTICE: _realloc: size 552: new server wsi (free heap 144760)
?[35;1m[2018/02/03 12:19:52:1742] NOTICE: lws_gate_accepts: on = 0
?[35;1m[2018/02/03 12:19:52:1758] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/03 12:19:52:1813] NOTICE: mbedtls_handshake: ssl ret -0 state 1
?[35;1m[2018/02/03 12:19:52:1900] NOTICE: mbbedtls_handshake: ssl ret -0 state 2
?[35;1m[2018/02/03 12:19:52:2000] NOTICE: mbedtls_handshake: ssl ret -0 state 3
?[35;1m[2018/02/03 12:19:52:2068] NOTICE: mbedtls_handshake: ssl ret -0 state 4
?[35;1m[2018/02/03 12:119:53:3520] NOTICE: mbedtls_handshake: ssl ret -0 state 5
?[35;1m[2018/02/03 12:19:53:3539] NOTICE: mbedtls_handshake: ssll ret -0 state 6
?[35;1m[2018/02/03 12:19:53:3604] NOTICE: mbedtls_handshake: ssl ret -0 state 7
?[35;1m[2018/02/03 12:19:53:3648] NOTICE: mbedtls_handshake: ssl ret -0 state 8
??[35;1m[2018/02/03 12:19:53:3725] NOTICE: mbedtls_handshake: ssl ret -6900 state 8
?[35;1m[2018/02/03 12:19:53:3797] NOTICE: _realloc: size 996: ah struct (free heap 131556)
?[35;1m[2018/02/03 12:19:53:3880] NOTICE: _realloc: size 4096: ah data (free heap 127216)
?[35;1m[2018/02/03 12:19:53:3966] NOTICE:  heap :127212 (-18940)
?[35;1m[2018/02/03 12:19:53:4024] NOOTICE: ssl_pm_handshake
?[35;1m[2018/02/03 12:19:54:2374] NOTICE: mbedtls_handshake: ssl ret -0 state 9
?[35;1m[2018/002/03 12:19:54:2381] NOTICE: mbedtls_handshake: ssl ret --0 state 10
?[35;1m[2018/02/03 12:19:54:2424] NOTICE: mbedtls_handshake: ssl ret -0 state 11
?[35;1m[2018/02/03 12:19:54:2517] NOTICE: mbedtls_handshake: ssl ret -0 state 12
?[35;1m[2018/02/03 12:19:54:2604] NOTICE: mbeddtls_handshake: ssl ret -0 state 13
?[35;1m[2018/02/03 12:19:54:2730] NOTICE: mbedtls_handshake: ssl ret -0 state 14
?[35;1m[2018/02/03 12:19:54:2736] NOTICE: mbedtls_handshake: ssl ret -0 state 15
?[35;1m[2018/02/03 12:19:54:2801] NOTICE: mbedtls_handshake: ssl ret -0 state 16
?[35;1m[2018/02/03 12:19:54:2865] NOTICE: _realloc: size 208: h2n (free heap 132736)
?[35;1m[2018/02/03 12:19:54:2940] NOTICE: _realloc: size 780: dynamic table entries (freee heap 131952)
?[35;1m[2018/02/03 12:19:54:3041] NOTICE:  heap :127388 (+176)
?[35;1m[2018/02/03 12:19:54:3120] NOTICE: _realloc: size 512: h2 rx scratch (free heap 131980)
?[35;1m[2018/02/03 12:19:54:3194] NOTICE: _realloc: size 48: pps (free heap 132100)
?[35;1m[2018/02/03 12:19:54:3259] NOTICE: _realloc: size 48: pps (free heap 132052)
?[35;1m[2018/02/03 12:19:54:3464] NOTICE: _realloc: size 552: new server wsi (free heap 131892)
??[35;1m[2018/02/03 12:19:54:3471] NOTICE: _realloc: size 996: ah struct (free heap 130892)
?[35;1m[2018/02/03 12:19:54:3529] NOTICE: _realloc: size 4096: ah data (free heap 126792)
?[35;1m[2018/02/03 12:19:54:3620] NOTICE: _realloc: size 9: hpack dyn (free heap 126779)
?[35;1m[2018/02/03 12:19:54:3697] NOTICE: _realloc: size 13: hpack dyn (free heap 126759)
?[35;1m[2018/02/03 12:19:54:3781] NOTICE: _realloc: size 4: hpack dyn (free heap 126748)
??[35;1m[2018/02/03 12:19:54:3859] NOTICE: _realloc: size 71: hpack dyn (free heap 126677)
?[35;1m[2018/02/03 12:19:54:3939] NOTICE: _realloc: size 22: hpack dyn (free heap 126654)
?[35;1m[2018/02/03 12:19:54:4021] NOTICE: _realloc: size 15: hpack dyn (free heap 126633)
?[35;1m[2018/02/03 12:19:54:4105] NOTICE: _realloc: size 77: hpack dyn (free heap 126555)
?[35;1m[2018/02/03 12:19:54:4184] NOTICE: _realloc: size 8: hpack dyn (free heap 126544)
?[35;1m[20118/02/03 12:19:54:4265] NOTICE: _realloc: size 9: hpack dyn (free heap 126535)
?[35;1m[2018/02/03 12:19:54:4364] NOTICE: _realloc: size 596: user space (free heap 125932)
?[35;1m[2018/02/03 12:19:54:4431] NOTICE: lws_http_action: 0x3ffdb150: LWSS_HTTP_BODY state set
?[35;1m[2018/02/03 12:19:54:4975] NOTICE: _realloc: size 48: pps (free heap 109232)
?[35;1m[2018/02/03 12:19:54:4985] NOTICE: _realloc: size 48: pps (free heap 109184)
?[35;1m[2018/02/03 12:19:54:5028] NOTICE:  _realloc: size 44: spa (free heap 109140)
?[35;1m[2018/02/03 12:19:54:5101] NOTICE: _realloc: size 4096: spa (free heap 105044)
?[35;1m[2018/02/03 12:19:54:5181] NOTICE: _realloc: size 4: spa params (free heap 105036)
?[35;1m[2018/02/03 12:19:54:5261] NOTICE: _realloc: size 552: stateful urldecode (free heap 1044484)
?[35;1m[2018/02/03 12:19:54:5352] NOTICE: boundary '
-----------------------------7e21c735b08ac'
?[35;1m[2018/02/03 12:19:54:5442] NOTICE: _realloc: size 4: spa param len (free heap 104476)
?[35;1m[2018/02/03 12:19:54:5529] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[2018/02/03 12:19:54:5628] NOTICE: Directing OTA to part type 0/16 start 0x110000
Task watchdog got triggered. The following tasks did nnot reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
[... cut ...]
?[34;1m[2018/02/03 12:20:18:4163] DEBUG:
?[34;1m[2018/02/03 12:20:18:4179] DEBUG:
?[34;1m[22018/02/03 12:20:18:4185] DEBUG:
?[35;1m[20118/02/03 12:20:28:7977] NOTICE: lws_service_fd: closing
?[35;1m[2018/02/03 12:20:28:8081] NOTICE: lws_ggate_accepts: on = 1
?[35;1m[2018/02/03 12:20:29:0086] NNOTICE:  heap :124252 (+13676)
?[35;1m[2018/02/03 12:20:36:00888] NOTICE: lws_spa_destroy: destroy spa 0x3ffdc680
?[35;1m[2018/02/03 12:20:36:0100] NOTICE:  heapp :135228 (+10976)
?[35;1m[2018/02/03 122:20:37:0122] NOTICE:  heap :159016 (+23788)

lws-team, 2018-02-03:
There is some state set somewhere in the flash by the upload action that is not set by the make flash_ota.  So it is necessary to do the upload once so it can boot the OTA thing.  After that, you can make flash_ota OK.  If someone wants to research what that state is and how to do the same from the make flash, patches welcome :-)

FredrikFornstad, 2018-02-03:
Nope, adding (int32_t)  did not help. Same result as before (I got the watchdog triggered this time too, even though I was only using Edge-browser)  see log (shorter this time).

Funny, yesterday I must have triggered the "boot-enable the ota-app" somehow without uploading the .bin file from the web browser. But I have no idea how...

?[35;1m[2018/02/03 12:57:03:5971] NOTICE: boundary '
-----------------------------7e2e823050c'
?[35;1m[2018/02/03 12:57:03:6059] NOTICE: _reallooc: size 4: spa param len (free heap 118864)
?[35;1m[2018/02/03 12:57:03:6146] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[2018/02/03 12:57:03:6246] NOTICE: Directing OTA to part type 0/16 start 0x110000
Task watchdog got triggered. The following tasks did not reset the watchdog in time::
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
?[35;1m[2018/02/03 12:57:11:6902] NOTICE:  heap :118540 (-24332)
?[35;1m[2018/02/03 12:57:11:6945]] NOTICE: lws_esp32_get_image_info: bad image header magic
?[35;1m[2018/02/03 12:57:11:7011] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 1: add 281920
?[35;1m[2018/02/03 12:57:11:7162] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 281920
?[35;1m[20118/02/03 12:57:11:7230] NOTICE: writing 0x110000...
?[35;1m[2018/02/03 12:57:12:0790] NOTICE:  heap :118892 (+352)
?[35;1m[2018/02/03 12:57:112:3076] NOTICE: writing 0x120fef...
?[35;1m[2018/02/03 12:57:12:8694]] NOTICE: writing 0x130fdf...
?[35;1m[2018/02/03 12:57:133:4007] NOTICE: writing 0x140fcf...
?[35;1m[2018/02/03 12:577:13:9343] NOTICE: writing 0x150fbf...
?[35;1m[20018/02/03 12:57:14:0026] NOTICE:  heap :118896 (+4)
?[35;1m[2018/02/03 112:57:14:4656] NOTICE: writing 0x160faf...
?[35;1m[2018/02/03 12:57:15:0086] NOTICE:  heap :124976 (+6080)
?[34;1m[2018/02/03 12::57:41:8096] DEBUG:
?[34;1m[2018/02/03 12:57:41:8103] DEBUG:
?[35;1m[2018/02/03 12:57:45:0087] NOTICE: lws_spa_destroy: destroy spa 0x3ffd8de0
?[35;1m[2018/02/03 12:57::45:0099] NOTICE:  heap :135940 (+10964)

FredrikFornstad, 2018-02-03:
I downloaded FireFox on my Windows 10 pc. FF works better, but also fails. I notice that it does not break at the same Point each run. Sometimes it almost completes (as shown below), other times it stop a lot earlier. I also tried uploading without make monitor running, but it made no difference.

?[35;1m[2018/02/03 13:16:04:9550] NOTICE: _realloc: size 4096: spa (free heap 120076)
?[35;1m[2018/02/03 13:16:04:9629] NOTICE: _realloc: size 4: spa params (free heap 120068)
?[35;1m[2018/02/03 13:16:04:9711] NOTICE: _realloc: size 552: stateful urldecode (free heap 119516)
?[35;1m[2018/02/03 13:16:04:9802] NOTICE: boundary '
------------------------------10763155648446'
?[35;1m[2018/02/03 13:16:04:9892] NOTICE: _realloc: size 4: spa param len (free heap 119508)
?[35;1m[2018/02/03 13:16:04:9979] NOTICE: LWS_UFS_OPEN Filename lwws-esp32-test-server-demos.bin
?[35;1m[2018/02/03 13:16:05:0078] NOTICE: Directing OTA to part type 0/16 start 0x110000
?[35;1m[2018/02/03 13:16:08:2077] NOTICE:  heap :119184 (-23164)
?[35;1m[2018/02/03 13::16:08:2089] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 15: add 281536
?[35;1m[2018/02/03 13::16:08:2150] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 281536
?[35;1m[2018/02/03 13:16:08:2245] NOTICE: writing 0x110000...
?[35;1m[2018/02/03 13:16:08:3572] NOTICE: SYSTEEM_EVENT_SCAN_DONE
?[35;1m[2018/02/03 13:16:08:3585] NOOTICE: scan_finished: count 2
?[35;1m[2018/02/03 13:16:08:7876] NOTICE: _realloc: ssize 48: pps (free heap 119572)
?[35;1m[2018/02/03 13:16:08::7886] NOTICE: _realloc: size 48: pps (free heap 119524)
?[35;1m[2018/02/03 133:16:08:8173] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 15: add 281536
??[35;1m[2018/02/03 13:16:08:8228] NOTICE: Issuing LWS__H2_PPS_UPDATE_WINDOW: sid 0: add 281536
?[35;1m[2018/02/03 13:16:088:8439] NOTICE: writing 0x120fed...
?[35;1m[2018/02/03 13:16:09:0087] NOTICE:  heap :119616 (+432)
?[35;1m[[2018/02/03 13:16:09:3629] NOTICE: writing 0x130fdd...
?[35;1m[2018/02/03 13:16:09:88335] NOTICE: writing 0x140fcd...
?[35;1m[2018/02/03 13:16:10:0401] NOTICE:  heap :119620 (+4)
?[35;1m[2018/02/03 13:16:10:4033] NOTICE: writing 0x150fbd...
?[35;1m[2018/02/03 13:16:10:9274] NOTICE: writing 0x160fad...
?[35;1m[2018/02/03 13:16:10:9936] NOTICE: _realloc: size 48: ppss (free heap 119568)
?[35;1m[2018/02/03 13:16:10:9945] NOTICE: _realloc: size 488: pps (free heap 119520)
?[35;1m[2018/02/03 13:16:10:9997] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 15: add 275944
?[35;1m[2018/02/03 13:16:11:0116] NOTICE:  heap :118004 (-1616)
?[35;1m[2018/02/03 13:16:11:0134] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 275944
?[35;1m[2018/02/03 13:16:11:4753]] NOTICE: writing 0x170f9d...
?[35;1m[2018/02/03 13:16:11:9925] NOTICE: writing 0x180f8d...

?[35;1m[2018/02/03 13:16:12:0281] NOTICE:  heap :1196616 (+1612)
?[35;1m[2018/02/03 13:16:12:5074]] NOTICE: writing 0x190f7d...
?[35;1m[2018/02/03 13:16:13:0242] NOTICE: writing 0x1a0f6d...

?[35;1m[2018/02/03  13:16:13:2325] NOTICE: _realloc: size 48: pps (free heaap 119568)
?[35;1m[2018/02/03 13:16:13:2334] NOTICE: _realloc: size 48:: pps (free heap 119520)
?[35;1m[2018/02/03 13:16:13:2635] NOOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 15: add 281536
?[35;1m[2018/02/03 13:16:13:26886] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 281536
?[35;1m[2018/02/03 13:16:13:5670] NOTTICE: writing 0x1b0f5d...
?[35;1m[2018/02/03 13:166:14:0079] NOTICE:  heap :125696 (+6080)
?[35;1m[2018/02/003 13:16:14:0804] NOTICE: writing 0x1c0f4d...
?[35;1m[2018/02/03 13:16:14:5971] NOTICE: writinng 0x1d0f3d...
?[35;1m[2018/022/03 13:16:15:0163] NOTICE:  heap :119620 (-6076)
?[35;1m[2018/02/03 13:16:15:1101] NOTICE:: writing 0x1e0f2d...
?[35;1m[2018/02/03 13:16:15:4356] NOTICE: _realloc:: size 48: pps (free heap 119568)
?[35;1m[2018//02/03 13:16:15:4366] NOTICE: _realloc: size 48: pps (free heap 119520)
?[35;1m[2018/02/03 13:16:155:4419] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 15: add 281536
??[35;1m[2018/02/03 13:16:15:4539] NOTICE: Issuing LWS_H2_PPS_UPDDATE_WINDOW: sid 0: add 281536
?[35;1m[2018/02/03 13::16:15:6377] NOTICE: writing 0x1f0f1d...
?[35;1m[22018/02/03 13:16:16:1462] NOTICE: writing 0x200f0d...
?[35;1m[2018/02/03 13:16:16:6543] NOTICE: writing 0x210efd...
?[35;1m[2018/02/03 13:16:17:1968]] NOTICE: writing 0x220eed...
?[35;1m[20018/02/03 13:16:17:7059] NOTICE: _realloc: size 488: pps (free heap 119568)
?[35;1m[2018/02/03 13:16:17:70699] NOTICE: _realloc: size 48: pps (free heap 119520)
?[35;1m[2018/02/03 13:16:17:7128] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 15: add 281536
?[35;1m[2018/02/03 13:16:17:7246] NOTICE:: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 281536
?[35;1m[2018/02/03 13:16:17:7341] NOTICE:: writing 0x230edd...
?[35;1m[2018/02/03 13:16:18:0699] NOTICE:  heap :119612 (-8)
?[35;1m[2018/02/03 13:16:18:2438] NOTICE: writing 0x240ecd...
?[35;1m[2018/02/03 13:16:18:7437] NOTICE: writing 0x250ebc...

?[35;1m[2018/02/03 13:16:19:0660] NOTIICE:  heap :119616 (+4)
?[35;1m[2018/02/03 13:16:19:2540] NOTICE: writingg 0x260eac...
?[35;1m[2018/02/03 133:16:19:7639] NOTICE: writing 0x270e9c...
?[35;1m[2018/02/03 13:16:19:8671] NOTICE: _realloc: size 48: ppps (free heap 119568)
?[35;1m[2018/02/03 13:16:19:8681] NOTICE: _reallocc: size 48: pps (free heap 119516)
?[35;1m[2018/02/03 133:16:19:8732] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 15: add 281464
?[35;1m[2018/02/03 13:16:19:8852] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 2281464
[... cut ...]
?[35;1m[2018/02/03 13:16:55:00955] NOTICE: lws_spa_destroy: destroy spa 0x3ffd3aa4
?[35;1m[[2018/02/03 13:16:55:0107] NOTICE:  heap :136656 (+109644)
?[35;1m[2018/02/03 13::16:56:0144] NOTICE:  heap :160824 (+24168)
?[35;1m[2018/02/03 13:16:57:00995] NOTICE:  heap :161160 (+336)
?[35;1m[2018/02/03 13:16:58:0094]] NOTICE:  heap :161164 (+4)
?[35;1m[2018/02/03 13:17:01:3149] NOTICE: lws_service_fd: closinng
?[35;1m[2018/02/03 13:17:01:3665] NOTICE: _realloc: size 552: new server wsi (free heap 173720)
?[35;1m[2018/02//03 13:17:01:3698] NOTICE: ssl_pm_handshake

lws-team, 2018-02-03:
Hm it seems related to http/2 transmit credit still... it receives 8 x 64K before it stalls, without updating the window.  Prior to that it's updating the browser's tx window every 4 x 64K it was sent.  It's 21:30 here I'll study it further tomorrow.

FredrikFornstad, 2018-02-03:
I can confirm that everything works using a Windows 10 pc with Chrome browser (I was starting to believe it was me...). By the way, both Chrome and Firefox render all the pages better than Edge/IE that has a problem with "blocky transparent grey boxes" that goes away after a few seconds when loading new pages. With Chrome I get a much better experience overall from the lws-app.

 So right now:
Edge and IE : Upload stops early
Firefox: Upload stops generally later
Chrome: Works

FredrikFornstad, 2018-02-04:
Sorry, the latest change does not help. It still does not work with Edge or FireFox (under Win10). Firefox goes a bit further than Edge, but a lot less than yesterday.  Also, when writing goes "bad" it seems to wait a lot longer before it continues than it did yesterday.

I have not tested with chrome.

Log from try with Edge:
?[35;1m[2018/02/04 09:12:42:6733] NOTICE: _realloc: size 552: stateful urldecode (free heap 103000)
?[35;1m[2018/02/04 09:12:42:6825] NOTICE: boundary '
-----------------------------7e2e97304ac'
?[35;1m[2018/02/04 09:12:42:6912] NOTICE: _realloc: size 4: spa param len (free heap 102992)
?[35;1m[2018/022/04 09:12:42:6999] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[2018/02/04 09:12:42:7100] NOTICE: Directing OTA to part type 0/16 start 0x110000
?[35;1m[2018/02/04 09:12:44:3531] NOTICE:  heap :102924 (-23068)
?[35;1m[2018/02/04 09:112:44:3542] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 1: add 98304
?[35;1m[2018/02/04 09:12:44:3622] NOTICCE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 98304
?[35;1m[2018/02/04 09:12:44::3732] NOTICE: writing 0x110000...
?[35;1m[2018/02/04 09:12:44:9218] NOTICE: writing 0x120fef...
?[35;1m[[2018/02/04 09:12:45:1015] NOTICE:  heap :103024 (+100)
?[35;1m[2018/02/04 09:12:45:4499] NOTICE: writing 0x130fdf...
?[35;1m[2018/02//04 09:12:46:0033] NOTICE:  heap :109100 (+6076)
?[35;1m[2018/022/04 09:12:58:0032] NOTICE:  heap :109304 (+204)
?[34;1m[2018/02/04 09:12:59:9519] DEBUG:
?[34;1m[2018/02/04 09:12:59:9527] DEBUG:
?[35;1m[2018/02/04  09:13:16:0033] NOTICE: lws_spa_destroy: destroy spa 0x3ffd9424
?[35;1m[2018/02/04 09:13:16:0047] NOTICE:  hheap :120272 (+10968)
?[35;1m[2018/02/04 09:13:17:0088] NOTICE: lws_gate_accepts: on = 1

lws-team, 2018-02-04:
Hmmm is everything up to date git submodule-wise?  Have a look with git status from the toplevel dir.

With Firefox on Linux

```
[2018/02/04 00:37:21:6319] NOTICE: _realloc: size 552: new server wsi (free heap 148700)
[2018/02/04 00:37:21:6327] NOTICE: _realloc: size 996: ah struct (free heap 147700)
[2018/02/04 00:37:21:6396] NOTICE: _realloc: size 4096: ah data (free heap 140560)
[2018/02/04 00:37:21:6480] NOTICE: _realloc: size 8: hpack dyn (free heap 140744)
[2018/02/04 00:37:21:6550] NOTICE: _realloc: size 85: hpack dyn (free heap 140659)
[2018/02/04 00:37:21:6647] NOTICE: _realloc: size 596: user space (free heap 140056)
[2018/02/04 00:37:21:6718] NOTICE: lws_http_action: 0x3ffd9364: LWSS_HTTP_BODY state set
[2018/02/04 00:37:21:6884] NOTICE: _realloc: size 48: pps (free heap 137064)
[2018/02/04 00:37:21:6892] NOTICE: _realloc: size 48: pps (free heap 137016)
[2018/02/04 00:37:21:6954] NOTICE: _realloc: size 48: pps (free heap 136964)
[2018/02/04 00:37:21:7033] NOTICE: _realloc: size 44: spa (free heap 136916)
[2018/02/04 00:37:21:7107] NOTICE: _realloc: size 4096: spa (free heap 132816)
[2018/02/04 00:37:21:7186] NOTICE: _realloc: size 4: spa params (free heap 132808)
[2018/02/04 00:37:21:7267] NOTICE: _realloc: size 552: stateful urldecode (free heap 132256)
[2018/02/04 00:37:21:7359] NOTICE: boundary '
-----------------------------262915315368687401826357415'
[2018/02/04 00:37:21:7471] NOTICE: _realloc: size 4: spa param len (free heap 132248)
[2018/02/04 00:37:21:7546] NOTICE: LWS_UFS_OPEN Filename lws-esp32-serial.bin
[2018/02/04 00:37:21:7636] NOTICE: Directing OTA to part type 0/16 start 0x110000
Task watchdog got triggered. The following tasks did not reset the watchdog in time:
 - IDLE (CPU 0)
Tasks currently running:
CPU 0: main
CPU 1: IDLE
[2018/02/04 00:37:27:2316] NOTICE:  heap :131848 (-19116)
[2018/02/04 00:37:27:2328] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 68814
[2018/02/04 00:37:27:2403] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 68814
[2018/02/04 00:37:27:2925] NOTICE: _realloc: size 48: pps (free heap 118544)
[2018/02/04 00:37:27:2945] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
[2018/02/04 00:37:27:3036] NOTICE: writing 0x110000...
[2018/02/04 00:37:27:6005] NOTICE: _realloc: size 48: pps (free heap 118544)
[2018/02/04 00:37:27:6016] NOTICE: _realloc: size 48: pps (free heap 118496)
[2018/02/04 00:37:27:6066] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 97536
[2018/02/04 00:37:27:9189] NOTICE: _realloc: size 48: pps (free heap 118496)
[2018/02/04 00:37:27:9483] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
[2018/02/04 00:37:27:9735] NOTICE: writing 0x120fef...
[2018/02/04 00:37:28:0387] NOTICE:  heap :122904 (-8944)
[2018/02/04 00:37:28:6529] NOTICE: _realloc: size 48: pps (free heap 118492)
[2018/02/04 00:37:28:6537] NOTICE: _realloc: size 48: pps (free heap 118444)
[2018/02/04 00:37:28:6589] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 95616
[2018/02/04 00:37:28:7214] NOTICE: writing 0x130fdf...
[2018/02/04 00:37:28:9789] NOTICE: _realloc: size 48: pps (free heap 118440)
[2018/02/04 00:37:28:9809] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
[2018/02/04 00:37:29:1025] NOTICE:  heap :124368 (+1464)
[2018/02/04 00:37:29:3507] NOTICE: writing 0x140fcf...
[2018/02/04 00:37:29:6027] NOTICE: _realloc: size 48: pps (free heap 118440)
[2018/02/04 00:37:29:6036] NOTICE: _realloc: size 48: pps (free heap 118392)
[2018/02/04 00:37:29:6087] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 97536
[2018/02/04 00:37:29:9283] NOTICE: _realloc: size 48: pps (free heap 118384)
[2018/02/04 00:37:29:9302] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
[2018/02/04 00:37:29:9886] NOTICE: writing 0x150fbf...
[2018/02/04 00:37:30:0484] NOTICE:  heap :124132 (-236)
[2018/02/04 00:37:30:5483] NOTICE: _realloc: size 48: pps (free heap 118388)
[2018/02/04 00:37:30:5492] NOTICE: _realloc: size 48: pps (free heap 118340)
[2018/02/04 00:37:30:5554] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 97536
[2018/02/04 00:37:30:6163] NOTICE: writing 0x160faf...
[2018/02/04 00:37:30:8884] NOTICE: _realloc: size 48: pps (free heap 118332)
[2018/02/04 00:37:30:9176] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
[2018/02/04 00:37:31:0079] NOTICE:  heap :124072 (-60)
[2018/02/04 00:37:31:2541] NOTICE: writing 0x170f9f...
[2018/02/04 00:37:31:5024] NOTICE: _realloc: size 48: pps (free heap 118332)
[2018/02/04 00:37:31:5034] NOTICE: _realloc: size 48: pps (free heap 118284)
[2018/02/04 00:37:31:5356] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 97536
[2018/02/04 00:37:31:8234] NOTICE: _realloc: size 48: pps (free heap 118280)
[2018/02/04 00:37:31:8527] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
[2018/02/04 00:37:31:8790] NOTICE: writing 0x180f8f...
[2018/02/04 00:37:32:0082] NOTICE:  heap :118328 (-5744)
[2018/02/04 00:37:32:4447] NOTICE: _realloc: size 48: pps (free heap 118276)
[2018/02/04 00:37:32:4456] NOTICE: _realloc: size 48: pps (free heap 118228)
[2018/02/04 00:37:32:4779] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 97536
[2018/02/04 00:37:32:5041] NOTICE: writing 0x190f7f...
[2018/02/04 00:37:32:7633] NOTICE: _realloc: size 48: pps (free heap 118232)
[2018/02/04 00:37:32:7955] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
...
```

It times out after ~30s, that's what is expected if it stalls

```
?[35;1m[2018/02/04 09:12:45:4499] NOTICE: writing 0x130fdf...
...
?[35;1m[2018/02/04 09:13:16:0033] NOTICE: lws_spa_destroy: destroy spa 0x3ffd9424
```

lws-team, 2018-02-04:
Well, it's happy here, which makes it difficult.

Please try this on components/libwebsockets

```
diff --git a/lib/http2/http2.c b/lib/http2/http2.c
index b935637b..f7012921 100644
--- a/lib/http2/http2.c
+++ b/lib/http2/http2.c
@@ -815,7 +815,7 @@ lws_h2_parse_frame_header(struct lws *wsi)
                wsi->h2.peer_tx_cr_est -= h2n->length;
                h2n->swsi->h2.peer_tx_cr_est -= h2n->length;
 
-               lwsl_debug("   peer_tx_cr_est %d, parent %d\n",
+               lwsl_notice("   peer_tx_cr_est %d, parent %d\n",
                           h2n->swsi->h2.peer_tx_cr_est, wsi->h2.peer_tx_cr_est);
 
                if (h2n->swsi->h2.peer_tx_cr_est < (int)(2 * h2n->length) + 65536) {
```

it will dump lws' idea of how much tx window the peer has, for each frame.

FredrikFornstad, 2018-02-04:
Here is the log (running Edge-browser). I included the log from the first time I saw the notice-message.

?[35;1m[2018/02/04 10:30:32:5833] NOTICE: _realloc: size 36: hpack dyn (free heap 144016)
?[35;1m[2018/02/04 10:30:32:5923] NOTICE: _realloc: size 4696: user space (free heap 139316)
?[35;1m[2018/02/04 10:30:32:5991] NOTICE: lws_http_action: 0x3ffdfd64:  LWSS_HTTP_BODY state set
?[35;1m[2018/02/04 10:30:32:6098] NOTICE:    peer_tx_cr_est 65197, parent -338
?[35;1m[2018/02/04 10:30:32:6149] NOTICE: _realloc: size 48: pps (free heap 139676)
?[35;1m[2018/02/04 10:30:32:6228] NOTICE: _realloc: size 48: pps (free heap 139624)
?[35;1m[2018/02/04 10:30:32:6305] NOTICE: _realloc: size 48: pps (free heap 139572)
?[35;1m[2018/02/04 10:30:32:6383] NOTICE: _realloc: size 44: spa (free heap 1139524)
?[35;1m[2018/02/04 10:30:32:6456] NOTICE: _realloc: size 1024: spa (free heap 138500)
?[35;1m[2018/02/04 10:30:32:6534] NOTICE: _realloc: size 32: spa params (free heap 138464)
?[35;1m[2018/02/04 10:30:32:6617] NOTICE: _realloc: size 552: stateful urldecode (free heap 13791
22)
?[35;1m[2018/02/04 10:30:32:6708] NOTICE: boundary '
------WebKitFormBoundarygh1Rw7Wf560hfRSH'
?[35;1m[2018/02/04 10:30:32:6795] NOTICE: _realloc: size 32: spa param len (free heap 137876)
?[35;1m[2018/02/04 10:30:32:6886] NOTICE: HTTP_BODY_COMPLETION
?[35;1m[2018/02/04 10:30:32:6939] NOTICE: LWS_CALLBACK_HTTP_BODY_COMPLETION (scan)
?[35;1m[2018/02/04 10:30:32:7015] NOTICE:  Param text: (none)
?[35;1m[2018/02/04 10:30:32:7072] NOTICE:  Param pub: (none)
?[35;1m[2018/02/04 10:30:32:7129] NOTICE:  Param pri: (none)
?[35;1m[2018/02/04 10:30:32:7186] NOTICE:  Param serial: 0E99E5
?[35;1m[2018/02/04 10:30:32:7246] NOTICE:  Param opts: (none)
?[35;1m[2018/02/04 10:30:32:7307] NOTICE:  Param group: Testunnan
?[35;1m[2018/02/04 10:30:32:7365] NOTICE:  Param role: IoT
?[35;1m[2018/02/04 10:30:32:7420] NOTICE:  Param updsettings: (none)
?[35;1m[2018/02/04 10:30:32:7558] NOTICE: lws_spa_destroy: destroy spa 0x3ffc0e64
?[35;1m[2018/02/04 10:30:33:0057] NOTICE:  heap :159576 (+18064)
?[35;1m[2018/02/04 10:30:34:0056] NOTICE:  heap :159592 (+16)
?[35;1m[2018/02/04 10:30:35:0057] NOTICE:  heap :159512 (-80)
?[35;1m[2018/02/04 10:30:36:0058] NOTICE:  heap :159272 (-240)
?[35;1m[2018/02/04 10:30:37:0056] NOTICE:  heap :159036 (-236)
?[35;1m[2018/02/04 10:30:37:0805] NOTICCE: SYSTEM_EVENT_SCAN_DONE
?[35;1m[2018/02/04 10:30:37:0812] NOTICE: scan_finished: count 7
?[35;1m[2018/02/04 10:30:37:0907]  NOTICE: lws_esp32_get_image_info: bad image header magic
?[35;1m[2018/02/04 10:30:38:0057] NOTICE:  heap :159592 (+556)
?[35;1m[2018//02/04 10:30:45:0057] NOTICE:  heap :159512 (-80)
?[35;1m[2018/02/04 10:30:46:0056] NOTICE:  heap :159356 (-156))
?[35;1m[2018/02/04  10:30:47:0056] NOTICE:  heap :159196 (-160)
?[35;1m[2018/02/04 10:30:47:0808] NOTICE: SYSTEM_EVENT_SCAN_DOONE
?[35;1m[2018/02/04 10:30:47:0817] NOTICE: scan_finished: count 5
?[35;1m[2018/02/04 10:30:47:0891] NOTICE: lws_esp32_get_image_info: bad image header magic
?[35;1m[2018/02/04 10:30:47:6401] NOTICE: _realloc: size 552: new server wsi (free heap 1447720)
?[35;1m[2018/02/04 10:30:47:6434] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/04 10:30:47:6440] NOTICE: mbedtls_handshake: ssl ret -0 state 1
?[35;1m[2018/02/04 10:30:47:6529] NOTICE: mbedtls_handshake: ssl ret -0 state 2
?[35;1m[22018/02/04 10:30:47:6617] NOTICE: mbedtls_handshake: ssl ret -0 state 3
?[35;1m[2018/02/04 10:30:47:6694] NOTICE: mbedtls_handshake: ssl ret -0 state 4
?[35;1m[2018/02/04 10:30:48:8117] NOTICE: mbbedtls_handshake: ssl ret -0 state 5
?[35;1m[2018/02/04 10:30:48:8130] NOTICE: mbedtls_handshake: ssl ret -0 state 6
?[35;1m[2018/02/04 10:30:48:81889] NOTICE: mbedtls_handshake: ssl ret -0 state 7
?[35;1m[2018/02/04 10:30:48:8240] NOTICE: mbedtls_handshake: ssl ret -0 state 8
?[35;1m[2018/02/04 10:30:48:8318] NOTICE: mbedtls_handshake: ssl ret -6900 state 8
?[35;1m[2018/02/04 10:30:48:8389] NOTICE: _realloc: size 996: ah struct (free heap 144968)
?[35;1m[2018/02/04 10:30:48:8470] NOTICE: _realloc: size 4096: ah data (free heap 140868)
?[35;1m[2018/02/04 10:30:48:8558] NOTICE:  heap :140864 (-18332)
?[35;1m[20018/02/04 10:30:48:9049] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/04 10:30:48:9070] NOTICE: mbedtls_handshake: ssl ret -7280 state 8
ssl_pm_handshake: mbedtls_ssl_handshake() returned -0x7280
?[35;1m[2018/02/04  10:30:49:0058] NOTICE:  heap :159500 (+18636)
?[35;1m[2018/02/04 10:30:50:0058] NOTICE:  heap :159504 (+4)
?[35;1m[2018/02/04 10:30:52:8178] NOTICE: _realloc: size 552: new server wsi (free heap 158140)
?[35;1m[2018/02/04 10:30:52:8211] NOTICE: ssl_pm_handshake
?[35;1m[2018/02/04 10:30:52:8218] NOTICE: mbedtls_handdshake: ssl ret -0 state 1
?[35;1m[2018/02/04 10:30:52:8304] NOTICE: mbedtls_handshake: ssl ret -0 state 2
?[35;1m[2018/02/04 10:30:52:8408] NOTICE: mbedtls_handshake: ssl ret -0 state 3
?[35;1m[2018/02/04 10:30:52:8463] NOTICE: mbedtls_handshake: ssl ret -0 state 4
?[35;1m[2018/02/04 10:30:54:0019] NOTICE: mbedtls_handshake: ssl ret -0 state 5
?[35;1m[2018/02/04 10:30:54:0026] NOTICE: mbedtlss_handshake: ssl ret -0 state 6
[... 298 lines cut ...]
?[35;1m[2018/02/04 10:32:36:2230] NOTICE: mbedtls_handshake: ssl ret -0 state 16
?[35;1m[2018/02/04 10:32:36:2266] NOTICE: _realloc: size 208: h2n (free heap 131392)
?[35;1m[2018/02/04 10:32:36:2345] NOTICE: _realloc: size 780: dynamic table entries (free heap 1256644)
?[35;1m[2018/02/04 10:32:36:2433] NOTICE:  heap :125660 (+3632)
[... cut ...]
?[35;1m[2018/02/04 10:32:36:3737] NOTICE: _realloc: size 8: hpack dyn (free heap 121584)
?[35;1m[2018/02/04 10:32:36:3802] NOTICE: _realloc: size 9: hpack dyn (free heap 124687)
?[35;1m[2018/02/04 10:332:36:3899] NOTICE: _realloc: size 596: user space (free heap 124084)
?[35;1m[2018/02/04 10:32:36:3965] NOTICE: lws_http_action: 0x3ffd9694: LWSS_HTTP_BODY state set
?[35;1m[2018/02/04 10:32:36:5670] NOTICE:    peer_tx_cr_est 49151, parent -16384
?[35;1m[22018/02/04 10:32:36:5678] NOTICE: _realloc: size 48: pps (free heap 111948)

FredrikFornstad, 2018-02-04:
I also did a run with FireFox. The behaviour is different from Edge, since it prints the notice message much more frequent. Also: when running Edge, the upload-try is being repeated numerous times Before it finally give up and the http server serves the first "tab". In Firefox, it only tries once.

?[35;1m[2018/02/04 11:00:07:0082] NOTICE:  heap :143976 (-7620)
?[35;1m[20118/02/04 11:00:07:1777] NOTICE:    peer_tx_cr_est 63897, parent -16338
?[35;1m[2018/02/04 11:00:07:1808] NOTICE: _realloc: size 48: pps (free heap 137656)
?[35;1m[2018/02/04 11:00:07:1842] NOTICE:: _realloc: size 48: pps (free heap 137796)
?[35;1m[2018/02/04 11:00:07:1916] NOTICE: _realloc: size 48: pps (free heap 137744)
?[35;1m[2018/02/04 11:00:07:1996] NOTICE: _realloc: size 44: spa (free heap 137696)
?[35;1m[2018/02/04 11:00:07:2069] NOTTICE: _realloc: size 4096: spa (free heap 133600)
?[35;1m[2018/02/04 11:00:07:2148] NOTICE: _realloc: size 4: spa params (free heap 133592)
?[35;1m[2018/02/04 11:00:07:2230] NOTICE: _realloc: size 552: stateful urldecode (free heap 133040)
?[35;1m[2018/02/04 11:00:07:2320] NOTICE: boundary '
-----------------------------278891150314871'
?[35;1m[2018/02/04 11:00:07:2411] NOTICE: _realloc: size 4: spa param len (free heap 133032)
?[35;1m[2018/02/04 11:00:07:2499] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[[2018/02/04 11:00:07:2598] NOTICE: Directing OTA to part type 0/16 start 0x110000
?[35;1m[2018/02/04 11::00:08:9577] NOTICE:  heap :132960 (-11016)
?[35;1m[2018/02/04 11:00:08:9587] NOTICE:: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 68812
?[35;1m[2018/02/04 11:00:08:9666] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 68812
?[35;1m[2018/02/04 11:00:09:0049] NOTICE:  heap :123744 (-9216)
?[35;1m[2018/02/04 11:00:09:0403] NOTICE:    peer_tx_cr_est 116709, parent 51174
?[35;1m[2018/02/04 111:00:09:0410] NOTICE: _realloc: size 48: pps (free heap 119328)
?[35;1m[2018/02/04 11:00:09:0463] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
?[35;1m[2018/02/04 11:00:09:0597] NOTICE: writing 0x110000...
?[35;1m[2018/02/04 11:00:09:1916] NOTICE:    peer_tx_cr_est 101989, parent 133990
?[35;1m[2018/02/04 11:00:09:26654] NOTICE:    peer_tx_cr_est 100693, parent 132694
?[35;1m[2018/02/04 11:00:09:3387] NOTICE:    peer_tx_cr_est 84693, parent 116694
?[35;1m[2018/02/04 11:00:09:3396] NOTICE: _realloc: size 48: pps (free heap 119324)
?[35;1m[2018/02/04 11:00:09:3437] NOTICE: _realloc: size 48: pps (free heap 1192766)
?[35;1m[2018/02/04 11:00:09:3521] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 97536
?[35;1m[2018/02/04 11:00:09:4998] NOTICE:    peer_tx_cr_est 166757, parent 101222
?[35;1m[2018/02/04 11:00:09:5701] NOTICE:    peer_tx_cr_est 166348,  parent 100813
?[35;1m[2018/02/04 11:00:09:6097]] NOTICE:    peer_tx_cr_est 150348, parent 84813
?[35;1m[2018/02/04 11:00:09:6105] NOTICE: _realloc: size 48: pps (free heap 119276)
?[35;1m[2018/02/04 11:00:09:6412] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
?[35;1m[2018/02/04 11:00:09:6621] NOTICE: writing 0x120fed...
?[35;1m[2018/02/04 11:00:09::7412] NOTICE:    peer_tx_cr_est 134348, parent 166349
?[35;1m[2018/02/04 11:00:09:8312] NOTICE::    peer_tx_cr_est 134332, parent 166333
?[35;1m[2018/02/04 11:00:09:8338] NOTICE:    peer_tx_cr_est 133989, parent 165990
?[35;1m[2018/02/04 11:00:09:8830] NOTICE:    peer_tx_cr_est 117989, parent 149990
?[35;1m[2018/02/04 11:00:10:01182] NOTICE:    peer_tx_cr_est 102316, parent 134317
?[35;1m[2018/02/04 11:00:10:1089] NOTICE:  heap :119324 (-4420)
?[35;1m[[2018/02/04 11:00:10:1127] NOTICE:    peer_tx_cr_est 101221, parent 133222
?[35;1m[2018/02/04 11:00:10:1309] NOTICE:    peer_tx_cr_est 97536, parent 129537
?[35;1m[2018/02/04 11:00:10:2059] NOTICE:    peer_tx_cr_est 81536, parent 1135537
?[35;1m[2018/02/04 11:00:10:2067] NOTICE: _realloc: size 48: pps (free heap 119276)
?[35;1m[2018/02/04 11:00:10:2108] NOTICE: _realloc: size 48: pps (free heap 119228)
?[35;1m[2018/02/04 11:00:10:2193] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 25: add 97536
?[35;1m[2018/02/04 11:00:10:2578] NOTICE: writing 0x130fdd...
?[35;1m[20118/02/04 11:00:10:3677] NOTICE:    peer_tx_cr_est 165989, parent 100454
?[35;1m[2018/02/04 11:00:10:4432] NOTICE:    peer_tx_cr_est 163056, parent 97521
?[35;1m[2018/02/04 11:00:10::5187] NOTICE:    peer_tx_cr_est 147056, parent 81521
?[35;1m[20018/02/04 11:00:10:5195] NOTICE: _realloc: size 48: pps (free heap 119224)
?[35;1m[2018/02/04 11:00:10:5247] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 97536
?[35;1m[2018/02/04 11:00:10:6690] NOTICE:    peer_tx_cr_est 133221, parent 165222
?[35;1m[20018/02/04 11:00:10:7439] NOTICE:    peer_tx_cr_est 131040, parent 163041
?[35;1m[2018/02/04 11:00:10:8111] NOTICE:    peer_tx_cr_est 115040, parent 147041
?[35;1m[2018/02/04 111:00:10:8362] NOTICE: writing 0x140fcd...
?[35;1m[2018/02/04 11:00:10:9376] NOTICE:    peer_tx_cr_est 101205, parent 133206
?[35;1m[2018/02/04 11:00:11:0039] NOTICE:  heap :119272 (-52)
?[35;1m[2018/02/04 11:00:11:0156] NOTICE:    peer_tx_cr_est 100453, parent 1324454
?[35;1m[2018/02/04 11:000:11:3938] NOTICE: writing 0x150fbd...
?[35;1m[2018/02/044 11:00:11:9271] NOTICE: writing 0x160fad...
?[35;1m[2018/02/04 11:00:12:0072] NOTICE:  heap :125348 (+6076)
?[35;1m[2018/02/04 11:00:43:0084] NOTICE: lws_spa_destroy: destroy sppa 0x3ffd65dc
?[35;1m[2018/02/04 11:00:43:0097] NOTICE:  heap :136308 (+10960)
?[35;1m[2018/02/04 11:00:44:0142] NOTICE:  heap ::160780 (+24472)
[... cut ...]
?[35;1m[2018/02/04 11:00:50:2576] NOTICE: _realloc: size 996: ah struct (free heap 160180)
?[35;1m[2018/02/04 11:00:50:2657] NOTICE: _realloc: size 4096: ah data (free heap 156080)
?[35;1m[2018/02/04 11:00:50:2760] NOTICE:  heap :156076 (-4796)
?[35;1m[2018/02/04 11:00:50:2808] NOTICE: ssl_pm_handshake
?[35;1m[20188/02/04 11:00:51:0884] NOTICE: mbedtls_handshake: ssl ret -0  state 9
?[35;1m[2018/02/04 11:00:51:0892] NOTICE: mbedtls_handshake: ssl ret -0 state 10
?[35;1m[2018/02/04 11:00:51:0950] NOTICE: mbedtls_handshake: ssl ret -0 sttate 11

lws-team, 2018-02-05:
I found there exists a https://browserstack.com that lets me drive various browsers from Chrome here and render my ESP32 on a portforward.  So I could reproduce the problem by this strange way.

I added some additional tx credit when the stream is established, that allowed it to complete (although I didn't get all the logging I was expecting).  So please try once more.

FredrikFornstad, 2018-02-05:
No, it still does not work... I tested both Edge and with FireFox.
I double checked that I got your latest Changes in libwebsockets.
I also changed the "debug" message to "notice". I also added a notice where you assign the new 4*65536 so you can see that it is really doing it.

HOWEVER, in ONE of the trials with Edge, it got MUCH further in Writing to the the flash. That particular try is the 2:nd try in the below log. I have not seen this Before. Is it timing related? Or what other condition could have caused the flashing to go further this time?

?[35;1m[2018/02/05 16:12:29:0021] NOTICE:  heap :150672 (-96)
?[35;1m[2018/022/05 16:12:30:4555] NOTICE: _realloc: size 552: new server wsi (free heap 145556)
?[35;1m[2018/02/05 16:12:30:4563] NOTICE: _realloc: size 48: pps (free heap 145504)
?[35;1m[2018/02/05 16:12:30:4615] NOTICE: Set pps->u.update_window.credit to: 262144 , and h2n->swsi->h2.peer_tx_cr_est to: 327679
?[35;1m[2018/02/05 16:12:30:4736] NOTICE: _realloc: size 48: pps (free heap 145456)
?[35;1m[2018/02/05 166:12:30:4810] NOTICE: _realloc: size 996: ah struct (free heap 144456)
?[35;1m[2018/02/05 16:12:30:4892] NOTICE: _realloc: size 4096: ah data (free heap 140360)
?[35;1m[2018/02/05 16:12:30:4982] NOTICE: _realloc: size 9: hpack dyn (free heap 140347)
?[35;1m[2018/02/05 116:12:30:5056] NOTICE: _realloc: size 13: hpack dyn (free heap 140331)
?[35;1m[2018/02/05 16:12:30:5141] NOTICE: _realloc: size 4: hpack dyn (free heap 140320)
?[35;1m[2018/02/05 16:12:30:5220] NOTICE: _realloc: size 22: hpack dyn (free heap 140318)
?[35;1m[2018/02/05 16:12:30:5301] NOTICE:  _realloc: size 15: hpack dyn (free heap 140317)
?[35;1m[2018/02/05 16:12:30:5387] NOTICE: _realloc: size 130: hpack dyn (free heap 140310)
?[35;1m[2018/02/05 16:12:30:5467] NOTICE: _realloc: size 71: hpack dyn (free heap 140385)
?[35;1m[2018/02/05 16:12:30:5548] NOTICE: _realloc: size 18: hpack dyn (free heap 140362)
?[35;1m[2018/02/05 16:12:30:5629] NOTICE: _realloc: size 8: hpack dyn (free heap 140352)
?[35;1m[2018/02/05 16:12:30:57299] NOTICE: _realloc: size 9: hpack dyn (free heap 140343)
?[35;1m[2018/02/05 16:12:30:5802] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 11: add 262144
?[35;1m[2018/02/05 16:12:30:5938] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 262144
?[35;1m[2018/02/05 16:12:30:6021] NOTICE: _realloc: size  596: user space (free heap 139848)
?[35;1m[2018/02/05 16:12:30:6047] NOTICE: lws_http_action: 0x3ffdc924: LWSS_HTTP_BODY state set
?[35;1m[2018/02/05 16:12:30:6569] NOTICE:    peer_tx_cr_est 311295, parent 1556480
?[35;1m[2018/02/05 16:12:30:6579] NOTICE: _realloc: size 44: spa (free heap 123200)
?[35;1m[2018/02/05 16:12:30:6618] NOTICCE: _realloc: size 4096: spa (free heap 119100)
?[35;1m[2018/02/05 16:12:30:6696] NOTICE: _realloc: size 4: spa params (free heap 119092)
?[35;1m[2018/02/05 16:12:30:6780] NOTICE: _realloc: size 552: stateful urldecode (free heap 118544)
?[35;1m[2018/02/05 16:12:30:6868] NOTICE: boundary '
-----------------------------7e22021f103c0'
?[35;1m[2018/02/05 16:12:30:6958] NOTICE: _realloc: size 4: spa paramm len (free heap 118536)
?[35;1m[2018/02/05 16:12:30:7044] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[2018/02/05 16:12:30:7147] NOTICE: Directing OTA to part type 0/16 start 0x110000
?[35;1m[2018/02/05 16:12:32:6837] NOTICE: writing 0x110000...
?[35;1m[2018/02/05 16:12:32:7458] NOTICE:  heap :118300 (-32372)
?[35;1m[2018/02/05 16:12:33:0067]  NOTICE:  heap :118464 (+164)
?[35;1m[2018/02/05 16:12:33:2234] NOTICE: writing 0x120fef...
?[35;1m[2018/02/05 16:12:33:7363] NOTICE: writing 0x130fdf...
?[35;1m[2018/02/05 16:12:34:2620] NOTICE: writing 0x140fcf...
?[35;1m[2018/02/05 16:112:34:7801] NOTICE: writing 0x150fbf...
?[35;1m[2018/02/05 16:112:35:2922] NOTICE: writing 0x160faf...
?[35;1m[2018/02/05 16:12:35:8065] NOTICE: writing 0x170f9f...
?[35;1m[2018/02/05 16:12:36:3177] NOTICE: writing 0x180f8f...
?[35;1m[2018/02/05 16:12:366:8289] NOTICE: writing 0x190f7f...
?[35;1m[2018/02/05 16:12:37:0244] NOOTICE:  heap :118468 (+4)
?[35;1m[2018/02/05 16:12:37:3353] NOTIICE: writing 0x1a0f6f...
?[35;1m[2018/02//05 16:12:37:8428] NOTICE: writing 0x1b0f5f...
?[35;1m[2018/02/05 16:12:38:0402] NOTICE:  heap :118464 (-4)
?[35;1m[2018/02/05 16:12:38:3551] NOTICE: writing 0x1c0f4f...
?[35;1m[2018/02/05 16:12:38:8673] NOTICE: writing 0x1d0f3f...
?[35;1m[2018/02/05 16:12:39:0603] NOTICCE:  heap :118468 (+4)
?[35;1m[20118/02/05 16:12:39:3763] NOTICE: writing 0x1e0f2f...
?[35;1m[2018/02/05 16:12:39:8840] NOTICE: writing 0x1f0f1f...
?[35;1m[2018/02/05 16:12:40:0805] NOTICE:  heap :118464 (-4)
?[35;1m[2018/02/05 16:12:40:3915] NOTICE: writing 0x200f0f...
?[35;1m[2018/02/05 16:12:40:8919] NOTICE: writing 0x210eff...
?[35;1m[2018/02/05 16:12:41:3986] NOTICE: writing 0x220eef...
?[35;1m[20188/02/05 16:12:41:9055] NOTICE: writing 0x230edf...
?[35;1m[2018/02/05 16:12:42:1004] NOTICE:  heap :118468 (+4)
?[35;1m[2018/02/05 16:12:42:4156] NOTICE: writing 0x240ecf...
?[35;1m[2018/02/05 16:12:42:9192] NOTICE: writing 0x250ebf...
?[35;1m[2018/02/05 16:12:43:0089] NOTICE:  heap :124540 (+6072)
?[35;1m[2018/02/05 16:12:43:4267] NOTICE: writing 0x260eaf...
?[35;1m[2018/02/05 16:12:43:9354] NOTICE: wwriting 0x270e9f...
?[35;1m[2018/02/05 16:12:44:0031] NOTICE:  heap :118464 (-6076)
?[35;1m[2018/02/05 16:12:44:4334] NOTICE: wriiting 0x280e8f...
[... 138 lines cut ...]
?[35;1m[2018/02/05 16:14:02:2995] NOTICE: _realloc: size 596: user space (free heap 140092)
[... cut ...]
?[35;1m[2018/02/05 16:14:05:7111] NOTICE: writing 0x1130fdf...
?[35;1m[2018/02/05 16:14:06:0203] NOTICE:  heap :118732 (-4)
?[35;1m[2018/02/05 16:14:06:2322] NOTICE: writing 0x140fcf...
?[35;1m[2018/02/05 16:14:06:7513] NOTTICE: writing 0x150fbf...
?[35;1m[[2018/02/05 16:14:08:0023] NOTICE:  heap :124812 (+6080)
?[34;1m[2018/02/055 16:14:27:1557] DEBUG:
?[34;1m[2018/02/05 16:14:27:1565] DEBUG:
?[35;1m[2018/02/05 16:14:38:0023] NOTICE: lws_spa_destroy: destroy spaa 0x3ffc0eac
?[35;1m[2018/02/05 16:14:38:0037] NOTICE:  heap :135780 (+10968)

FredrikFornstad, 2018-02-05:
Off topic: Here is an example on the rendering in Edge: NOTE, it is not always it looks like this. I really think it is an Edge problem so it will be hard for you to solve it. But it can be good to understand the experience "windows people" may have once in a while when they are using the default browser...

The "Square" in the middle is most likely a remaining from the "software download tab" as it has the same placement as the text boxes containing the firmware versions.

[image]
[image]

FredrikFornstad, 2018-02-05:
Here is an example log from a FireFox try. 
I also tested with Another Windows 10 PC with the same result, FireFox and Edge does not work.
The Chrome browser is however working OK.

?[35;1m[2018/02/05 17:13:11:7590] NOTICE: Set pps->u.update_window.credit to: 262144 , and h2n->swsi->h2.peer_txx_cr_est to: 327679
?[35;1m[2018/02/05 17:13:11:7703] NOTICE: _realloc: size 48: pps (free heap 144216)
?[35;1m[2018/02/05 17:13:11:7777] NOTICE: _realloc: size 996: ah struct (free heap 143220)
??[35;1m[2018/02/05 17:13:11:7858] NOTICE: _realloc: size 4096: ah data (free heap 139124)
?[35;1m[2018/02/05 17:13:11:7952] NOTICE: _realloc: size 8: hpack dyn (free heap 139112)
?[35;1m[2018/02/05 17:13:11:8025] NOTICE: _realloc: size 72: hpack dyn (free heap 139040)
?[35;1m[2018/02/05 17:13:11:8118] NOTICE: Issuing LWS_HH2_PPS_UPDATE_WINDOW: sid 27: add 262144
?[35;1m[2018/02/05 17:13:11:8239] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 262144
?[35;1m[2018/02/05 17:13:11:8338] NOTICE: _realloc: size 596: user space (free heap 138536)
?[35;1m[22018/02/05 17:13:11:8361] NOTICE: lws_http_action: 0x3ffd8548: LWSS_HTTP_BODY state set
?[35;1m[2018/02/05 17:13:11:8551] NOTICE:    peer_tx_cr_est 326040, parent 1833369
?[35;1m[2018/02/05 17:13:11:8562] NOTICE: _realloc: size 44: spa (free heap 135884)
?[35;1m[2018/02/05 17:13:11:8623] NOTICE: _rrealloc: size 4096: spa (free heap 131784)
?[35;1m[2018/02/05 17:13:11:8700] NOTICE: _realloc: size 4: spa params (free heap 131776)
?[35;1m[2018/02/05 17:13:11:8781] NOTICE: _realloc: size 552: stateful urldecode (free heap 131224)
?[35;1m[2018/02/05 17:13:11:8872] NOTICE: boundary '
------------------------------12053109739298'
?[35;1m[2018/02/05 17:13:11:8962] NOTICE: _realloc: size 4: spa param len (free heap 131216)
?[35;1m[2018/02/05 17:13:11:9049] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[2018/02/05 17:13:11:9149] NOTICE: Directing OTA to part type 0/16 start 0x110000
?[35;1m[2018/02/05 17:13:14:0793] NOTICE:  heap :131144 (-18604))
?[35;1m[2018/02/05 17:13:14:1230] NOTICE:    peer_tx_cr_est 310040, parent 1817369
?[35;1m[2018/02/05 17:13:14:1254] NOTICE: writing 0x110000...
?[35;1m[2018/02/05 17:13:14:2453] NOTICE:    peer_txx_cr_est 295320, parent 1802649
?[35;1m[2018/02/05 17:13:14:3160] NOTICE:    peer_tx_cr_est 294024, parent 1801353
?[35;1m[2018/02/05 17:13:14:3826] NOTICE:    peer_tx_cr_est 278024, parent 1785353
?[35;1m[2018/02/05 17:13:14:5182] NOTICE:    peer_tx_cr_est 262552, parent 1769881
?[35;1m[2018/02/05 17:13:14:5876] NOTICE:    peer_tx_cr_est 262008, parent 1769337
?[35;1m[2018/02/05 17:13:14:6300] NOTICE:    peer_tx_cr_est 2246008, parent 1753337
?[35;1m[2018/02/05 17:13:14:6692] NOTICE: writing 0x120fed...
?[35;1m[2018/02/05 17:13:14:7530] NOTICE:    peer_tx_cr_est 230008, parent 1737337
?[35;1m[2018/02/05 17:13:14:8463] NOTICE:    peer_tx_cr_est 229992, parent 1737321
?[35;1m[2018/02/05 17:13:14:8487] NOTICE:    peer_tx_cr_est 229784, parent 1737113
?[35;1m[22018/02/05 17:13:14:8915] NOTICE:    peer_tx_cr_est 213784, parent 1721113
?[35;1m[2018/02/05 17:13:15:0228] NOTICE:    peer_tx_cr_est 197976, parent 1705305
?[35;1m[2018/02/05 17:13:15:1148] NOTICE:  heap :1117464 (-13680)
?[35;1m[2018/02/05 17:13:15:1185] NOTICE:    peer_tx_cr_est 197016, parent 1704345
?[35;1m[2018/02/05 17:13:15:1252] NOTICE:    peer_tx_cr_est 196607, parent 1703936
?[35;1m[2018/02/05 17:13:15:1668] NOTICE:    peer_tx_cr_est 180607, parent 1687936
?[35;1m[2018/02/05 17:13:15:2139] NOTICE: writing 0x130fdd....
?[35;1m[2018/02/05 17:13:15:2897] NOTICE:    peer_tx_cr_est 164607, parent 1671936
?[35;1m[2018/02/05 17:13:15:3864] NOTICE:    peer_tx_cr_est 164591, parent 1671920
?[35;1m[20018/02/05 17:13:15:7499] NOTICE: writing 0x140fcd...
?[35;1m[2018/02/05 17:13:16:2724] NOTIICE: writing 0x150fbd...
?[35;1m[2018/02/05 17:13:166:8064] NOTICE: writing 0x160fad...
?[35;1m[2018/02/05 17:13:17:1059] NOTICE:  heap :117456 (-8)
?[35;1m[22018/02/05 17:13:18:0013] NOTICE:  heap :123540 (+6084)
?[35;1m[2018/02/05 17:13:45:0011] NOTICE:  heap :123500 (-40)
?[35;1m[2018/02/05 17:13:47:0011] NOTICE:  heap :123496 (-4)
?[35;1m[2018/02/05 17:13:48:0012] NOTICE: lws_spa_destroyy: destroy spa 0x3ffd1f10
?[35;1m[2018/02/05 17:13:48:0026] NOTICE:  heap :134468 (+10972)

lws-team, 2018-02-06:
I moved the window management from the frame to the actual data block usages... then tested on firefox, chrome and edge via the browserstack thing (this needs me to disable the OTA parts, since I can't upload a real image to their server to upload back to my ESP32, but otherwise it completes their 1MB zip upload).  Please try once more.

FWIW Edge16 on browserstack doesn't have the strange rectangles... chrome and ffox don't get them as you found... there's nothing in the html about those kind of colours... so I also think some bug with edge or windows.  The html makes a lot of use of transparency, but nothing really requires it.  The css could change dynamically if it found it was running on edge, eg not use alpha.

FredrikFornstad, 2018-02-06:
Getting closer!

It seems that all(?) loading works on both FireFox and Edge on Win10. However, there must be something missing in the final stage, because:
1. The test-demo app does not start automatically as it does when uploading with Chrome
2. The "app-slot" is still marked as empty

Here is a log from an Edge try:
?[35;1m[2018/02/06 16:27:12:3584] NOTICE:  heap :126816 (+144)
?[35;1m[2018/02/06 16:27:12:3648] NOTICE: _realloc: size 512: h2 rx scratch (free heap 131396)
?[35;1m[2018/02/006 16:27:12:3742] NOTICE: _realloc: size 48: pps (free heap 131520)
?[35;1m[2018/02/06 16:27:12:3808] NOTICE: 0x3ffd89d4 (0x3ffd89d4): fr hdr: typ 0x4, flags 0x0, sid 0x0, len 0
xc
?[35;1m[2018/02/06 16:27:12:3904] NOTICE: _realloc: size 48: pps (free heap 131468)
?[35;1m[2018/02/06 16:27:12:3986] NOTICE: 0x3ffd89d4 (0x3ffd89d4): fr hdr: typ 0x8, flags 0x0, sid 0x0, len 0
x4
?[35;1m[2018/02/06 16:27:12:4217] NOTICE: 0x3ffd89d4 (0x0): fr hdr: typ 0x1, flags 0x4, siid 0x1, len 0x107
?[35;1m[2018/02/06 16:27:12:4227] NOTICE: _realloc: size 552: new server wsi (free heap 131312)
?[35;1m[2018/02/06 16:27:12:4298] NOTICE: _realloc: size 48: pps (free heap 131260)
?[35;1m[2018/02/06 16:27:12:4376] NOTICE: _realloc: size 48: pps (free heap 131208)
?[35;1m[2018/02/06 16:27:12:4452] NOTICE: _realloc: size 996: ah struct (free heap 130208)
?[35;1m[2018/02/06 16:27:12:4534] NOTICE: _realloc: size 4096: ah data (free heap 126112)
?[35;1m[20018/02/06 16:27:12:4624] NOTICE: _realloc: size 9: hpack dyn (free heap 126103)
?[35;1m[2018/02/06 16:27:12:4699] NOTICE: _realloc: size 13: hpack dyn (free heap 126087)
?[35;1m[2018/02/06 16:27:12:4781] NOTICE: _realloc: size 4: hpack dyn (free heap 126076)
?[35;1m[2018/02/06 16:27:12:4863] NOTICE: _realloc: size 21: hpack dyn (free heap 126055)
?[35;1m[2018/02/06 16:27:12:4962] NOTICE: _realloc: size 22: hpack dyn (free heap 126030)
?[35;1m[2018/02/06 16:277:12:5026] NOTICE: _realloc: size 15: hpack dyn (free heap 126009)
?[35;1m[2018/02/06 16:27:12:5111] NOTICE: _realloc: size 130: hpack dyn (free heap 125874)
?[35;1m[2018/02/06 16:27:12:5196] NOTICE: _realloc: size 70: hpack dyn (free heap 125798)
?[35;1m[2018/02/06 16:27:12:5273] NOTICE: _realloc: size 18: hpack dyn (free heap 125774)
?[35;1m[2018/02/06 16:27:12:5354] NOTICE: _realloc: size 8: hpack dyn (free heap 125764)
?[35;1m[2018/02/06 16:27:12:5435]  NOTICE: _realloc: size 9: hpack dyn (free heap 125751)
?[35;1m[2018/02/06 16:27:12:5528] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 1: add 262144
?[35;1m[2018/02/06 16:27:12:5658] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 262144
?[35;1m[2018/02/06 16:27:12:5744] NOTICE: _realloc: size 596: user space (free heap 125252)
?[35;1m[20018/02/06 16:27:12:5771] NOTICE: lws_http_action: 0x3ffdb528: LWSS_HTTP_BODY state set
?[35;1m[2018/02/06 16:27:12:6343] NOTICE: 0x3ffd89d4 (0x3ffdb528): fr hdr: typ 0x0, flags 0x0, sid 0x1, len 0
x4000
?[35;1m[2018/02/06 16:27:12:6354] NOTICE: _realloc: size 44: spa (free heaap 108628)
?[35;1m[2018/02/06 16:27:12:6420] NOTICE: _realloc: size 4096: spa (free heap 104532)
?[35;1m[2018/02/06 16:27:12:6498] NOTICE: _realloc: size 4: spa params (free heap 104524)
?[35;1m[2018/02/06 16:27:12:6580] NOTICE: _realloc: size 552: stateful urldecode (free heap 103972)
?[35;1m[2018/02/06 16:27:12:6671] NOTICE: boundary '

-----------------------------7e2360bd03e2'
?[35;1m[2018/02/06 16:27:12:6759] NOTICE: _realloc: size 4: spa param len (free heap 103964)
?[35;1m[2018/02/06 16:27:12:6846] NOTICE: LWS_UFS_OPEN Filename lws-esp32-test-server-demos.bin
?[35;1m[2018/02/06 16:27:12:6946] NOTICE: Directing OTA to part type 0/16 start 0x110000
?[35;1m[2018/02/06 16:27:14:1063] NOTICE: writing 0xx110000...
?[35;1m[2018/02/06 16:27:14:1687] NOTICE:  heap ::103728 (-23088)
?[35;1m[2018/02/06 16:27:14:6619] NOTICE: writing 0x120fef...
?[35;1m[2018/02/06 16:27:15:0063] NOTICE:  heap :106940 (+3212)
?[35;1m[2018/02/06 16:27:15:2038] NNOTICE: writing 0x130fdf...
?[35;1m[2018/02/06 16:27:15:3868] NOTICE: _realloc: size 48: pps (free heap 103848)
?[35;1m[2018/02/06 16:27:15:3884] NOTTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 98304
?[35;1m[2018/02/06 16:27:15:7342] NNOTICE: writing 0x140fcf...
?[35;1m[2018/02/06 16:27:15:9135] NOTICE: _reallocc: size 48: pps (free heap 103848)
?[35;1m[2018/02/06 16:27:15:9151] NOTICE: Issuing LWWS_H2_PPS_UPDATE_WINDOW: sid 1: add 98304
?[35;1m[20118/02/06 16:27:16:0542] NOTICE:  heap :103896 (-3044)
?[35;1m[20118/02/06 16:27:16:2104] NOTICE: _realloc: size 48: pps (free heap 105368)
??[35;1m[2018/02/06 16:27:16:2120] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 98304
?[35;1m[2018/02/06 16:27:16:2794] NOTICE: writing 0x150fbf...
?[35;1m[2018/02/006 16:27:16:7089] NOTICE: _realloc: size 48: pps (free heap 103852)
?[35;1m[2018/02/06 16:27:16:7105]  NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 1: add 98304
?[35;1m[2018/02/06 116:27:16:8010] NOTICE: writing 0x160faf...
?[35;1m[2018/02/06 16:27:16:9953] NOTICE: _realloc: size 48: pps (free heap 103844)
?[35;1m[2018//02/06 16:27:16:9969] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 98304
?[35;1m[2018/02/06 16:27:17:0059] NOTICE:  heap :102328 (-1568)
?[35;1m[2018/02/06 16:27:17:3162] NOTICE: writing 0x170f9f...
?[35;1m[2018/02/06 16:27:17:5129]  NOTICE: _realloc: size 48: pps (free heap 103848)
[... 267 lines cut ...]
?[35;1m[2018/02/06 16:27:42:8093] NOTICE: 0x3ffd89d4 (0x0): fr hdrr: typ 0x1, flags 0x5, sid 0x9, len 0xd8
[... cut ...]
?[35;1m[2018/02/06 16:27:42:9454] NOTICE: esp32_lws_fops_open: /ap/Gudea-Regular.ttf
?[35;1m[2018/02/06 16:27:43:0068] NOTICE:  heap :133964 (+54452)
?[35;1m[2018/02/06 16:27:44:0068] NOTICE:  heap :133968 (+4)
?[35;1m[2018/02/06 16:27:47:0068] NOTICE:  heap :133888 (-80)
?[35;1m[2018/02/06 16:27:48:0068] NOTICE:  heap :133728 (-160)
?[35;1m[2018/02/06 16:27:49:0310] NOTICE: SYSTEM_EVENT_SCAN_DONE
?[35;1m[2018/02/06 16:27:49:0321] NOTICE: scaan_finished: count 3
?[35;1m[2018/02/06 16:27:50:0068] NOTICE:  heap :133968 (+240)

FredrikFornstad, 2018-02-06:
Here is a log with the end of the loading when using Chrome (on Win10). 
1. The behaviour is different during the whole upload process with a lot of Notice messages between each write when running Chrome. --> I guess this indicates that different parts of lws-code is executed depending on browser. Do you understand why?
2. With Chrome I get LWS_UFS_FINAL_CONTENT at the end. I did not see that with Edge or FireFox.

?[35;1m[2018/02/06 17:11:07:4473] NOTICE: writing 0x330ddc...
?[35;1m[2018/02/06 17:11:07:4726] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:4849] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:5220] NOTICE: 0x3ffd71d0 (0x3ffd7b338): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0x84c
?[35;1m[20118/02/06 17:11:07:5568] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:5590] NOTICE: _realloc: size 48: pps (free heap 131412)
?[35;1m[2018/02/06 17:11:07:5652] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 13: add 71240
?[35;1m[2018/02/06 17::11:07:5875] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:6249] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:6481] NOTICCE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:6605] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:6934] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0x84c
?[35;1m[2018/02/06 17:11:07:7297] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:7316] NOTICE: _realloc: size 48: pps (frree heap 131412)
?[35;1m[2018/02/06 17:11:07:7382] NOTICE: Issuing LWS_H2_PPS_UPDATE_WINDOW: sid 0: add 71240
?[35;1m[2018/02/06 17:11:07:7603] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:7971] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:8216] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:8340] NOTICE: 0x3ffd71d0 (0x33ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:8660] NOTICE: 0x3ffd71d0 (0x33ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0x84c
?[35;1m[[2018/02/06 17:11:07:9013] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:9144] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:9492] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:9727] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:07:9852] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:08:0104] NOTICE:  heap :130560 (-900)
?[35;1m[2018/02/06 17:11:08:0185] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid  0xd, len 0x84c
?[35;1m[2018/02/06 17:11:08:0534] NOTICE: 0x3ffd71d0  (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:08:0655] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:08:0984] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:08:1003] NOTICE: writing 0x340dcc...
?[35;1m[2018/02/06 17:11:08:1266] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:08:1390] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0xb24
?[35;1m[2018/02/06 17:11:08:1702] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x0, sid 0xd, len 0x84c
?[35;1m[2018/02/06 17:11:08:2018] NOTICE: 0x3ffd71d0 (0x3ffd7b38): fr hdr: typ 0x0, flags 0x1, sid 0xd, len 0xb08
?[35;1m[2018/02/06 17:11:08:2240] NOTICE: LWS_UFS_FINAL_CONTENT
?[35;1m[2018/02/06 17:11:08:7591] NOTICE: HTTP_BODY_COMPPLETION
?[35;1m[2018/02/06 17:11:08:7601] NOTICE: LWS_CALLBACK_HTTP_BODY_COMPLETION (ota)
?[34;1m[2018/02/06 17:11:08:7689] DEBUG: LWS__CALLBACK_HTTP_WRITEABLE: sending 26
I (79377) wifi: state: run -> init (0)
I (79377) wifi: n:1 0, o:1 0, ap:1 1, sta:1  0, prof:1
I (79427) wifi: flush txq
I (79427) wifi: stop sw txq
I (79427) wiffi: lmac stop hw txq
?[35;1m[2018/02/06 17:11:09:0572] NOTICE: SYSTEM_EVENT_STA_DISCONNECTED
?[35;1m[20018/02/06 17:11:09:0597] NOTICE: gapss from LWS_GAPSS_STAT_HAPPY to LWS_GAPSS_SCAN
E (79437) wifi: esp_wifi_connect 839 wifi not start
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drvv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0020,len:4
load:0x3fff0024,len:3432
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:11724
entry 0x40078e3c
?[35;1m[2018/02/06 17:11:09:1336] NOTICE: gapss from LWS_GAPSS_INITIAL to LWS_GAPSS_SCAN

lws-team, 2018-02-07:
> ?[35;1m[2018/02/06 16:27:32:6662] NOTICE: HTTP_BODY_COMPLETION
?[35;1m[2018/02/06 16:27:32:6684] NOTICE: LWS_CALLBACK_HTTP_BODYY_COMPLETION (ota)

This indicates we received all of the post body treating it as a bunch of bytes.  We have been trying to solve the stalls at this level so far, I think that is solved now by the sound of it.

I did not actually understand yet why there are two paths through the code as you noticed.  I tried to manage things at the start of each DATA frame but while the bulk data is coming, the code that should handle the headers doesn't see enough of them somehow.  In the end just reacting to bulk data coming solved the symptom.

> ?[35;1m[2018/02/06 17:11:08:2240] NOTICE: LWS_UFS_FINAL_CONTENT

This means we understood we received the logical end of the encapsulated file in the post body... so this triggers setting the partition as bootable and finishes the flas update.

I am traveling for the next week or so, I can study it remotely when I get time but bandwidth to Taiwan isn't that great in Australia.

lws-team, 2018-02-18:
I'm back home now and reproduced it.  After some poking around I believe I found the root cause both for this and the original "where did my frame headers go" problem.

Please give the current stuff a try.

FredrikFornstad, 2018-02-19:
Great, it now works! 
I have tested with Edge on Windows. It will take a few days before I can test with FireFox, but IF the problem still exist there, I will re-open this issue.

closed 2018-02-19 by FredrikFornstad

lws-team, 2018-02-19:
Super... I tested it on ffox + chrome on Linux.  The difference we were seeing between the browsers at the end was to do with if they had more than one frame of http/2 DATA arrived coalesced at the ESP32, lws had a bug it didn't respect the logical frame boundary properly then, just treated it all as DATA.  Normally what arrived was packets that ended on the frame boundary, concealing the bug.  So thanks a lot for the help.