warmcat/lws-esp32-factory #29: Can we upload binary using websocket protocoll instate of http forms

nodi-andy, 2018-02-14:
hi,

i can upload binaries with `curl --form @ota=/` etc. But is there a way the factory downloads the binaries or connects to another websocket and gets binaries using websocket?

thanks in advance

lws-team, 2018-02-15:
There is actually stuff in the esp32-related plugins where the device itself will go out in client mode and look for newer firmware from a remote server.  I checked this basically worked when I wrote it, but I haven't tested it in a while.

https://github.com/warmcat/libwebsockets/blob/master/plugins/protocol_esp32_lws_scan.c#L216
https://github.com/warmcat/libwebsockets/blob/master/plugins/protocol_esp32_lws_scan.c#L1136

If you define these things

Var|example
---|---
CONFIG_LWS_MODEL_NAME|"bfg9000"
CONFIG_LWS_OTA_SERVER_BASE_URL|"/url-base-on-server", eg, "/updates"
CONFIG_LWS_OTA_SERVER_FQDN|"myserver.com"

It will check for, eg

https://myserver.com/updates/bfg9000/manifest.json

This is the same file generated during build in ./build.  If it wants it, it will download .../bfg9000/projectname.bin, again from ./build.

at startup when it gets internet in factory mode.  It gets projectname.bin from the existing firmware's manifest.

nodi-andy, 2018-02-15:
do you think i can convert these HTTP POST Requests:

LWS_CALLBACK_HTTP_BODY
LWS_CALLBACK_HTTP_BODY_COMPLETION

to WebSocket requests and change POST Reqest parsing to websocket parsing.

This can be the way?

Downloading a file is good but think to opportunity that a factory can send userapp to another factory via websocket. This sound very interessting and i can just use my server with websocketd to send userapp.bin

would you prefer that i should use libwebsocket with c++ in ubuntu server to transmit userapp.

I would appreciate if you give hints about my concept because i think this kind of a concept has a good future.

thanks

nodi-andy, 2018-02-15:
i found out, that you handle http and ws protocoll in same routine?? That makes the things easier,

Is there a format that i can start ws.send(....) and the data goes into esp_ota_begin, ota_write and ota_end??

lws-team, 2018-02-15:
You can have it also support wss, but overall, isn't it easier to do it with https, and copy the files somewhere that any http server can serve them?  Ws doesn't really buy you anything afaik.

nodi-andy, 2018-02-16:
how do you download a file?
And if you download, than you have to read and call esp_ota_begin , write, end anyway. But i am also OK with this solution, just think that it is the same complicity.

Can you give me a hint or i will try to route ws data to esp_ota_begin etc...

lws-team, 2018-02-16:
https://github.com/warmcat/libwebsockets/blob/master/plugins/protocol_esp32_lws_scan.c#L216
https://github.com/warmcat/libwebsockets/blob/master/plugins/protocol_esp32_lws_scan.c#L1136

nodi-andy, 2018-02-18:
ok, i've solved. It is the best practice to change server data from sdkconfig

closed 2018-02-18 by nodi-andy

lws-team, 2018-02-18:
It does let you control it from sdkconfig

```
CONFIG_LWS_OTA_SERVER_FQDN="libwebsockets.org"
CONFIG_LWS_OTA_SERVER_BASE_URL="/esp32-ota"
CONFIG_LWS_OTA_SERVER_UPLOAD_USER="root"
CONFIG_LWS_OTA_SERVER_UPLOAD_PATH="/var/www/libwebsockets.org"
```

Btw there is a make target `make upload` that lets you scp the current project binary to the remote server along with the manifest, using the SERVER_UPLOAD_USER / _PATH vars.