warmcat/lws-esp32-factory #32: Auto-update not working

snoutmate, 2018-02-26:
Hello, the Auto-update feature is not working, because Makefile sets LWS_IS_FACTORY_APPLICATION to '1' while libwebscokets/plugins/protocol_esp32_lws_scan.c is comparing it to 'y'.

#if CONFIG_LWS_IS_FACTORY_APPLICATION == 'y' ...

Thanks.

lws-team, 2018-02-27:
I'm willing to believe something broke with the update stuff, but confusing as it is, LWS_IS_FACTORY_APPLICATION (== 1) is completely distinct from CONFIG_LWS_IS_FACTORY_APPLICATION (==y).

The latter is set in sdkconfig which for me is

    CONFIG_LWS_IS_FACTORY_APPLICATION=y

We pass LWS_IS_FACTORY_APPLICATION into the build as it is, without any CONFIG_ ... I think nothing uses it atm

    -DLWS_IS_FACTORY_APPLICATION=$(LWS_IS_FACTORY_APPLICATION)

jas39, 2018-04-02:
This is due to an error in components/libwebsockets/plugins/protocol_esp32_lws_scan.c where the following #ifdef on row 218 should be changed from
#if CONFIG_LWS_IS_FACTORY_APPLICATION == 'y' && defined(CONFIG_LWS_OTA_SERVER_BASE_URL) && defined(CONFIG_LWS_OTA_SERVER_FQDN)

to 

#if defined(CONFIG_LWS_IS_FACTORY_APPLICATION) && defined(CONFIG_LWS_OTA_SERVER_BASE_URL) && defined(CONFIG_LWS_OTA_SERVER_FQDN)

(it is actually defined as == 1 in the generated sdkconfig.h  but the above is the way it is used in other files)