warmcat/lws-esp32-factory #33: lws-esp32-factory/components/bootloader/Makefile.projbuild:37: *** multiple target patterns.  Stop.

markosankovic, 2018-03-10:
Hi,

What would be the reason for the following error:

    lws-esp32-factory/components/bootloader/Makefile.projbuild:37: *** multiple target patterns.  Stop.

I followed the README:

```
➜  github:warmcat git clone git@github.com:warmcat/lws-esp32-factory.git
Cloning into 'lws-esp32-factory'...
remote: Counting objects: 733, done.
remote: Total 733 (delta 0), reused 0 (delta 0), pack-reused 733
Receiving objects: 100% (733/733), 1.22 MiB | 523.00 KiB/s, done.
Resolving deltas: 100% (270/270), done.
➜  github:warmcat cd lws-esp32-factory
➜  lws-esp32-factory git:(master) git submodule update --init --recursive
Submodule 'components/libwebsockets' (https://github.com/warmcat/libwebsockets) registered for path 'components/libwebsockets'
Submodule 'components/mbedtls' (https://github.com/lws-team/mbedtls.git) registered for path 'components/mbedtls'
Cloning into '/Users/marko/github:warmcat/lws-esp32-factory/components/libwebsockets'...
Cloning into '/Users/marko/github:warmcat/lws-esp32-factory/components/mbedtls'...
Submodule path 'components/libwebsockets': checked out 'a0581a926bb841ca6f2e1299d47206e67ee780f9'
Submodule path 'components/mbedtls': checked out '4efb69fc0c5c7a007395d55044fafb86a97ec4ed'
➜  lws-esp32-factory git:(master) make erase_flash
/Users/marko/github:warmcat/lws-esp32-factory/components/bootloader/Makefile.projbuild:37: *** multiple target patterns.  Stop.
```
The hello_world example from https://github.com/espressif/esp-idf works as expected and I can run `make erase_flash` from there successfully.

I'm using the SparkFun ESP32 Thing on macOS High Sierra.

Thanks!

lws-team, 2018-03-10:
Esp-idf provides the build pieces, but it changes rapidly.  To rule that out as the source of the problem, read tge README.md on -factory and force the esp-idf commit to one that has been tested as shown there.

markosankovic, 2018-03-10:
The commit is already at cded92bd3c39232295be6d901befe847db22bc74:
```
commit cded92bd3c39232295be6d901befe847db22bc74 (HEAD -> master)
Merge: c3bec5b1 4322e31c
Author: Angus Gratton <angus@espressif.com>
Date:   Tue Feb 20 04:29:40 2018 +0800

    Merge branch 'feature/easy_lookup_of_errors' into 'master'

    Make it easier to look up error messages

    See merge request idf/esp-idf!1929
```

markosankovic, 2018-03-10:
The issue is solved. I cloned lws-esp32-factory into ~/github:warmcat and since it has colon character in the path the build failed. I just moved the cloned directory into another without colon in the name.

closed 2018-03-10 by markosankovic