warmcat/lws-esp32-factory #34: oto_choose_part will accept partition outside range

jas39, 2018-04-03:
I believe that the following statement on line 75 should be changed from:
		if (part->subtype < ESP_PARTITION_SUBTYPE_APP_OTA_MIN ||
		    part->subtype >= ESP_PARTITION_SUBTYPE_APP_OTA_MIN +
		    		     ESP_PARTITION_SUBTYPE_APP_OTA_MAX)
			goto next;

to:
		if (part->subtype < ESP_PARTITION_SUBTYPE_APP_OTA_MIN ||
		    part->subtype >= ESP_PARTITION_SUBTYPE_APP_OTA_MAX)
			goto next;

Since definition of ESP_PARTITION_SUBTYPE_APP_OTA_MAX is:
ESP_PARTITION_SUBTYPE_APP_OTA_MAX = ESP_PARTITION_SUBTYPE_APP_OTA_MIN + 16,//!< Max subtype of OTA partition

FredrikFornstad, 2018-09-26:
It seems that this fix has been implemented a few months ago, so I guess this issue can be closed...

jas39, 2018-10-06:
Agree

closed 2018-10-06 by jas39