warmcat/lws-esp32-factory #35: INFO: ota update of factory partition fails

jas39, 2018-04-03:
The ota update of factory partition fails since esp-idf now have checks to make sure that spi operations are not done on active partition. These checks can be disabled in the 'make menuconfig' -> Component Config -> SPI Flash driver -> Writing dangerous flash regions

This will unfortunately disable these checks for the whole bootloader and create a security hole.

Is there a way to rewrite the ota-invalidation found in libwebsockets/lib/plat/lws-plat-esp32.c on line 1681 and make this part of the code the only one allowed to overwrite active partition?

lws-team, 2018-04-03:
> esp-idf now have checks

> disable these checks for the whole bootloader

> rewrite the ota-invalidation found in libwebsockets

If I understood it, it's esp-idf that decided there's only one kind of "dangerous flash region" and says "no", not lws.  So a solution would be esp-idf separates bootloader overwrite and active partition overwrite.

jas39, 2018-04-04:
That is of course one solution but it might not happen soon. I thought the idea of lws-factory was to build upon an existing OS using the underlying functions to provide the service. In this case the invalidation function seems not to be working.

A more secure way of doing it would be that instead of trying to invalidate the partition info do something like:

Set a flag in NVS that ota_x is supposed to be cleared
Execute from the boot partition and IF boot is working then invalidate the ota_x partition

In that way you make sure that the boot partition actually is OK before invalidating a working partition.

lws-team, 2018-04-04:
> That is of course one solution but it might not happen soon.

Stuff that's in esp-idf domain is in their domain... it's not really my problem.

> A more secure way of doing it would be

Patches are very welcome.

I didn't complete this update stuff before I had to work on other things, so if you have a better vision for how to complete it, I would be very happy to cooperate on it.

So we are clear the idea at the moment is that all updates go into the OTA area.  On reboot, the factory / bootloader stuff may see from the manifest there is a factory image in the OTA area.  In that case, it will copy it to the factory area and boot into factory so you can upload a new OTA image into the OTA area.

Incomplete uploads should boot into factory for recovery.

jas39, 2018-04-04:
The idea seems sound but is not the behaviour I observe. It will copy it to the boot area, try to invalidate the OTA partition and  boot into factory but since invalidation isnt successfull you will end up in OTA again repeating the procedure of copy, invalidate etc.

I'll work on it some more and see if I can come up with something that I can contribute with. I'm far from as efficient with coding as you are :-) . What would the right procedure be to create a pull request?

Do you use this code/mechanisms for any live product deployments?

lws-team, 2018-04-04:
I use the code myself and I know from support requests others are using it in shipping products.  What they have done about updates I dunno; I don't owe anybody anything.  However cooperation is a different matter, I provided all this stuff in order to get contributions that help me too.  So if it looks useful to me cooperation is good.

If PRs are a stumbling block don't worry about it, emailing or pasting the result of `git diff` is usually enough.  If you want to learn about making PRs generally, the idea is

1) click on the `> fork` button at the top right of this project.  That will make you a copy under your own name in your own repo... it will be like https://github.com/jas39/lws-esp32-factory

2) commit your changes in your local git on top of the default master branch in your local copy.  (hint `git commit --amend` is your friend)

3) push your changed copy with your patch(es) to your forked repo, it will be like `git push ssh://git@github.com/jas39/lws-esp32-factory +master:master`  with that bit at the end, the remote repo is like a puppet you can force to the state of your local repo, as many times as you like and without caring about being consistent with what was there before, so don't worry about mistakes.

4) check in https://github.com/jas39/lws-esp32-factory if it reflects your local state.  At this point although what you pushed is public, you can change your local stuff and re-push until it is right, nobody is getting notified yet.

5) There will be a "create pull request" button or so on your remote repo... create it against this repo master branch

If you need to patch on lws, it's the same flow but starting from the lws repo, push to your forked lws repo, and make a PR against the original lws repo.

As I say if the code is good I don't care how it arrives, pasting `git diff`s is fine.  But the above is a good skill to have anyway.

FredrikFornstad, 2018-09-14:
I have not seen any progress on this so here is my proposal for fixing this: Simply add a new magic byte pattern and let the bootloader erase the OTA partition header instead of trying erase the header of the partition that is currently executing.

In libsockets:
`diff --git a/include/libwebsockets/lws-esp32.h b/include/libwebsockets/lws-esp32.h
index e11d735e..2d1c0f42 100644
--- a/include/libwebsockets/lws-esp32.h
+++ b/include/libwebsockets/lws-esp32.h
@@ -109,7 +109,7 @@ static LWS_INLINE void uv_close(uv_handle_t *h, void *v)
 #define LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY 0xb00bcafe
 #define LWS_MAGIC_REBOOT_TYPE_FORCED_FACTORY 0xfaceb00b
 #define LWS_MAGIC_REBOOT_TYPE_FORCED_FACTORY_BUTTON 0xf0cedfac
-
+#define LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY_ERASE_OTA 0xfac0eeee

 /* user code provides these */

diff --git a/lib/plat/esp32/esp32-helpers.c b/lib/plat/esp32/esp32-helpers.c
index 9bd4cd4e..11eda871 100644
--- a/lib/plat/esp32/esp32-helpers.c
+++ b/lib/plat/esp32/esp32-helpers.c
@@ -1032,14 +1032,12 @@ lws_esp_ota_get_boot_partition(void)
                                }
                        }

-                       /* destroy our OTA image header */
-                       spi_flash_erase_range(ota->address, 4096);
-
                        /*
-                        * with no viable OTA image, we will come back up in
+                        * We send a message to the bootloader to erase the OTA header, we will come back up in
                         * factory where the user can reload the OTA image
                         */
                        lwsl_notice("  FACTORY copy successful, rebooting\n");
+                       lws_esp32_restart_guided(LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY_ERASE_OTA);
 retry:
                        esp_restart();
                }

`

and in lws-esp32-factory:
`diff --git a/components/bootloader/subproject/main/bootloader_start.c b/components/bootloader/subproject/main/bootloader_start.c
index 300c5e3..1acf872 100644
--- a/components/bootloader/subproject/main/bootloader_start.c
+++ b/components/bootloader/subproject/main/bootloader_start.c
@@ -24,6 +24,7 @@
 #include "bootloader_common.h"
 #include "sdkconfig.h"
 #include "esp_image_format.h"
+#include "bootloader_flash.h"

 static const char* TAG = "boot";

@@ -48,6 +49,7 @@ static bool check_force_button(void)
 #define LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY 0xb00bcafe
 #define LWS_MAGIC_REBOOT_TYPE_FORCED_FACTORY 0xfaceb00b
 #define LWS_MAGIC_REBOOT_TYPE_FORCED_FACTORY_BUTTON 0xf0cedfac
+#define LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY_ERASE_OTA 0xfac0eeee

 static int select_partition_number (bootloader_state_t *bs);
 static int selected_boot_partition(const bootloader_state_t *bs);
@@ -73,8 +75,12 @@ void call_start_cpu0()
     /* we can leave a magic at end of fast RTC ram to force next boot into FACTORY */
        uint32_t *p_force_factory_magic = (uint32_t *)LWS_MAGIC_REBOOT_TYPE_ADS;

-       if (*p_force_factory_magic == LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY) {
+       if (*p_force_factory_magic == LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY ||
+                       *p_force_factory_magic == LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY_ERASE_OTA) {
                boot_index = FACTORY_INDEX;
+               if (*p_force_factory_magic == LWS_MAGIC_REBOOT_TYPE_REQ_FACTORY_ERASE_OTA)
+                       bootloader_flash_erase_range(bs.ota[0].offset, 4096);  // Erase the OTA header to mark it empty
+
                /* mark as having been forced... needed to fixup wrong reported boot part */
                *p_force_factory_magic = LWS_MAGIC_REBOOT_TYPE_FORCED_FACTORY;
        } else
`

FredrikFornstad, 2018-09-14:
Sorry about the formatting in my previous post. I guess its a Windows/edge thing…  (I did use the "insert code")

lws-team, 2018-09-16:
Thanks a lot for the patches... they're both pushed and the two projects aligned on lws with them.

For quoting code, the magic ingredient is three backticks at the start and end.

closed 2018-09-16 by lws-team