warmcat/lws-esp32-factory #36: lws-esp32-factory is always listed as dirty

JimGaluska, 2018-04-09:
**Issue**:  In the device setup screen, the git information for lws-esp32-factory is always listed as 'dirty'.

**Cause**:   Before the application can be compiled, 'make menuconfig' must be run to generate the file 'sdkconfig'.  This generates file 'sdkconfig', which causes git to mark the repository as dirty due to an untracked file.  This , in turn, causes lws-esp32-factory to list itself as 'dirty' in the status window or the Device Update page.

**Cure**:  add 'sdkconfig' to the project's '.gitignore' file

lws-team, 2018-04-09:
Do you have any idea about how to square this with wanting to provide and maintain a reference sdkconfig?

Every time I update lws-esp-factory against esp-idf, there are new sdkconfig options... most of these are irrelevant, but some of these are critical design decisions about integration with lws at all (eg, lws provides dynamic "stretchy" tls buffers by patching mbedtls, which requires the related hack option in sdkconfig to be 16384).  Another example is the hacks in lwip around "offset for socket fds".

If users don't align the sdkconfig with the necessary options for lws to do what it's doing, it won't work.

JimGaluska, 2018-04-09:
This request is not asking for file 'sdkconfig' to be added to the repository.  This request is to add a line containing the file name to the existing '.gitignore' file.

From a user perspective, this change will be virtually invisible.  He will still have to run 'make menuconfig' before he compiles.  However git will no longer call the repository dirty when the file 'sdkconfig' is present.

lws-team, 2018-04-09:
... ignoring a file is inconsistent with git tracjing the file, right?  Sdkconfig is already in the project

https://github.com/warmcat/lws-esp32-factory/blob/master/sdkconfig

Adding stuff being tracked to gitignore has no effect...

https://help.github.com/articles/ignoring-files/

What you suggested means **removing** sdkconfig from the project.

Please re-read my original reply with this in mind.

JimGaluska, 2018-04-10:
I now understand the issue and see where I was mistaken.  I agree that the way the code is now is correct.  Please close this issue.

Thanks
Jim

closed 2018-04-10 by lws-team