warmcat/lws-esp32-factory #37: Does it link on ESP32

snahmad, 2018-06-19:
I am getting linker errors.

C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipper10closeEntryEv+0x0): undefined reference to `unzCloseCurrentFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipper9openEntryEPKc+0x0): undefined reference to `unzLocateFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipper9openEntryEPKc+0x4): undefined reference to `unzOpenCurrentFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipper5closeEv+0x0): undefined reference to `unzClose'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipper12getEntrySizeEv+0x0): undefined reference to `unzGetCurrentFileInfo64'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipperrsERSo+0x0): undefined reference to `unzReadCurrentFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipper11readEntriesEv+0x0): undefined reference to `unzGetGlobalInfo64'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipper11readEntriesEv+0x4): undefined reference to `unzGoToNextFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o):(.literal._ZN8ziputils8unzipper4openEPKc+0x0): undefined reference to `unzOpen64'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o): In function `ziputils::unzipper::closeEntry()':
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/char_traits.h:243: undefined reference to `unzCloseCurrentFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o): In function `ziputils::unzipper::openEntry(char const*)':
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/char_traits.h:243: undefined reference to `unzLocateFile'
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/char_traits.h:243: undefined reference to `unzOpenCurrentFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o): In function `ziputils::unzipper::close()':
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/char_traits.h:243: undefined reference to `unzClose'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o): In function `ziputils::unzipper::getEntrySize()':
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/char_traits.h:243: undefined reference to `unzGetCurrentFileInfo64'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o): In function `ziputils::unzipper::operator>>(std::basic_ostream<char, std::char_traits<char> >&)':
c:\msys32\opt\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\5.2.0\bits/char_traits.h:243: undefined reference to `unzReadCurrentFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o): In function `ziputils::unzipper::readEntries()':
C:/Work/LibDev/esp32/esp-idf/components/cpp_utils/unzipper.cpp:138: undefined reference to `unzGetGlobalInfo64'
C:/Work/LibDev/esp32/esp-idf/components/cpp_utils/unzipper.cpp:146: undefined reference to `unzGetCurrentFileInfo64'
C:/Work/LibDev/esp32/esp-idf/components/cpp_utils/unzipper.cpp:159: undefined reference to `unzGoToNextFile'
C:/Work/Actisense/esp32/build/cpp_utils\libcpp_utils.a(unzipper.o): In function `ziputils::unzipper::open(char const*)':
C:/Work/LibDev/esp32/esp-idf/components/cpp_utils/unzipper.cpp:30: undefined reference to `unzOpen64'

snahmad, 2018-06-20:
https://github.com/warmcat/lws-esp32-factory/tree/master/components/zlib/contrib/minizip

is not building for esp32.

lws-team, 2018-06-20:
I have no idea what you're trying to do, but the zlib component builds fine if you build it as part of, eg, the factory sample.  You seem to be building something in esp-idf... either way it's not my problem.

closed 2018-06-20 by lws-team