warmcat/lws-esp32-factory #38: Connection terminates after approximetly

formus14, 2018-07-16:
Hi,

I Just noticed a strange issue when trying to connect from a WSS client to the ESP board Connection opens successfully normally and it functions as expected but after appx. 10 minutes,  i see a notice message that the heap is just occupied with some data and then the connection is terminated.

P.S. I am using IDF3 ( the latest version of the framework)

Console is down below:

> ESP just recieved this from WSS client: {"command":"ek_keepAlive","id":193}email":"defaultadmin@xxxx","password":"xxxxxx"},"id":129}et-Protocol: http-only
> Host: 192.168.1.1 
> ▒'▒▒    H▒F▒z▒)�▒oy/d
>
> JSON Package back over WSS 
> ESP sent 31 bytes succesfully. Frame Type: 0. Bytes: {
>         "id":   193,
>         "status":       "ok"
> }
>
> [1970/01/01 00:13:33:0110] NOTICE:  heap :4235516 (-1272)
> 
> writeable
> disconnect in lws callback
> I (832578) lwscallback:  # SSL CONNECTION TERMINATED
> 
> [1970/01/01 00:13:52:5613] DEBUG:
> [1970/01/01 00:13:52:5621] DEBUG: 0000: 03 E9                                              ..
> [1970/01/01 00:13:52:5661] DEBUG:
> writeable
> disconnect in lws callback
> I (832608) lwscallback:  # SSL CONNECTION TERMINATED
> 
> I (832608) lwscallback:  # SSL CONNECTION CLOSED
> 
> I (832618) lwscallback:  # WSI DESTROY

any help is appreciated :)

lws-team, 2018-07-16:
Heap is a global resource, so anything can corrupt it.

The only way to attack it afaik is start removing and disabling things, or increasing the frequency of things until you can associate something with the problem.

This spi as heap thing would be #1 on my list...

Lws itself has a life outside of esp32 where it runs under valgrind.  So core lws wouldn't be my first suspect.

You need to do some work to try to point tge finger yourself.

formus14, 2018-07-18:
Just increased the main task size and it worked fine.

Thanks for the tip, ill close the issue now :)

closed 2018-07-18 by formus14