warmcat/lws-esp32-factory #54: build fail for lws-esp32-factory

KishanPala556, 2019-11-25:
I am building lws-esp32-factory from windows 10 using MINGW32 as per suggest here: https://libwebsockets.org/git/lws-esp32-factory.
but getting some error for ASIO as below:

lws-esp32-factory/esp-idf/components/asio/asio/asio/include/asio/execution_context.hpp:127:27: error: macro "shutdown" requires 2 arguments, but only 1 given ASIO_DECL void shutdown();

lws-esp32-factory/esp-idf/components/asio/asio/asio/include/asio/detail/scheduler.hpp:66:50: error: macro "poll" requires 3 arguments, but only 1 given ASIO_DECL std::size_t poll(asio::error_code& ec);

lws-esp32-factory/esp-idf/components/asio/asio/asio/include/asio/detail/socket_ops.hpp:67:47: error: macro "accept" passed 4 arguments, but takes just 3 std::size_t* addrlen, asio::error_code& ec);

Please let me know what causing this problem.

lws-team, 2019-11-25:
This isn't making any attempt at the moment to keep up with current esp-idf.  If you want to try it, read the README about what to do to esp-idf.

closed 2019-11-25 by lws-team

KishanPala556, 2019-11-25:
I have read about what to do with current esp-idf as it is mentioned to hard reset on last commit in which it is tested, I have did the same but not able to build.

lws-team, 2019-11-25:
From the errors, it looks like whatever defines accept() etc disagrees with esp-idf expectation of it... maybe newer toolchains provide those and are no longer able to build the old esp-idf commit... you can try an older version of the espressif toolchain.

KishanPala556, 2019-11-25:
you are right. I don't find old tool chain but I have removed the ASIO component because it is not needed. but still have following errors:
WARNING: esp-idf git submodule components/coap/libcoap may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/json/cJSON may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/lwip/lwip may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/esptool_py/esptool may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: esp-idf git submodule components/esptool_py/esptool may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
Python requirements from lws-esp32-factory/esp-idf/requirements.txt are satisfied.
CC build/coap/libcoap/src/async.o
In file included from lws-esp32-factory/esp-idf/components/coap/port/include/coap_config.h:22,
                 from lws-esp32-factory/esp-idf/components/coap/libcoap/src/async.c:16:
lws-esp32-factory/esp-idf/components/coap/port/include/coap_config_posix.h:29: warning: "IP_PKTINFO" redefined
 #define IP_PKTINFO   IP_MULTICAST_IF

In file included from /lws-esp32-factory/esp-idf/components/lwip/port/esp32/include/sys/socket.h:33,
                 from /lws-esp32-factory/esp-idf/components/coap/port/include/coap_config_posix.h:23,
                 from /lws-esp32-factory/esp-idf/components/coap/port/include/coap_config.h:22,
                 from /lws-esp32-factory/esp-idf/components/coap/libcoap/src/async.c:16:
/lws-esp32-factory/esp-idf/components/lwip/lwip/src/include/lwip/sockets.h:273: note: this is the location of the previous definition
 #define IP_PKTINFO         8

In file included from /lws-esp32-factory/esp-idf/components/coap/libcoap/src/async.c:17:
lws-esp32-factory/esp-idf/components/coap/port/include/coap/coap.h:20:10: fatal error: libcoap.h: No such file or directory
 #include "libcoap.h"

lws-team, 2019-11-25:
None of this is anything to do with lws... I guess you need to make clean, remove CMakeCache.txt, and start from scratch.  But these are all esp-idf troubles and this isn't the right place.

KishanPala556, 2019-11-25:
There is no such file as CMakeCache.txt. I have tried with make clean but getting same error. yes it is esp-idf troubles but to get rid of it either back porting is required or lws-esp32-factory should be updated as per current esp-idf. I am trying to back port it but it is not good option, it would be appreciated if you can release lws-esp32-factory with current esp-idf.

lws-team, 2019-11-25:
> it would be appreciated if you can release lws-esp32-factory with current esp-idf.

You're very welcome to contribute it.

KishanPala556, 2019-11-25:
Thank you, I am trying to build it with latest esp-idf.

KishanPala556, 2019-11-25:
Almost all errors related to esp-idf is resolved. there is only one issue and that is, In the commit 17ac4bad7381e579e5a7775755cc25480da47d97 there exist components/esp32/lib folder which is empty but in current esp-idf it has been removed, that's why getting below error:
WARNING: Missing submodule components/esp32/lib...
Attempting 'git submodule update --init components/esp32/lib' in esp-idf root directory...
error: pathspec 'components/esp32/lib' did not match any file(s) known to git
let me know how can I solve it.

lws-team, 2019-11-25:
AFAIK the dir itself isn't meaningful for git submodule... there's a file .gitmodules at the top level that knows what should be where.

It sounds like a big red flag though if that file is knowing about older commit situation and you think you are building against latest commit.  It sounds like the state of the checkout is mixed with old and new.

KishanPala556, 2019-11-25:
what step I can take to overcome?

lws-team, 2019-11-25:
It's a condition that exists in your checkout afaik not the world at large.

Check your .submodules against the upstream esp-idf one

https://github.com/espressif/esp-idf/blob/master/.gitmodules

If you made changes in esp-idf save them some where with git diff > ~/mychanges.diff then git reset --hard 93a0603  (there should be no more directory in there that doesn't match upstream) and git submodule update --init --recursive